paveldn/haier-esphome #1: Great job + question on reducing the setting range

sinclairfr, 2022-10-04:
This works great. Thank you for your hard work. 
I wish I knew how to better code to help you out on this project.

I was wondering, how hard would it be to restrict the range of temperature possible in the climate widget ?

I have tenants who don't respect our rules and see the AC to 30 degrees... I wish to keep the available setting range from 16 to 22C.

Any help would be greatly appreciated. 

Thank you

paveldn, 2022-10-04:
Thank you for the kind words. 
I already pushed the change to the dev branch. But you should keep in mind that it will not block the fact that this temperature can be set by remote control these changes restrict only the temperature range in the home assistant interface.
So to change the max and min temperature for the home assistant you need to use the following settings in your climate:
`
    visual:
      max_temperature: 26
      min_temperature: 18
`
It will not allow to set lower min than 16 and higher max than 30.

closed 2022-10-04 by paveldn

sinclairfr, 2022-10-05:
Wow, that's awesome! Thank you so much.
Just FYI, your code works great, out of the box, with my Haier Geos AC unit (AS35TAMHRA).
I've tested out all the commands.
So if people own the same unit, they can go ahead and use it safely.
Thanks again