paveldn/haier-esphome #3: AC turns off after power outage

bearpawmaxim, 2022-10-31:
If an AC was in working state before the power outage, it immediately turns off when the electricity comes back (at that stage when both wifi network and/or HA is not yet available).

When ha and/or wifi network is up, I can't turn on an AC from HA without retoggling the fan speed setting - AC just not reacts on nor operation mode nor temperature change.

This behavior occurs with this component only.

bearpawmaxim, 2022-11-01:
One more thing: when turning AC on, it turns on with 24°C setting

bearpawmaxim, 2022-11-07:
Hi @paveldn, any idea about what is the cause of my issue?

paveldn, 2022-11-08:
@bearpawmaxim 
Sorry I didn't find the root cause of the problem. I just can't reproduce it with my AC. Yes, it resets to 24 degrees (which is expected because the temperature is taken from AC itself and it is losing its settings). But now I am reworking the whole code for AC communication to make it more reliable and easier to support. Let's see if the problem will be gone with those changes.
I will let you know once this new code will be available for testing.

paveldn, 2022-11-08:
Oh, I just realized that probably the power outage problem is much bigger for you than I thought. I will make another attempt to fix it this evening.

bearpawmaxim, 2022-11-08:
@paveldn okay, thanks, will wait.
I think that it is somehow related to a fan/fan-only mode.

bearpawmaxim, 2022-11-08:
> the power outage problem is much bigger for you than I thought

I think that I got used to this) It just annoys me to sometimes turn the AC on, late at night for example))

bearpawmaxim, 2022-11-08:
@paveldn here the logs,
https://pastebin.com/pYu2nyxd - step 1, when can't turn on by setting heat mode. Just after network and ha boot from power outage
https://pastebin.com/gGHB0raG - step 2, when set automatic fan mode
https://pastebin.com/EQ9HPqm6 - step 3, when set heat mode and the AC finally turned on

looks like something's is causing message timeouts at step 1

paveldn, 2022-11-09:
Yes, it looks like AC side problem. Not much I can do.  I have plans to use a different command to control AC probably that one will work better in your case. But it will take some time.

bearpawmaxim, 2022-11-09:
@paveldn, thank you! Any estimates?

Just in case it would be helpful for you: as I mentioned earlier, the [original version](https://github.com/albetaCOM/esp-haier)  does not have such bug

paveldn, 2022-11-10:
@bearpawmaxim 
> Any estimates?
I don't want to give any promises it really depends on how much free time I will have. Rough estimation 2 weeks. 
> Just in case it would be helpful for you: as I mentioned earlier, the [original version](https://github.com/albetaCOM/esp-haier) does not have such bug
I can't reproduce it so not really helpful but it gives hope that the problem will disappear after my ew code will start working.

bearpawmaxim, 2022-11-12:
> I can't reproduce it
BTW, running sequentially actions from [this](https://github.com/bearpawmaxim/hass-config/blob/master/automations/climate.yaml#L1)  and [this](https://github.com/bearpawmaxim/hass-config/blob/master/automations/climate.yaml#L36)  automations also reproduces the problem.

bearpawmaxim, 2022-11-18:
@paveldn, removing both the display/beeper switches and swing mode select (config is [here](https://github.com/bearpawmaxim/esphome-configs/blob/master/.haier_device_common.yaml) ) looks like fixes the problem.. I think that it is related to the switch/select restoring it's state on boot and sends a command to the AC while AC is initializing. For example, `turn_on_action/turn_off_action`s for switch are executing every time the device boots.

bearpawmaxim, 2022-11-21:
Hi @paveldn!
Removing both the display/beeper switches and swing mode select (config is here ) fixes the problem..
I think that it may be somehow related to the switch/select restoring it's state on boot and sends a command to the AC while AC is initializing.
For example, `turn_on_action/turn_off_action`-s for switch are executing every time the device boots.

paveldn, 2022-11-21:
Hi @bearpawmaxim !
Thanks. Will see how I can improve this. Need to dig in ESPHome code for this...

bearpawmaxim, 2022-11-21:
> Will see how I can improve this. Need to dig in ESPHome code for this...

The most easy way is to somehow deny or schedule switch/select (still don't know who of these caused the problem) action until the communication with AC is not initialized.

paveldn, 2022-12-07:
Hi @bearpawmaxim,
I probably found the root cause of this problem (and probably some other similar problems). It looks like in case the data part of the packet (header, data, checksum, or CRC) contains 0xFF, AC adds an additional 0x55 byte and also expects it from the ESP32 module. So if this situation happens in your case after a power cut, AC will ignore our packets (because it is waiting for 1 byte more) until something is changed (and 0xFF disappears from a packet). I am not 100% sure but it looks like it.
I see from your log that your packets at the beginning have FF in a low byte of CRC:
FF FF 14 40 00 00 00 00 00 01 60 01 08 06 81 00 02 01 00 00 00 00 48 E1 FF
So it is ignored. But when you change mode (actually any value) CRC changes and everything works fine.
I will need to change the handling of packets on the lowest level. It will take some time but at least now I know what is happening and why.

bearpawmaxim, 2022-12-07:
Hi @paveldn!
Thanks for your work!
Does this answers the question why AC turns off after the power loss and how this is related to the display/beeper switch? (as we already know it might be related to the executing of the turn on/off actions by the switch on boot)

bearpawmaxim, 2022-12-07:
Okay @paveldn, that sounds logical.
With the only exception that my AC doesn't forgets it's settings during power losses (it restores with the mode and temperature that were set before the power loss).
Anyway, waiting for the solution! Thanks for your work!

evlo, 2022-12-07:
I think forgetting depends of the revision of the mainboard.

bearpawmaxim, 2022-12-09:
Hi @paveldn! 
Have updated the component to the latest version and uncommented display/beeper switch alongside with swing mode select, but unfortunately that haven't helped: AC still turns off and forgets it's settings...
After commenting out the display, beepee and swing mode stuff again AC starts normally and restores the settings.

Can't there be situation when due to switch startup logic the ESP8266 sends an empty command to the AC? (that happens on the early boot stage, long before the 1st status reception)

bearpawmaxim, 2022-12-09:
@paveldn with the new version I can turn the AC on without that fan magic! So we are on the right way.

evlo, 2022-12-09:
To ADD to the confusion, I have tried with AC that did have the CRC issues before and it does remember last settings both with the wifi adapter present and without it the same after power outage.

bearpawmaxim, 2022-12-10:
> To ADD to the confusion, I have tried with AC that did have the CRC issues before and it does remember last settings both with the wifi adapter present and without it the same after power outage.

As I mentioned earlier the problem is caused by the display/beeper switch and/or swing mode select restore at boot logic: it executes on_turn_off/on_turn_on actions on the early boot stage and likely sends an empty commands to the AC.
Maybe you are using the component without that switches/selects ?

paveldn, 2022-12-10:
@bearpawmaxim  @evlo 
Thanks for the update. With confirmation that the CRC issue is fixed, I can try to make a fix or at least a workaround.

paveldn, 2022-12-26:
@bearpawmaxim  Can you check if power on problems still happens on the latest dev branch commits?

bearpawmaxim, 2022-12-26:
Hi @paveldn! Thanks for you work! Power off after power on is fixed!!)
The only things that remains is:
1. The display on AC blinks after power on
2. The outdoor temperature sensor reports -64 for some time after power on
but that is for another issues I think/

closed 2022-12-26 by bearpawmaxim

evlo, 2023-01-13:
_Haha I was wondering if it actually can froze that cold during heating cycle. :D_