paveldn/haier-esphome #7: Outdoor temperature sensor reports -64 right after power on

bearpawmaxim, 2022-12-26:
The outdoor temperature sensor reports -64 right after power on, with time the issue fixes by itself (when the ESP receives an actual status from the AC I think)

The logs are here https://pastebin.com/q19HraL4

paveldn, 2022-12-28:
It looks like it takes some time for the external module to show temperature. Not sure why this problem only popped up with new code it should also happen with the previous implementations.  -64 is offset for the outdoor temperature sensor. So on the startup outdoor module still report 0 for some time (and according to protocol docs it is the correct temperature). I wrote a workaround. I ignore any outdoor temperature until we got at least one non-zero value. It can cause problems if the outside is really -64 but in this case, the failed outdoor sensor will not be the biggest one 🤣 . Can you try the latest dev branch commit?

bearpawmaxim, 2022-12-28:
Updated AC units, will wait for the next power on (approximately in 4 hours and 15 minutes)

bearpawmaxim, 2022-12-30:
Hi @paveldn! The issue is fixed, thanks for your work!

closed 2022-12-30 by bearpawmaxim