paveldn/haier-esphome #9: ESP32 haier module doesn't connect to wifi after flashing

Napalmcold, 2023-03-26:
Hi! I got 3 Haier Flexis units with those modules. The mobile app is called EVO, not hOn and not smartair2.
Successfully made a backup following your guide. But after flashing the module through the esphome addon in home assistant, it doesn't show up on network. I tried different yaml configs from your repository, but everytime same result: no connection to wifi network and no falback wifi hotspot. 
After fIashing I get these logs while connected to serial without flashing mode:

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

When I flash it back to the original firmware it works well again.
What am I doing wrong?

paveldn, 2023-03-27:
Hi @Napalmcold ,

First of all great that you have made a backup. Just keep in mind that there are unique keys for every module so if you have three of them you will need to make a backup per module and remember which backup from which module.
About your problem: 
I was not able to find the EVO app in the play store. Apparently, it is some regional version of the application unavailable in my region. Anyway, the message you see is not from ESPhome itself but from the bootloader and it is a flash reading error. Something should be different about your hardware. Did you change something in the esp32 section of your yaml config? I suppose not just need to make sure :) 
To check your hardware I need to see your chip id. Please, put the device in flashing mode (the same way you did for making a backup) and run esptool with the following command:
`python esptool.py --port <port_name> chip_id`
and post your results here.  
It should be something like this:
```
Detecting chip type... ESP32
Chip is ESP32-S0WD (revision 1)
Features: WiFi, BT, Single Core, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: XX:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB

```

I am interested in chip type and flash size. You can remove your board MAC it is not needed.

Napalmcold, 2023-03-27:
> Hi @Napalmcold ,
>
> First of all great that you have made a backup. Just keep in mind that there are unique keys for every module so if you have three of them you will need to make a backup per module [...]

Hi, Pavel!

First of all, thanks for your respond, I much appreciate that.

Here is the config, I've only changed external components path to github:

```
substitutions:
  device_name: Haier AC
  device_id: haier_climate
  uart_id: ac_port 
  send_wifi: "true"

esphome:
  name: haier
  friendly_name: Haier
  name_add_mac_suffix: true

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

uart:
  baud_rate: 9600
  tx_pin: 17
  rx_pin: 16
  id: ${uart_id}  

logger:
  level: DEBUG

external_components:
  - source: github://paveldn/ESP32-S0WD-Haier@dev
    
api:
  encryption:
    key: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

ota:
  password: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

climate:
  - platform: haier
    id: ${device_id}
    protocol: hOn
    name: ${device_name} 
    uart_id: ${uart_id}
    wifi_signal: ${send_wifi}   # Optional, default true, enables WiFI signal transmission from ESP to AC
    beeper: true                # Optional, default true, disables beep on commands from ESP
    outdoor_temperature:        # Optional, outdoor temperature sensor
      name: ${device_name} outdoor temperature
    visual:                     # Optional, you can use it to limit min and max temperatures in UI (not working for remote!)
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:            # Optional, can be used to disable some modes if you don't need them
    - 'OFF'                     # always available
    - AUTO                      # always available
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    supported_swing_modes:      # Optional, can be used to disable some swing modes if your AC does not support it
    - 'OFF'                     # always available
    - VERTICAL                  # always available
    - HORIZONTAL
    - BOTH

switch:
  - platform: template
    id: ${device_id}_display_switch
    name: ${device_name} display
    icon: mdi:led-on
    entity_category: config
    restore_state: yes
    lambda: |-
      return id(${device_id}).get_display_state();
    turn_on_action:
      climate.haier.display_on: ${device_id}
    turn_off_action:
      climate.haier.display_off: ${device_id}
  - platform: template
    id: ${device_id}_beeper_switch
    entity_category: config
    name: ${device_name} beeper
    icon: mdi:volume-high
    restore_state: yes
    lambda: |-
      return id(${device_id}).get_beeper_state();
    turn_on_action:
      climate.haier.beeper_on: ${device_id}
    turn_off_action:
      climate.haier.beeper_off: ${device_id}
  - platform: template
    id: ${device_id}_health_mode
    name: ${device_name} health mode
    icon: mdi:leaf
    restore_state: yes
    lambda: |-
      return id(${device_id}).get_health_mode();
    turn_on_action:
      climate.haier.health_on: ${device_id}
    turn_off_action:
      climate.haier.health_off: ${device_id}
  - platform: restart
    name: ${device_name} restart

select:
    # Vertical airflow direction setting. Works only if the vertical swing is off
    # You can remove this control if you don't need it
  - platform: template
    id: ${device_id}_vertical_direction
    name: ${device_name} airflow vertical
    entity_category: config
    icon: mdi:arrow-expand-vertical
    optimistic: true
    restore_value: true
    initial_option: Center 
    options:
      - Up
      - Center
      - Down
    on_value:
      then:
        - climate.haier.set_vertical_airflow:
           id: ${device_id}
           vertical_airflow: !lambda |-
             return esphome::haier::AirflowVerticalDirection(i);
    # Horizontal airflow direction setting. Works only if the horizontal swing is off
    # You can remove this control if you don't need it
  - platform: template
    id: ${device_id}_horizontal_direction
    name: ${device_name} airflow horizontal
    entity_category: config
    icon: mdi:arrow-expand-horizontal
    optimistic: true
    restore_value: true
    initial_option: Center 
    options:
      - Left
      - Center
      - Right
    on_value:
      then:
        - climate.haier.set_horizontal_airflow:
           id: ${device_id}
           horizontal_airflow: !lambda |-
             return esphome::haier::AirflowHorizontalDirection(i);
text_sensor:
  - platform: template
    name: ${device_name} cleaning status
    icon: mdi:spray-bottle
    lambda: return id(${device_id}).get_cleaning_status_text();

button:
  - platform: template
    name: ${device_name} start self cleaning
    icon: mdi:spray-bottle
    on_press:
      - climate.haier.start_self_cleaning: ${device_id}
  - platform: template
    name: ${device_name} start 56°C steri-cleaning
    icon: mdi:spray-bottle
    on_press:
      - lambda: id(${device_id}).start_steri_cleaning();
```

And herу is the chip_id command results::

```
esptool.py v4.5.1
Serial port COM6
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Chip is ESP32-S0WD (revision v1.0)
Features: WiFi, BT, Single Core, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: XX:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: XX:XX:XX:XX:XX:XX
Hard resetting via RTS pin...
```

Napalmcold, 2023-03-27:
Also I've just noticed that the mac of module I'm trying to flash begins with 40:22:d8, while the other 2 modules have macs begining with CC:DB:A7. Looks like they are different. Later I'll try to flash one of those.

paveldn, 2023-03-27:
MAC shouldn't be important I have a similar situation 3 of my modules starts with 24:A1:60 and 1 starts with C4:DD:57 and all of them worked fine. Your chip info looks the same, it should be OK. The only explanation I have is that Haier used in the new version of the module different flash mode. You can try to play with different modes, and add in esphome section the board_build.flash_mode field:
```
esphome:
  name: haier
  friendly_name: Haier
  name_add_mac_suffix: true
  platformio_options:
    board_build.flash_mode: dio
```
Use this link https://docs.platformio.org/en/latest/platforms/espressif32.html#flash-mode to see available options, and try them one by one. Let me know if it worked.

Napalmcold, 2023-03-27:
> @Napalmcold Can you, please, run esptool command but with flash_id instead of chip_id? Maybe the size of the flash is different

Here's what I got:
```
Serial port COM7
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Chip is ESP32-S0WD (revision v1.0)
Features: WiFi, BT, Single Core, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: XX:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

paveldn, 2023-03-28:
Hi @Napalmcold ,
It looks exactly like mine. Size frequency etc. The problem here is not in the Haier component or even ESPHome itself. It is a bootloader that can't start ESPHome because it can't read flash. So it does not matter if you start it connected to HVAC or not. Actually, you can try to flash it with plain ESPHome without the Haier component and without the UART component and you will get the same problem.  I found on esp-idf forums topic where people complain that this kind of problem can happen if you flash your module when it is powered from 3.3V and not 5V. I was flashing mine with 3.3V power without any problems but who knows? Can you please use 5V instead of 3.3V and use the EN contact of the same UART (or use 5v and GND contacts of connector that connects to HVAC) and repeat flashing procedure?

Napalmcold, 2023-03-28:
Same situation after 5V flashing(

Napalmcold, 2023-03-28:
> Hi @Napalmcold , It looks exactly like mine. Size frequency etc. The problem here is not in the Haier component or even ESPHome itself. It is a bootloader that can't start ESPHome because it can't [...]

Same situation after 5V flashing(

Napalmcold, 2023-03-28:
Maybe I should try to flash it not through ESPHome addon, but through python esphome? How do you usually flash those modules?

paveldn, 2023-03-28:
> Maybe I should try to flash it not through ESPHome addon, but through python esphome? How do you usually flash those modules?

I am usually using python esphome. Both ways should work, but let's try this. Also, you can try to use this tool: https://github.com/esphome/esphome-flasher
Not sure if it helps but a lot of people recommend it. You can also install it for python with pip.
If it will not help we can try to update the bootloader, just not sure how to do it and not to kill the board. Will need to do some experiments with my board first.

Napalmcold, 2023-03-28:
Here is what I get when I try to flash it through esphome flasher on windows:

```
Using 'COM6' as serial port.
Connecting.......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-S0WD (revision 1)
 - Number of Cores: 1
 - Max CPU Frequency: 80MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 40:22:D8:11:14:60
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=23, should be E9)
```
I get this error even when I try to flash the stock firmware backup

paveldn, 2023-03-28:
Can you also check your partitions layout? In original module backup. For example, this is for one of my modules:
[image]
Just use any hex viewer and check offset 0x8000 from the beginning

Napalmcold, 2023-03-28:
Here's what I see in HEX. Couldn't get better result playing with encoding. I'm pretty new with HEX editors
[image]

Napalmcold, 2023-03-28:
> What are you flashing? Firmware should always start with 0xE9. In my case it is. It should be file firmware.bin in .esphome\build\haier.pioenvs\haier folder. If it is not starting with E9 something is wrong with esp-idf compiler in your case

Yeah, you're right, I tried to flash firmware-factory.bin... With factory.bin all flashes as expected. But result is the same again.

paveldn, 2023-03-29:
Yes, it doesn't look like a partition section.  Can you send me the original backup you made? I will try to parse flash structures and see what the hell is happening. There is nothing really secret or important inside so no worries. There is a key for Haier cloud servers but I have them anyway from my own modules. Anyway, I will understand if you will decide not to send it. In case you are willing to send my email paveldudn@gmail.com.
I will do my best to figure out how to make your modules work but usually, these things take time, so in case you want a faster solution you can use any other ESP module (I would recommend ESP32 but ESP89266 also will work). For example with one of my ACs I am using M5Stamp C3 https://shop.m5stack.com/collections/m5-controllers/products/m5stamp-c3-mate-with-pin-headers (to have the original module for experiments) and it works perfectly well. But of course, it is up to you.

Napalmcold, 2023-03-29:
> Yes, it doesn't look like a partition section. Can you send me the original backup you made? I will try to parse flash structures and see what the hell is happening. There is nothing really secret or important inside so no worries. There is a key for Haier cloud servers but I have them anyway from my own modules. Anyway, I will understand if you will decide not to send it. In case you are willing to send my email [paveldudn@gmail.com](mailto:paveldudn@gmail.com). I will do my best to figure out how to make your modules work but usually, these things take time, so in case you want a faster solution you can use any other ESP module (I would recommend ESP32 but ESP89266 also will work). For example with one of my ACs I am using M5Stamp C3 https://shop.m5stack.com/collections/m5-controllers/products/m5stamp-c3-mate-with-pin-headers (to have the original module for experiments) and it works perfectly well. But of course, it is up to you.

No problem, I've sent you an email with the backup file. Thanks for your help!

paveldn, 2023-03-29:
Hi @Napalmcold 
I think there is no way to update your board, sorry. There is a thing called secure boot: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html if it is enabled bootloader will load only software with the right digest. And to have digest you need a private key (which is hidden somewhere in haier build system). So it looks like there is no way to use this new generation of Haier modules. But you still can use any other ESP32 (or ESP8266) with AC. Also, the good news is that I extracted some data from your image from an unencrypted partition and there is a JSON file with the description of protocol structures.  It looks like hOn protocol should work for you.

Napalmcold, 2023-03-29:
> Hi @Napalmcold I think there is no way to update your board, sorry. There is a thing called secure boot: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html if [...]

Oh well, sorry to hear that. But anyways, thanks a lot for your involvement. At least you helped me to find out which protocol to use :)
I know that I can use any other esp, but I really didn't want to cut the stock wire, connect microusb cable to it and do all that stuff. That's why I was hoping I would be able to flash stock Haier module. Tried to find similar connector like in haier modules, but no luck. Is it proprietary?
And by the way, is there a chance to find out the commands to control eco sensor's mode and other some special HVAC functions, that are available through the haier app?

paveldn, 2023-03-30:
The connector there is JST SM04B-GHS-TB it is not easy to find but possible. And it is not easy to solder either.  
And about special functions: I have plans to implement all functions available through the native app. It just takes some time.

Napalmcold, 2023-03-30:
> The connector there is JST SM04B-GHS-TB

Thank you. Already found it out last night.

> And about special functions: I have plans to implement all functions available through the native app. It just takes some time.

Hope you'll figure that out)

paveldn, 2023-03-30:
Some ESP32 have a way to use USB as UART but not all of them. I think the easiest way is just to connect wires to GPIO

closed 2023-03-31 by paveldn