paveldn/haier-esphome #11: Any way to report CLIMATE_ACTION ?

bearpawmaxim, 2023-04-15:
Hi! Is there any way to report the current action of the climate? I mean I'm interested in getting to know whether it is idle now or heating/cooling? (compressor and electric_heating flags in protocol?)

paveldn, 2023-04-16:
Hi @bearpawmaxim, I thought about this and I don't think there is a way to do it. For hOn, we have a protocol but the only parameter I can use for this is an electricHeatingStatus (which is on or off) so if it is ON it is probably doing heating. But nothing for the cooling. So I am afraid there is no explicit way. Potentially there are such things as compressorStatus, expansionValveOpenDegree, fourWayValveStatus, etc. But I have no idea what it is and how can I use them to detect the state, beside that in my case, it is always 0. So probably they just placeholders for future models.

bearpawmaxim, 2023-04-23:
Okay, thanks!

closed 2023-04-23 by bearpawmaxim