paveldn/haier-esphome #12: Error on dev bracnh

tomassj9, 2023-04-24:
Hello, I've used this since quite ago, it did work all perfect. Today i've tried to do an update to the dev branch, but this is what appears on the ESPHome logs and I don't know if anything it's broken or it's a fault on my side.

Every thing you can need (tests, logs, ...) tell me and I will provide it. Thanks!

ESP Home Log
```
[13:31:28][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:31:28][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:31:28][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:31:29][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:31:29][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:31:38][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:31:38][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:31:39][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:31:39][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:31:39][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:31:43][D][text_sensor:064]: 'Aire Tom cleaning status': Sending state 'No cleaning'
[13:31:48][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:31:48][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:31:49][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:31:49][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:31:49][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:31:58][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:31:58][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:31:59][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:31:59][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:31:59][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:32:03][D][api:102]: Accepted 192.168.27.3
[13:32:03][W][api.connection:089]: 192.168.27.3: Reading failed: BAD_INDICATOR errno=11
[13:32:08][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:32:08][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:09][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:32:09][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:09][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:32:18][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:32:18][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:19][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:32:19][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:19][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:32:25][D][sensor:109]: 'Uptime': Sending state 1.76387 minutes with 0 decimals of accuracy
[13:32:28][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:32:28][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:29][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:32:29][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:29][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:32:38][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:32:38][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:39][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:32:39][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:39][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:32:43][D][text_sensor:064]: 'Aire Tom cleaning status': Sending state 'No cleaning'
[13:32:48][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:32:48][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:49][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:32:49][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:49][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:32:52][D][sensor:109]: 'Wifi Signal': Sending state -51.00000 dBm with 0 decimals of accuracy
[13:32:58][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:32:58][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:59][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:32:59][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:32:59][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:33:06][D][api:102]: Accepted 192.168.27.3
[13:33:06][W][api.connection:089]: 192.168.27.3: Reading failed: BAD_INDICATOR errno=11
[13:33:08][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:33:08][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:33:09][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:33:09][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:33:09][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:33:18][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[... 29 lines cut ...]
[13:34:09][W][haier.protocol:013]: Message handler error, msg=62, err=1
[13:34:09][D][api:102]: Accepted 192.168.27.3
[13:34:09][W][api.connection:089]: 192.168.27.3: Reading failed: BAD_INDICATOR errno=11
[13:34:18][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[13:34:19][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:34:19][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:34:19][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[... cut ...]
[13:34:49][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:34:49][W][haier.climate:175]: Answer timeout for command 61, phase WAITING_ANSWER_INIT_1
[13:34:49][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[13:34:49][W][haier.protocol:013]: Message handler error, msg=62, err=1
```

paveldn, 2023-04-25:
Hi @tomassj9,
Your logs look really strange. ESP is sending a command and gets an answer (and the answer is correct) from AC but for some reason does not consider it as an answer but as a command from AC. 
Probably your AC is too slow and timeout happens or ESP is too busy with something and not able to process data in time which also causes timeout. Have you tried to restart ESP and AC? Try just completely powercut by switching the AC from power. 
I can try to increase timeouts (or even make it configurable) and see if it helps but I am traveling now and will be back next Sunday so cannot do it sooner.

tomassj9, 2023-04-25:
Seems like rebooting ESP and AC didn't work. However, I deleted the .esphome folder, redownloaded the dev branch and it worked somehow. I don't know exactly what happened. Anyway, thanks for the help. Keep going on the project, it's very useful (I have 3 Haier AC's)

closed 2023-04-25 by tomassj9