paveldn/haier-esphome #16: How to install.

Oxmaster, 2023-06-27:
Pardon my ignorance, but I have no clue how to flash/config this thing. I'm using ESPHome for the first time.

I can't find any information about using this. In read me it just says to flash ESPHome. I can't see any .bin file in this repository or anything that could be flashed. I tried the configurations from tests/esp8266-hon-wifi.yaml and wemos_d1_mini.yaml, but I'm getting errors. 

So far I've flashed the ESPHome firmware on my D1 mini via HomeAssistant and tried editing the yaml config of the module, but I'm getting error `Invalid YAML syntax: Error reading file /config/esphome/./.wifi-base.yaml: [Errno 2] No such file or directory: '/config/esphome/./.wifi-base.yaml'`. I've combined basic configuration that came with ESPHome, esp8266-hon-wifi.yaml, wemos_d1_mini.yaml and sample configuration from readme

```
substitutions:
  device_name: Haier AC
  device_id: haier_climate
  uart_id: ac_port 
  send_wifi: "true"

esphome:
  name: esphome-web-9ccdac
  friendly_name: Haier AC

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "[hidden]"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-9Ccdac"
    password: "Wx352F0mVwL4"

captive_portal:
    
uart:
  baud_rate: 9600
  tx_pin: 1
  rx_pin: 3
  id: ${uart_id} 

external_components:
  source:
    type: local
    path: ../components

<<: !include ./.wifi-base.yaml

<<: !include ../.haier-hon-base.yaml

climate:
  - platform: haier
    id: haier_ac
    protocol: hOn
    name: Haier AC 
    uart_id: ${uart_id}
    wifi_signal: true           # Optional, default true, enables WiFI signal transmission from ESP to AC
    beeper: true                # Optional, default true, disables beep on commands from ESP
    outdoor_temperature:        # Optional, outdoor temperature sensor
      name: Haier AC outdoor temperature
    visual:                     # Optional, you can use it to limit min and max temperatures in UI (not working for remote!)
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:            # Optional, can be used to disable some modes if you don't need them
    - 'OFF'
    - AUTO
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    supported_swing_modes:      # Optional, can be used to disable some swing modes if your AC does not support it
    - 'OFF'
    - VERTICAL
    - HORIZONTAL
```

paveldn, 2023-06-29:
Hi @Oxmaster,

First of all get rid of both <<: !include sections. .wi-fi-base.yaml is not needed at all (I am using it with GitHub action tests for fake WiFI). And second one is pretty much covered by the climate section (unless you want to add some additional controls but you can do it later). Also, you need to set logging to UART1 (in the opposite case you will have a conflict with the AC port) like this:
```
logger:
  hardware_uart: UART1
```
Except for these remarks,  your config looks good. Try to fix them and give them another try.

Oxmaster, 2023-06-29:
Yeah, I tried that. I've started to remove all the lines that gave out errors, but it still won't work. Without the includes, I'm getting

```Failed config

external_components: [source /config/esphome/esphome-web-9ccdac.yaml:45]
  source: 
    type: local
    
    Could not find directory '/config/esphome/../components'. Please make sure it exists (full path: /config/components).
    path: ../components
```

and after removing external components config

```Failed config

climate.haier: [source /config/esphome/esphome-web-9ccdac.yaml:46]
  platform: haier
  id: haier_ac
  
  [protocol] is an invalid option for [climate.haier]. Please check the indentation.
  protocol: hOn
  name: Haier AC
  uart_id: ac_port
  
  [wifi_signal] is an invalid option for [climate.haier]. Please check the indentation.
  wifi_signal: True
  
  [beeper] is an invalid option for [climate.haier]. Please check the indentation.
  beeper: True
  
  [outdoor_temperature] is an invalid option for [climate.haier]. Please check the indentation.
  outdoor_temperature: 
    name: Haier AC outdoor temperature
  visual: 
    min_temperature: 16 °C
    max_temperature: 30 °C
    temperature_step: 1 °C
  
  [supported_modes] is an invalid option for [climate.haier]. Did you mean [supported_swing_modes]?
  supported_modes: 
    - OFF
    - AUTO
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
  supported_swing_modes: 
    
    Unknown value 'OFF', valid options are 'BOTH', 'VERTICAL', 'HORIZONTAL'.
    - OFF
    - VERTICAL
    - HORIZONTAL
```

it thinks it's the "climate.haier" without the hOn additions, the protocol field of climate is invalid.

after adding one more tab next to id, I'm getting `mapping values are not allowed here`

paveldn, 2023-06-30:
@Oxmaster 
You still need to have the component code. Replace the whole external_components section with this:
```
external_components:
  - source: github://esphome/esphome@dev
    components: [ haier ]
```
It should fix most of the problems. What is happening now: ESPHome can't find sources for my hair climate component and it takes the default component which is don't support all of the features of the standard one.

Oxmaster, 2023-06-30:
thank you. It seems to work now. Sorry for the trouble.
[image]

closed 2023-06-30 by Oxmaster