paveldn/haier-esphome #19: Problem with AD71S2SM3FA(H)

knathan1987, 2023-07-08:
Hello,

I can not get my ac to work with the module. When I want to set a parameter I always get the message "Answer handler error, msg=01, answ=03, err=7".

> Jul  8 23:23:35 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 61, data: 00 07
> Jul  8 23:23:35 haier haier.protocol [D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 30 30 30 30 30 30 34 32 F1 00 00 30 30 30 30 30 30 30 31 01 55 2D 42 41 43 00 00 00 00 04
> Jul  8 23:23:37 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 70, data: <empty>
> Jul  8 23:23:37 haier haier.protocol [D][haier.protocol:019]: Frame found: type 71, data: 20 1C 10 C7 08 80 81 00 0D 12 05 46 45 44 85 00 00 00 9C D6 8E 69 2C 10 4E 2A 33 3E AB 95 D1 40
> Jul  8 23:23:39 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
> Jul  8 23:23:39 haier haier.protocol [D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 00 23 00 02 01 14 00 00 00 00 00 01 00 00 03 09 03 30 32 58 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> Jul  8 23:23:39 haier sensor [D][sensor:094]: 'Haier AC outdoor temperature': Sending state -63.00000 °C with 0 decimals of accuracy
> Jul  8 23:23:39 haier climate [D][climate:378]: 'Haier AC' - Sending state:
> Jul  8 23:23:39 haier climate [D][climate:381]:   Mode: COOL
> Jul  8 23:23:39 haier climate [D][climate:386]:   Fan Mode: LOW
> Jul  8 23:23:39 haier climate [D][climate:392]:   Preset: NONE
> Jul  8 23:23:39 haier climate [D][climate:398]:   Swing Mode: OFF
> Jul  8 23:23:39 haier climate [D][climate:401]:   Current Temperature: 0.00°C
> Jul  8 23:23:39 haier climate [D][climate:407]:   Target Temperature: 16.00°C
> Jul  8 23:23:39 haier haier.climate [I][haier.climate:824]: HVAC values changed
> Jul  8 23:23:39 haier haier.climate [I][haier.climate:826]: HVAC Mode = 0x1
> Jul  8 23:23:39 haier haier.climate [I][haier.climate:828]: Fan speed Status = 0x3
> Jul  8 23:23:39 haier haier.climate [I][haier.climate:830]: Horizontal Swing Status = 0x0
> Jul  8 23:23:39 haier haier.climate [I][haier.climate:832]: Vertical Swing Status = 0x0
> Jul  8 23:23:39 haier haier.climate [I][haier.climate:834]: Set Point Status = 0x0
> Jul  8 23:23:39 haier haier.climate [I][haier.climate:206]: First HVAC status received
> Jul  8 23:23:41 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 73, data: <empty>
> Jul  8 23:23:41 haier haier.protocol [D][haier.protocol:019]: Frame found: type 74, data: 0F 5A 00 00 00 00 00 00 00 00 00 00
> Jul  8 23:23:41 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:41 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:41 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> Jul  8 23:23:41 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> Jul  8 23:23:42 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:42 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:42 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> Jul  8 23:23:42 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> Jul  8 23:23:42 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:42 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:42 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> Jul  8 23:23:42 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> Jul  8 23:23:42 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> Jul  8 23:23:43 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> Jul  8 23:23:43 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> Jul  8 23:23:43 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> Jul  8 23:23:43 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> Jul  8 23:23:43 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> Jul  8 23:23:44 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:44 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:44 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> Jul  8 23:23:44 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> Jul  8 23:23:44 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:44 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:44 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> Jul  8 23:23:44 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> Jul  8 23:23:45 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:45 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:45 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> Jul  8 23:23:45 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> Jul  8 23:23:45 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:45 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:45 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> Jul  8 23:23:45 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> Jul  8 23:23:45 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:45 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:45 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
[... 21 lines cut ...]
> Jul  8 23:23:48 haier haier.climate [I][haier.climate:418]: Control packet sent
> Jul  8 23:23:48 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> Jul  8 23:23:48 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> Jul  8 23:23:48 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> Jul  8 23:23:48 haier haier.climate [W][haier.climate:408]: Sending control packet timeout!
> Jul  8 23:23:50 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
> Jul  8 23:23:50 haier haier.protocol [D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 00 23 00 02 01 14 00 00 00 00 00 01 00 00 03 09 03 30 32 58 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> Jul  8 23:23:50 haier climate [D][climate:378]: 'Haier AC' - Sending state:
> Jul  8 23:23:50 haier climate [D][climate:381]:   Mode: COOL
> Jul  8 23:23:50 haier climate [D][climate:386]:   Fan Mode: LOW
> Jul  8 23:23:50 haier climate [D][climate:392]:   Preset: NONE
> Jul  8 23:23:50 haier climate [D][climate:398]:   Swing Mode: OFF
> Jul  8 23:23:50 haier climate [D][climate:401]:   Current Temperature: 0.00°C
> Jul  8 23:23:50 haier climate [D][climate:407]:   Target Temperature: 16.00°C
> Jul  8 23:23:50 haier haier.climate [D][haier.climate:826]: HVAC Mode = 0x1
[... cut ...]
> Jul  8 23:23:52 haier haier.protocol [D][haier.protocol:019]: Sending frame: type FC, data: <empty>
> Jul  8 23:23:52 haier haier.protocol [D][haier.protocol:019]: Frame found: type FD, data: 00 00 00 01 00 00
> Jul  8 23:23:54 haier haier.climate [D][haier.climate:371]: WiFi signal is: -38dBm => 70%
> Jul  8 23:23:54 haier haier.protocol [D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 46
> Jul  8 23:23:54 haier haier.protocol [D][haier.protocol:019]: Frame found: type 05, data: <empty>

paveldn, 2023-07-09:
Hi @knathan1987 
Your logs look strange. It looks like hOn but some packets are bigger. And some values shifted. It is not clear to me yet whether is this some kind of problem related to UART port settings or a new version of the hOn protocol.
To fix this I will need your help. 
Do you have the original WiFi module for this AC?
I will need a log from you with setting different parameters (like destination temperature, AC mode, and swing mode for AC) from the remote with comments on what values you set. Also, make sure that there are at least 20 seconds between your setting parameters. It can be separate logs for each action.

knathan1987, 2023-07-10:
Hi @paveldn

I'm using a "Wemos D1 Mini ESP32"-Module. TX-Pin: 19, RX-Pin: 18, Baudrate: 9600.

> Do you have the original WiFi module for this AC?

Yes. But I could not get it to connect to my Wi-Fi. 
It looks exactly like the hOn module displayed on the "readme"-Page. (Maybe a different revision number - have to check that 😅)

What exactly do you need? The whole logs? Or is something like that sufficient?

// OFF
> Frame found: type 02, data: 6D 01 00 00 21 00 02 00 14 00 00 00 00 00 01 00 00 03 09 03 3A 32 66 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

// Fan only, Fan speed: Low
> Frame found: type 02, data: 6D 01 08 00 C3 00 02 01 14 00 00 00 00 00 01 00 00 03 09 03 3A 32 57 00 00 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

// Fan only, Fan speed: Medium
> Frame found: type 02, data: 6D 01 08 00 C2 00 02 01 14 00 00 00 00 00 01 00 00 03 09 03 3A 32 57 00 00 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

// Fan only, Fan speed: High 
> Frame found: type 02, data: 6D 01 08 00 C1 00 02 01 14 00 00 00 00 00 01 00 00 03 09 03 3A 32 57 00 00 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

(im going to log some more settings later 😉)

paveldn, 2023-07-11:
I am afraid it wouldn't be so easy. In general, the situation is the following: Usually, the hOn packet with status contains 2 parts first one - 10 bytes control group with different AC settings, and second - 17 byes sensors group with different sensors and counters.
When you change some settings you need to send only the control group back. But in your case - the control group has extra 6 bytes. I have no idea why and what these extra 6 bytes contain. I will try to simulate your AC for my ESP32 haier module and see how it goes. It will take some time. Thnks for the logs. It will also help if you will provide comments about all parameters that you can get from the remote like target temp, room temperature, swing mode, etc. Will help me to detect packet structure.
It looks like for your AC I will need to make separate protocol type something like hon_v2. Will see.

knathan1987, 2023-07-12:
Sooo.... Here are some more log entries.
Since it's a duct Unit it does not support any "Swing Modes".

Off, Room Temperature: 30 °C, Room Humidity: 50 %
> 6D 01 07 00 21 00 02 08 14 00 00 00 00 00 01 00 00 03 09 03 3E 32 5D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Mode: Cooling, Room Temperature: 30 °C, Target Temperature: 16.0°C, Fan speed: Auto
> 6D 01 00 00 25 00 02 01 14 00 00 00 00 00 01 00 00 03 09 03 3C 32 5E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Mode: Cooling, Room Temperature: 30 °C, Target Temperature: 17.0°C, Fan speed: Auto
> 6D 01 01 00 25 00 02 01 14 00 00 00 00 00 01 00 00 03 09 03 3C 32 5E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Mode: Cooling, Room Temperature: 30 °C, Target Temperature: 16.0°C, Fan speed: Low
> 6D 01 00 00 23 00 02 01 14 00 00 00 00 00 01 00 00 03 09 03 3C 32 5F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Mode: Cooling, Health Mode ON, Room Temperature: 30 °C, Target Temperature: 16.0°C, Fan speed: Low
> 6D 01 00 00 23 00 02 03 14 00 00 00 00 00 01 00 00 03 09 03 3C 32 5E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Mode: Dry, Health Mode ON, Dehydrate, Room Temperature: 30 °C, Target Temperature: 24.0°C, Fan speed: Auto
> 6D 01 08 02 45 00 02 03 14 00 00 00 00 00 01 00 00 03 09 03 3C 32 5E 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Mode: Dry, Health Mode OFF, Dehydrate, Room Temperature: 30 °C, Target Temperature: 24.0°C, Fan speed: Auto
> 6D 01 08 00 45 00 02 01 14 00 00 00 00 00 01 00 00 03 09 03 3C 32 5E 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Mode: Cooling, ECO MODE, Room Temperature: 31 °C, Target Temperature: 23.0°C, Fan speed: Low
> 6D 01 07 00 23 00 02 01 14 00 00 00 00 00 01 00 00 03 09 03 3E 32 5E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Mode: Cooling, TURBO MODE, Room Temperature: 31 °C, Target Temperature: 23.0°C, Fan speed: Turbo (same as "high"?!)
> 6D 01 07 00 21 00 02 09 14 00 00 00 00 00 01 00 00 03 09 03 3E 32 5E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

scaraps, 2023-07-14:
Hi, I have the same machine as you and I use it with smartair2, can you tell me which configurations you used, because I can't connect it in any way? can you pass me your yaml?

> Hello,
> 
> I can not get my ac to work with the module. When I want to set a parameter I always get the message "Answer handler error, msg=01, answ=03, err=7".
> 
> > Jul  8 23:23:35 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 61, data: 00 07
> > Jul  8 23:23:35 haier haier.protocol [D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 30 30 30 30 30 30 34 32 F1 00 00 30 30 30 30 30 30 30 31 01 55 2D 42 41 43 00 00 00 00 04
> > Jul  8 23:23:37 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 70, data: 
> > Jul  8 23:23:37 haier haier.protocol [D][haier.protocol:019]: Frame found: type 71, data: 20 1C 10 C7 08 80 81 00 0D 12 05 46 45 44 85 00 00 00 9C D6 8E 69 2C 10 4E 2A 33 3E AB 95 D1 40
> > Jul  8 23:23:39 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
> > Jul  8 23:23:39 haier haier.protocol [D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 00 23 00 02 01 14 00 00 00 00 00 01 00 00 03 09 03 30 32 58 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> > Jul  8 23:23:39 haier sensor [D][sensor:094]: 'Haier AC outdoor temperature': Sending state -63.00000 °C with 0 decimals of accuracy
> > Jul  8 23:23:39 haier climate [D][climate:378]: 'Haier AC' - Sending state:
> > Jul  8 23:23:39 haier climate [D][climate:381]:   Mode: COOL
> > Jul  8 23:23:39 haier climate [D][climate:386]:   Fan Mode: LOW
> > Jul  8 23:23:39 haier climate [D][climate:392]:   Preset: NONE
> > Jul  8 23:23:39 haier climate [D][climate:398]:   Swing Mode: OFF
> > Jul  8 23:23:39 haier climate [D][climate:401]:   Current Temperature: 0.00°C
> > Jul  8 23:23:39 haier climate [D][climate:407]:   Target Temperature: 16.00°C
> > Jul  8 23:23:39 haier haier.climate [I][haier.climate:824]: HVAC values changed
> > Jul  8 23:23:39 haier haier.climate [I][haier.climate:826]: HVAC Mode = 0x1
> > Jul  8 23:23:39 haier haier.climate [I][haier.climate:828]: Fan speed Status = 0x3
> > Jul  8 23:23:39 haier haier.climate [I][haier.climate:830]: Horizontal Swing Status = 0x0
> > Jul  8 23:23:39 haier haier.climate [I][haier.climate:832]: Vertical Swing Status = 0x0
> > Jul  8 23:23:39 haier haier.climate [I][haier.climate:834]: Set Point Status = 0x0
> > Jul  8 23:23:39 haier haier.climate [I][haier.climate:206]: First HVAC status received
> > Jul  8 23:23:41 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 73, data: 
> > Jul  8 23:23:41 haier haier.protocol [D][haier.protocol:019]: Frame found: type 74, data: 0F 5A 00 00 00 00 00 00 00 00 00 00
> > Jul  8 23:23:41 haier haier.climate [I][haier.climate:418]: Control packet sent
> > Jul  8 23:23:41 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> > Jul  8 23:23:41 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> > Jul  8 23:23:41 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> > Jul  8 23:23:42 haier haier.climate [I][haier.climate:418]: Control packet sent
> > Jul  8 23:23:42 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> > Jul  8 23:23:42 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> > Jul  8 23:23:42 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> > Jul  8 23:23:42 haier haier.climate [I][haier.climate:418]: Control packet sent
> > Jul  8 23:23:42 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> > Jul  8 23:23:42 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> > Jul  8 23:23:42 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> > Jul  8 23:23:42 haier haier.climate [I][haier.climate:418]: Control packet sent
> > Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> > Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> > Jul  8 23:23:43 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> > Jul  8 23:23:43 haier haier.climate [I][haier.climate:418]: Control packet sent
> > Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> > Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> > Jul  8 23:23:43 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> > Jul  8 23:23:43 haier haier.climate [I][haier.climate:418]: Control packet sent
> > Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 00 06 23 00 00 81 14 00 00 00
> > Jul  8 23:23:43 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
> > Jul  8 23:23:43 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
> > Jul  8 23:23:44 haier haier.climate [I][haier.climate:418]: Control packet sent
[... cut ...]
> > Jul  8 23:23:52 haier haier.protocol [D][haier.protocol:019]: Sending frame: type FC, data: 
> > Jul  8 23:23:52 haier haier.protocol [D][haier.protocol:019]: Frame found: type FD, data: 00 00 00 01 00 00
> > Jul  8 23:23:54 haier haier.climate [D][haier.climate:371]: WiFi signal is: -38dBm => 70%
> > Jul  8 23:23:54 haier haier.protocol [D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 46
> > Jul  8 23:23:54 haier haier.protocol [D][haier.protocol:019]: Frame found: type 05, data:

OwlBawl, 2023-07-15:
It should be an SmartAir2 in-ceiling AC, as I have similar AD35S2SS1FA and it uses older WiFi dongle and defenetly has some protocol differences from standart SmartAir2. But works basically with @paveldn integration. Did you try 
"protocol: smartAir2" ?

OwlBawl, 2023-07-15:
my .yaml for similar model AD35S2SS1FA if it helps.

`substitutions:
  device_name: "haier-guestroom"
  device_id: haier_guestroom
  uart_id: ac_guestroom
  send_wifi: "false"

esphome:
  name: ${device_name}

esp8266:
    board: d1_mini

logger:
  level: DEBUG
  baud_rate: 0 #Important. You can't use serial port
  hardware_uart: UART1
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  #important, otherwise the IP could change and then the entity is unavailable from home assistant.
  manual_ip:
    static_ip: 0.0.0.0
    gateway: 0.0.0.0.
    subnet: 255.255.255.0

external_components:
    - source: github://paveldn/haier-esphome@experimental
      components: [ haier ]
      refresh: 1d

# Enable Home Assistant API
api:

ota:

uart:
  baud_rate: 9600
  tx_pin: 1
  rx_pin: 3
  id: ${uart_id}

climate:
  - platform: haier
    id: ${device_id}
    name: ${device_name}
    protocol: smartAir2
    wifi_signal: ${send_wifi}
    uart_id: ${uart_id}
    supported_modes:            # Optional, can be used to disable some modes if you don't need them
      - 'OFF'
      - AUTO
      - COOL
      - HEAT
      - DRY
      - FAN_ONLY
    supported_swing_modes: []
    visual:                     # Optional, you can use it to limit min and max temperatures in UI (not working for remote!)
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 0.5 °C`

knathan1987, 2023-07-16:
> It should be an SmartAir2 in-ceiling AC, as I have similar AD35S2SS1FA and it uses older WiFi dongle and defenetly has some protocol differences from standart SmartAir2. But works basically with @paveldn integration. Did you try
> "protocol: smartAir2" ?

Tried that. Did not work at all.
As I said, my AC had the newer "ESP32-for-Haier" module pre-installed from the factory.

paveldn, 2023-07-16:
@knathan1987 
I have a feeling that haier have both smartair2 and hOn control modules compatible with any AC and installs different modules to the same AC model depending on region (or maybe some other reasons). I already saw the same models of AC working with smartair2 and with hOn.
I tried to make a simulator based on your logs. It really helped me with figuring out how smartair2 works. But in your case my esp32-for-haier reboots :( It looks like your esp32 module has a different firmware version. So probably we will need to go a long way. Try and error. I will make the first test version just with basic functionality and you will need to try it. 
BTW it looks like your AC has a way to set target humidity for dry mode. Is that so? Can you see it on the remote?

knathan1987, 2023-07-16:
> It looks like your esp32 module has a different firmware version.

Maybe i could extract the Firmware from my Module... Dont know if you could flash that on yours 🤷🏻‍♂️

As i said i wasnt able to connect the Module to my WiFi.
I could try to use a different AccessPoint and try it again. If that works i could sniff the UART packages. I think that would help, right? 😊

> BTW it looks like your AC has a way to set target humidity for dry mode. Is that so? Can you see it on the remote?

Yes. Will try to log some settings.

But i may need some time to do than since im a little "handicapped" right now (fractured metacarpal bone) 😔

paveldn, 2023-07-16:
@knathan1987,
Can you try this branch https://github.com/paveldn/haier-esphome/tree/bigger_control_packet ? Let me know if it worked.

sergey909, 2023-07-17:
@paveldn 
I tried it on a new branch, received these messages -
[08:32:34][D][haier.protocol:019]: Frame found: type 03, data: 00 00
[08:32:34][W][haier.climate:126]: It looks like your ESPHome Haier climate configuration is wrong. You should use the smartAir2 protocol instead of hOn
[08:32:34][W][haier.protocol:013]: Answer handler error, msg=61, answ=03, err=7
ok, switched the protocol to smartAir2, had to turn it off cleaning status, beeper, outdoor_temperature, airflow vertical, airflow horizontal and self_cleaning.
Looks like it's working -
[08:55:45][C][haier.climate:228]: Haier Climate 'haier-ac-01 Bedroom'
[08:55:45][C][haier.climate:229]:   Device communication status: none
[08:55:45][C][haier.climate:130]:   Protocol version: smartAir2
[08:55:45][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1A 00 3B 00 00 00 00 00 00 00 01 00 03 00 00 00 00 00 00 00 00 00 07
[08:55:45][D][climate:378]: 'haier-ac-01 Bedroom' - Sending state:
[08:55:45][D][climate:381]:   Mode: OFF
[08:55:45][D][climate:386]:   Fan Mode: AUTO
[08:55:45][D][climate:392]:   Preset: NONE
[08:55:45][D][climate:398]:   Swing Mode: OFF
[08:55:45][D][climate:401]:   Current Temperature: 26.00°C
[08:55:45][D][climate:407]:   Target Temperature: 23.00°C
[08:55:45][I][haier.climate:531]: HVAC values changed
[08:55:45][I][haier.climate:533]: HVAC Mode = 0x1
[08:55:45][I][haier.climate:535]: Fan speed Status = 0x3
[08:55:45][I][haier.climate:537]: Horizontal Swing Status = 0x0
[08:55:45][I][haier.climate:539]: Vertical Swing Status = 0x0
[08:55:45][I][haier.climate:541]: Set Point Status = 0x7
[08:55:45][I][haier.climate:041]: First HVAC status received
[08:55:45][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1A 00 3B 00 00 00 00 00 00 00 01 00 03 00 00 00 00 00 20 00 00 00 07
[08:55:45][D][haier.climate:533]: HVAC Mode = 0x1
[08:55:45][D][haier.climate:535]: Fan speed Status = 0x3
[08:55:45][D][haier.climate:537]: Horizontal Swing Status = 0x0
[08:55:45][D][haier.climate:539]: Vertical Swing Status = 0x0
[08:55:45][D][haier.climate:541]: Set Point Status = 0x7
[08:55:47][D][haier.climate:128]: WiFi signal is: -34dBm => 73%
[08:55:47][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 49
[08:55:47][D][haier.protocol:019]: Frame found: type 05, data: <empty>
[08:55:50][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[08:55:50][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1A 00 3B 00 00 00 00 00 00 00 01 00 03 00 00 00 00 00 20 00 00 00 07

knathan1987, 2023-07-17:
Tried it.
This is what i get after restart:

> Jul 17 08:13:58 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 61, data: 00 07
Jul 17 08:13:58 haier haier.protocol [D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 30 30 30 30 30 30 34 32 F1 00 00 30 30 30 30 30 30 30 31 01 55 2D 42 41 43 00 00 00 00 04
Jul 17 08:14:00 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 70, data: <empty>
Jul 17 08:14:00 haier haier.protocol [D][haier.protocol:019]: Frame found: type 71, data: 20 1C 10 C7 08 80 81 00 0D 12 05 46 45 44 85 00 00 00 9C D6 8E 69 2C 10 4E 2A 33 3E AB 95 D1 40
Jul 17 08:14:02 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
Jul 17 08:14:02 haier haier.protocol [D][haier.protocol:019]: Frame found: type 02, data: 6D 01 06 00 23 00 02 00 14 00 00 00 00 00 01 00 00 03 09 03 30 32 5D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Jul 17 08:14:02 haier climate [D][climate:378]: 'Haier AC' - Sending state:
Jul 17 08:14:02 haier climate [D][climate:381]:   Mode: OFF
Jul 17 08:14:02 haier climate [D][climate:386]:   Fan Mode: LOW
Jul 17 08:14:02 haier climate [D][climate:392]:   Preset: NONE
Jul 17 08:14:02 haier climate [D][climate:398]:   Swing Mode: OFF
Jul 17 08:14:02 haier climate [D][climate:401]:   Current Temperature: 24.00°C
Jul 17 08:14:02 haier climate [D][climate:407]:   Target Temperature: 22.00°C
Jul 17 08:14:02 haier haier.climate [I][haier.climate:824]: HVAC values changed
Jul 17 08:14:02 haier haier.climate [I][haier.climate:826]: HVAC Mode = 0x1
Jul 17 08:14:02 haier haier.climate [I][haier.climate:828]: Fan speed Status = 0x3
Jul 17 08:14:02 haier haier.climate [I][haier.climate:830]: Horizontal Swing Status = 0x0
Jul 17 08:14:02 haier haier.climate [I][haier.climate:832]: Vertical Swing Status = 0x0
Jul 17 08:14:02 haier haier.climate [I][haier.climate:834]: Set Point Status = 0x6
Jul 17 08:14:02 haier haier.climate [I][haier.climate:206]: First HVAC status received
Jul 17 08:14:04 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 73, data: <empty>
Jul 17 08:14:04 haier haier.protocol [D][haier.protocol:019]: Frame found: type 74, data: 0F 5A 00 00 00 00 00 00 00 00 00 00
Jul 17 08:14:05 haier haier.climate [I][haier.climate:418]: Control packet sent
Jul 17 08:14:05 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 06 06 23 00 00 80 14 00 00 00 00 00 01 00 00 03 09 03
Jul 17 08:14:05 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
Jul 17 08:14:05 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
Jul 17 08:14:05 haier haier.climate [I][haier.climate:418]: Control packet sent
Jul 17 08:14:05 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 06 06 23 00 00 80 14 00 00 00 00 00 01 00 00 03 09 03
Jul 17 08:14:05 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
Jul 17 08:14:05 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
Jul 17 08:14:05 haier haier.climate [I][haier.climate:418]: Control packet sent
Jul 17 08:14:05 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 06 06 23 00 00 80 14 00 00 00 00 00 01 00 00 03 09 03
Jul 17 08:14:05 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
Jul 17 08:14:05 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
Jul 17 08:14:06 haier haier.climate [I][haier.climate:418]: Control packet sent
Jul 17 08:14:06 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 06 06 23 00 00 80 14 00 00 00 00 00 01 00 00 03 09 03
Jul 17 08:14:06 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
Jul 17 08:14:06 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
Jul 17 08:14:06 haier haier.climate [I][haier.climate:418]: Control packet sent
Jul 17 08:14:06 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 06 06 23 00 00 80 14 00 00 00 00 00 01 00 00 03 09 03
Jul 17 08:14:06 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
Jul 17 08:14:06 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
Jul 17 08:14:07 haier haier.climate [I][haier.climate:418]: Control packet sent
Jul 17 08:14:07 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 06 06 23 00 00 80 14 00 00 00 00 00 01 00 00 03 09 03
Jul 17 08:14:07 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
Jul 17 08:14:07 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
Jul 17 08:14:07 haier haier.climate [I][haier.climate:418]: Control packet sent
Jul 17 08:14:07 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 06 06 23 00 00 80 14 00 00 00 00 00 01 00 00 03 09 03
Jul 17 08:14:07 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
Jul 17 08:14:07 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
Jul 17 08:14:07 haier haier.climate [I][haier.climate:418]: Control packet sent
[... cut ...]
Jul 17 08:14:11 haier haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 60 01 06 06 23 00 00 80 14 00 00 00 00 00 01 00 00 03 09 03
Jul 17 08:14:11 haier haier.protocol [D][haier.protocol:019]: Frame found: type 03, data: 00 01
Jul 17 08:14:11 haier haier.protocol [W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
Jul 17 08:14:11 haier haier.climate [W][haier.climate:408]: Sending control packet timeout!

The only difference i see: Current Temperature and Outdoor Temperature are now displayed (outdoor Temp seems a little bit too high 🤷🏻‍♂️).
But i still can not send any commands.

sergey909, 2023-07-17:
[09:39:38][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:39:38][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1A 00 38 00 05 00 00 00 00 00 01 00 02 00 00 00 01 00 20 00 00 00 07
[09:39:38][D][haier.climate:533]: HVAC Mode = 0x1
[09:39:38][D][haier.climate:535]: Fan speed Status = 0x2
[09:39:38][D][haier.climate:537]: Horizontal Swing Status = 0x0
[09:39:38][D][haier.climate:539]: Vertical Swing Status = 0x0
[09:39:38][D][haier.climate:541]: Set Point Status = 0x7
[09:39:43][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:39:43][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 19 00 38 00 05 00 00 00 00 00 01 00 02 00 00 00 01 00 20 00 00 00 07
[09:39:43][D][climate:378]: 'haier-ac-01 Bedroom' - Sending state:
[09:39:43][D][climate:381]:   Mode: COOL
[09:39:43][D][climate:386]:   Fan Mode: LOW
[09:39:43][D][climate:392]:   Preset: NONE
[09:39:43][D][climate:398]:   Swing Mode: OFF
[09:39:43][D][climate:401]:   Current Temperature: 25.00°C
[09:39:43][D][climate:407]:   Target Temperature: 23.00°C
[09:39:43][I][haier.climate:531]: HVAC values changed
[09:39:43][I][haier.climate:533]: HVAC Mode = 0x1
[09:39:43][I][haier.climate:535]: Fan speed Status = 0x2
[09:39:43][I][haier.climate:537]: Horizontal Swing Status = 0x0
[09:39:43][I][haier.climate:539]: Vertical Swing Status = 0x0
[09:39:43][I][haier.climate:541]: Set Point Status = 0x7
[09:39:45][D][haier.climate:128]: WiFi signal is: -34dBm => 73%
[09:39:45][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 49
[09:39:45][D][haier.protocol:019]: Frame found: type 05, data: <empty>
[09:39:48][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:39:48][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 19 00 38 00 05 00 00 00 00 00 01 00 02 00 00 00 01 00 20 00 00 00 07
[09:39:48][D][haier.climate:533]: HVAC Mode = 0x1
[09:39:48][D][haier.climate:535]: Fan speed Status = 0x2
[09:39:48][D][haier.climate:537]: Horizontal Swing Status = 0x0
[09:39:48][D][haier.climate:539]: Vertical Swing Status = 0x0
[09:39:48][D][haier.climate:541]: Set Point Status = 0x7
[09:39:53][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:39:53][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 19 00 38 00 05 00 00 00 00 00 01 00 02 00 00 00 01 00 20 00 00 00 07
[09:39:53][D][haier.climate:533]: HVAC Mode = 0x1
---
09:40:27][D][climate:010]: 'haier-ac-01 Bedroom' - Setting
[09:40:27][D][climate:014]:   Mode: OFF
[09:40:27][D][Control:300]: Control call
[09:40:27][I][haier.climate:208]: Control packet sent
[09:40:27][D][haier.protocol:019]: Sending frame: type 01, data: 4D 5F 00 19 00 38 00 00 00 00 00 00 00 01 00 02 00 00 00 00 00 20 00 00 00 07
[09:40:28][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 19 00 38 00 00 00 00 00 00 00 01 00 02 00 00 00 00 00 20 00 00 00 07
[09:40:28][D][climate:378]: 'haier-ac-01 Bedroom' - Sending state:
[09:40:28][D][climate:381]:   Mode: OFF
[09:40:28][D][climate:386]:   Fan Mode: LOW
[09:40:28][D][climate:392]:   Preset: NONE
[09:40:28][D][climate:398]:   Swing Mode: OFF
[09:40:28][D][climate:401]:   Current Temperature: 25.00°C
[09:40:28][D][climate:407]:   Target Temperature: 23.00°C
[09:40:28][I][haier.climate:531]: HVAC values changed
[09:40:28][I][haier.climate:533]: HVAC Mode = 0x1
[09:40:28][I][haier.climate:535]: Fan speed Status = 0x2
[09:40:28][I][haier.climate:537]: Horizontal Swing Status = 0x0
[09:40:28][I][haier.climate:539]: Vertical Swing Status = 0x0
[09:40:28][I][haier.climate:541]: Set Point Status = 0x7
[09:40:29][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:40:30][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 19 00 39 00 00 00 00 00 00 00 01 00 02 00 00 00 00 00 20 00 00 00 07
[09:40:30][D][haier.climate:533]: HVAC Mode = 0x1
[09:40:30][D][haier.climate:535]: Fan speed Status = 0x2
[09:40:30][D][haier.climate:537]: Horizontal Swing Status = 0x0
[09:40:30][D][haier.climate:539]: Vertical Swing Status = 0x0
[09:40:30][D][haier.climate:541]: Set Point Status = 0x7
--
Look's like working with config:

external_components:
    source: github://paveldn/haier-esphome@bigger_control_packet
    components:
      - haier
climate:
  - platform: haier
    protocol: smartAir2

knathan1987, 2023-07-17:
This is what i get using "smartAir2".

> Jul 17 08:57:00 haier-sa haier.protocol [D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
Jul 17 08:57:00 haier-sa haier.protocol [D][haier.protocol:019]: Frame found: type 02, data: 6D 01 06 00 23 00 02 00 14 00 00 00 00 00 01 00 00 03 09 03 30 32 5D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Jul 17 08:57:00 haier-sa climate [D][climate:378]: 'Haier AC' - Sending state:
Jul 17 08:57:00 haier-sa climate [D][climate:381]:   Mode: AUTO
Jul 17 08:57:00 haier-sa climate [D][climate:386]:   Fan Mode: HIGH
Jul 17 08:57:00 haier-sa climate [D][climate:392]:   Preset: BOOST
Jul 17 08:57:00 haier-sa climate [D][climate:398]:   Swing Mode: BOTH
Jul 17 08:57:00 haier-sa climate [D][climate:401]:   Current Temperature: 0.00°C
Jul 17 08:57:00 haier-sa climate [D][climate:407]:   Target Temperature: 16.00°C
Jul 17 08:57:00 haier-sa haier.climate [I][haier.climate:437]: HVAC values changed
Jul 17 08:57:00 haier-sa haier.climate [I][haier.climate:439]: HVAC Mode = 0x0
Jul 17 08:57:00 haier-sa haier.climate [I][haier.climate:441]: Fan speed Status = 0x0
Jul 17 08:57:00 haier-sa haier.climate [I][haier.climate:443]: Horizontal Swing Status = 0x0
Jul 17 08:57:00 haier-sa haier.climate [I][haier.climate:445]: Vertical Swing Status = 0x1
Jul 17 08:57:00 haier-sa haier.climate [I][haier.climate:447]: Set Point Status = 0x0
Jul 17 08:57:00 haier-sa haier.climate [I][haier.climate:043]: First HVAC status received

Those values are all wrong 🤷🏻‍♂️

paveldn, 2023-07-17:
@sergey909  It looks like this model has different control modules with different protocols. Your device has smartAir2 protocol but @knathan1987's device has some variation of hOn.

@knathan1987 OK I tried. So let me give you a small summary. Your device talking almost normal hOn with extra 8 bytes (with values 00 00 01 00 00 03 09 03) between the control block and sensors block. Is this a constant value or does it control something - have no idea. In your logs, it is always the same. I tried just to copy them and include them in the control command when we trying to control AC but it didn't work. 
The only way to continue as I see it is to make your original Hier ESP32 device work and to sniff what is happening between that device and your AC. So you will need somehow to get between them. 
Sniffing traffic between ESP32 and Haier cloud is useless because it is a completely different protocol and gives no information about the protocol with AC. What I need to see is what your ESP32 module sends to AC when you change something in the official hOn application.
Making a copy of your FW will not work for me, those devices have unique certificates. If I will flash mine with your firmware it will not talk to the Haier cloud. I can guide you but you will need to do it mostly yourself.

paveldn, 2023-07-17:
@sergey909  And BTW smartAir2 protocol on the new bigger_control_packet branch works the same way as on the experimental branch all changes for hOn only. SO you can switch to experimental.

sergey909, 2023-07-17:
it's just that initially I had the same errors - Answer handler error, msg=01, answ=03, err=7

paveldn, 2023-07-17:
@sergey909 That is by design. It works like this with the official WiFi module. answ=03 means that AC doesn't support the command. I made this to be able to detect in logs if somebody used the wrong configuration (smartAir2 instead of hOn and vice versa). And it worked in your way 😄

sergey909, 2023-07-18:
@paveldn 
Hi, how to reduce log level haier protocol?
By defaut - INFO Detected log level for Haier protocol: DEBUG

knathan1987, 2023-07-18:
> The only way to continue as I see it is to make your original Hier ESP32 device work and to sniff what is happening between that device and your AC. So you will need somehow to get between them.

Thats what i meant in my answer above.... I will do that - but as i said - it will (maybe) take a few days because i cant hold a soldering iron right now 😬

> Making a copy of your FW will not work for me, those devices have unique certificates. If I will flash mine with your firmware it will not talk to the Haier cloud.

Oh okay, didnt know that. Thought they were simply identified by ESP-chipID or something like that.
**If nothing else works** i could send you my Module for research purposes. 🙃

scaraps, 2023-07-21:
> It should be an SmartAir2 in-ceiling AC, as I have similar AD35S2SS1FA and it uses older WiFi dongle and defenetly has some protocol differences from standart SmartAir2. But works basically with @paveldn integration. Did you try "protocol: smartAir2" ?

yes, i tried both with the official, and with your yaml. I am attaching the log, maybe you can help me! Even for a fee, because I'm going crazy and I need it!
[logs_test_run (1).txt](https://github.com/paveldn/haier-esphome/files/12121886/logs_test_run.1.txt)

My yaml:

esphome:
  name: test
  friendly_name: Test
  name_add_mac_suffix: true

esp8266:
  board: esp01_1m

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0
  

# Enable Home Assistant API
api:
  encryption:
    key: "fJI5DzsLJxso1+Gh7OTTpfojaokXIhDeUmM9r+U9HkY="

ota:
  password: "baab0fef81c379b501f53b406bcc6b73"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.90
    gateway: 192.168.1.254
    subnet: 255.255.255.0
  fast_connect: True  

uart:
  baud_rate: 115200
  tx_pin: 1
  rx_pin: 3
  id: acport

external_components:
- source: github://paveldn/haier-esphome@experimental

captive_portal:
    
climate:
  - platform: haier
    id: haier_ac
    protocol: smartAir2
    name: Haier AC 
    uart_id: acport
    visual:                     # Optional, you can use it to limit min and max temperatures in UI (not working for remote!)
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:            # Optional, can be used to disable some modes if you don't need them
    - 'OFF'
    - AUTO
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    supported_swing_modes:      # Optional, can be used to disable some swing modes if your AC does not support it
    - 'OFF'
    - VERTICAL
    - HORIZONTAL
    - BOTH

paveldn, 2023-07-21:
@scaraps 
Hi. All I can see from the logs is that your ESP is not communicating with AC. The first thing to change is the baud rate for the UART port, it  should be:
baud_rate: 9600
If this will not help check your wiring.

scaraps, 2023-07-21:
Hi! @paveldn 
I also tried with the 9600, as I checked both welds several times.

paveldn, 2023-07-21:
@scaraps 
try to change tx <-> rx

scaraps, 2023-07-21:
> @scaraps try to change tx <-> rx

no change!

paveldn, 2023-07-21:
> no change!
I see only two options here:
1. Something is wrong with your connection to AC, it can be anything like wires too long, too much electrical noise, or your AC control module doesn't understand 3.3 V levels.
2. You have another control module for your AC which is not speaking Haier protocol
I am afraid I can't help you either way. 
If you have an original module from your AC and have knowledge of how to sniff UART communication you can try it. Then we can check what module is sending to your AC and see if your AC speaks Haier. That is what I was doing when started this project. Checking what original module sends to AC

scaraps, 2023-07-21:
Oh shit! i connect the esp to the 5V!

knathan1987, 2023-07-24:
OK, finally got the original module connected to my wifi.

This is the output of my uart sniffer while the AC is off:
```
2023-07-24T20:01:34.630Z >>> FF FF 08 40 00 00 00 00 00 73 BB 87 01
2023-07-24T20:01:34.675Z <<< FF FF 14 40 00 00 00 00 00 74 0F 5A 00 00 00 00 00 00 00 00 00 00 31 A1 A3
2023-07-24T20:01:34.676Z ---------------------------------------------
2023-07-24T20:01:34.856Z >>> FF FF 0A 40 00 00 00 00 00 01 4D 01 99 B3 B4
2023-07-24T20:01:34.953Z <<< FF FF 32 40 00 00 00 00 00 02 6D 01 06 00 25 00 02 00 14 00 00 00 00 00 01 00 00 03 09 03 32 32 5C 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 F6 A9 71
2023-07-24T20:01:34.954Z ---------------------------------------------
2023-07-24T20:01:35.047Z >>> FF FF 08 40 00 00 00 00 00 FC 44 23 40
2023-07-24T20:01:35.094Z <<< FF FF 0E 40 00 00 00 00 00 FD 00 00 00 01 00 00 4C 01 8C
2023-07-24T20:01:35.095Z ---------------------------------------------
2023-07-24T20:01:36.091Z >>> FF FF 0A 40 00 00 00 00 00 01 4D FE 96 F3 F4
2023-07-24T20:01:36.206Z <<< FF FF 40 40 00 00 00 00 00 02 7D 01 06 00 25 00 02 00 14 00 00 00 00 00 01 00 00 03 09 03 32 32 5C 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 53 5C 00 00 00 00 01 FF 55 02 20 01 68 A3 7F 25
2023-07-24T20:01:36.207Z ---------------------------------------------
2023-07-24T20:01:36.327Z >>> FF FF 0C 40 00 00 00 00 00 F7 00 00 00 35 78 13 70
2023-07-24T20:01:36.347Z <<< FF FF 08 40 00 00 00 00 01 05 4E F1 81
2023-07-24T20:01:36.350Z ---------------------------------------------
2023-07-24T20:01:39.738Z >>> FF FF 08 40 00 00 00 00 00 73 BB 87 01
2023-07-24T20:01:39.778Z <<< FF FF 14 40 00 00 00 00 00 74 0F 5A 00 00 00 00 00 00 00 00 00 00 31 A1 A3
2023-07-24T20:01:39.779Z ---------------------------------------------
2023-07-24T20:01:39.940Z >>> FF FF 0A 40 00 00 00 00 00 01 4D 01 99 B3 B4
2023-07-24T20:01:40.052Z <<< FF FF 32 40 00 00 00 00 00 02 6D 01 06 00 25 00 02 00 14 00 00 00 00 00 01 00 00 03 09 03 32 32 5C 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 F6 A9 71
2023-07-24T20:01:40.053Z ---------------------------------------------
2023-07-24T20:01:40.158Z >>> FF FF 08 40 00 00 00 00 00 FC 44 23 40
2023-07-24T20:01:40.181Z <<< FF FF 0E 40 00 00 00 00 00 FD 00 00 00 01 00 00 4C 01 8C
2023-07-24T20:01:40.182Z ---------------------------------------------
2023-07-24T20:01:44.919Z >>> FF FF 08 40 00 00 00 00 00 73 BB 87 01
2023-07-24T20:01:44.966Z <<< FF FF 14 40 00 00 00 00 00 74 0F 5A 00 00 00 00 00 00 00 00 00 00 31 A1 A3
2023-07-24T20:01:44.967Z ---------------------------------------------
2023-07-24T20:01:45.133Z >>> FF FF 0A 40 00 00 00 00 00 01 4D 01 99 B3 B4
2023-07-24T20:01:45.224Z <<< FF FF 32 40 00 00 00 00 00 02 6D 01 06 00 25 00 02 00 14 00 00 00 00 00 01 00 00 03 09 03 32 32 5C 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 F6 A9 71
2023-07-24T20:01:45.225Z ---------------------------------------------
2023-07-24T20:01:45.320Z >>> FF FF 08 40 00 00 00 00 00 FC 44 23 40
2023-07-24T20:01:45.372Z <<< FF FF 0E 40 00 00 00 00 00 FD 00 00 00 01 00 00 4C 01 8C
2023-07-24T20:01:45.373Z ---------------------------------------------
2023-07-24T20:01:46.531Z >>> FF FF 0C 40 00 00 00 00 00 F7 00 00 00 35 78 13 70
2023-07-24T20:01:46.555Z <<< FF FF 08 40 00 00 00 00 01 05 4E F1 81
2023-07-24T20:01:46.556Z ---------------------------------------------
``` 

Will start playing with some settings now 😇

paveldn, 2023-07-27:
Thanks @knathan1987 !
I did a short analysis of your logs and start getting a little bit what is happening. So your AC is using a different way of setting parameters. "Normal" hOn AC (by normal I mean all that I saw before) sends AC state with one command 01 60 01, your AC is sending data per parameter, for example, command 01 5D 01 00 01 - turn AC on, 01 5D 01 00 00 - turn off etc.
From your logs, I got a couple of commands:
5D 01 - turn on/off
5D 05 - vertical swing mode
5D 04 - ??? (have no idea)
5D 02 - set target temperature
5D 19 - Quiet mode
5D 1A - Boost mode
So to support your AC I will need to reimplement all control parts and I can't make it without you. You will need to help me with figuring out all other commands and with testing of course.

knathan1987, 2023-07-27:
Hi @paveldn, 

sure, i'll do whatever i can ☺️
But it seems that there are some functions of my AC are not available with the hOn App (like setting target humidity in dry mode - i can set that on the remote but don't know if it works 😅)

knathan1987, 2023-07-27:
Oh... And i tried to log the debug output of the original module.... the module was really talkative at first.
but i think it got updated as soon it was connected to the internet.... now it only outputs an AWS ServerAddress and some other variables at boot.... nothing more 😔

paveldn, 2023-07-29:
@knathan1987 
OK, So I would like you to do some experiments from the hOn app that can help me to figure out some commands:
1. Set different fan modes
2.  Change horizontal swing mode (if your AC supports it)
3. Turn on/off health mode (ions)
4. Start cleaning (if possible)
With these functions (especially 1), I can try to start working on basic functionality for your AC.

knathan1987, 2023-07-29:
> 5D 05 - vertical swing mode

I think 5D 05 is fan speed.
- 5D 05 00 05 -> Auto (see log1690446661.txt)
- 5D 05 00 03 -> Low (see log1690443490.txt)
- 5D 05 00 02 -> Medium (see log1690443923.txt)
- 5D 05 00 01 -> High (see log1690444396.txt)

> 1. Set different fan modes

see above 😇

> 2\. Change horizontal swing mode (if your AC supports it)

My AC does not support any swing modes.

> 3\. Turn on/off health mode (ions)

I think 5D 0B 00 01 enables health mode. Because that command only occurs in log1690446661.txt (where health mode was enabled)

> 4\. Start cleaning (if possible)

Theres no option to do that on the hOn-App. Dont know if my AC even supports that 😅

knathan1987, 2023-07-29:
5D 04 seems to be the operating mode (i think)...
- 5D 04 00 00 -> Auto
- 5D 04 00 01 -> Cooling
- 5D 04 00 02 -> Dry
- 5D 04 00 04 -> Heating
- 5D 04 00 06 -> Fan

paveldn, 2023-08-03:
OK. I have a test version for you:
https://github.com/paveldn/haier-esphome/tree/AD71S2SM3FA_test
So you will need to add some functions in your yaml file,  something like this:
```
packages:
  local_haier: !include configs/external_components/local_haier.yaml
  haier_base: !include .haier-hon-base.yaml
  test_cmd1: !include configs/number/command_code.yaml
  test_cmd2: !include configs/number/command_arg.yaml
  test_btn: !include configs/button/send_test_command.yaml
```
It will add your device some controls:
[image]
Test command code is the command code
Test command argument - argument.
So for example to call set cooling mode you will need to set the Test command code to 4 (keep in mind that this is decimal digits not hexadecimal) Test command argument to 1. Then click Send test command button.
You will need to go over commands one by one. try all unknown. Probably some of them will be not supported. I will need logs for that. Preferably with your comment if you noticed some changes in AC. These experiments should be done with AC that is working. I would suggest trying every command with argument 1.

knathan1987, 2023-08-05:
My ESP keeps rebooting with the test version installed.

> [D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
> [D][haier.protocol:019]: Frame found: type 02, data: 6D 01 0C 00 23 00 02 00 14 00 00 00 00 00 01 00 00 03 09 03 2E 32 58 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 0
> [D][sensor:094]: 'Haier AC outdoor temperature': Sending state 24.00000 °C with 0 decimals of accuracy
> [D][climate:378]: 'Haier AC' - Sending state:
> [D][climate:381]:   Mode: OFF
> [D][climate:386]:   Fan Mode: LOW
> [D][climate:392]:   Preset: NONE
> [D][climate:398]:   Swing Mode: OFF
> [D][climate:401]:   Current Temperature: 23.00°C
> [D][climate:407]:   Target Temperature: 28.00°C
> [I][haier.climate:816]: HVAC values changed
> [I][haier.climate:818]: HVAC Mode = 0x1
> [I][haier.climate:820]: Fan speed Status = 0x3
> [I][haier.climate:822]: Horizontal Swing Status = 0x0
> [I][haier.climate:824]: Vertical Swing Status = 0x0
> [I][haier.climate:826]: Set Point Status = 0xC
> 
> Stack smashing protect failure!
> 
> abort() was called at PC 0x401005eb on core 1
> 
> Backtrace:0x400837cd:0x3ffb25600x4008b805:0x3ffb2580 0x40090d85:0x3ffb25a0 0x401005eb:0x3ffb2620 0x400dce5a:0x3ffb2640 0x400dcf11:0x3ffb26a0 0x40173f8a:0x3ffb26c0 0x40 
> 
> 
> ELF file SHA256: 0000000000000000
> 
> Rebooting...

I used an Stacktrace decoder:

> 0x401005eb: __stack_chk_fail at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/stack_check.c:36
> 0x400837cd: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
> 0x4008b805: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
> 0x40090d85: abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
> 0x401005eb: __stack_chk_fail at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/stack_check.c:36
> 0x400dce5a: esphome::haier::HonClimate::process_status_message_(unsigned char const*, unsigned char) at /config/.esphome/build/haier/src/esphome/components/haier/hon_climate.cpp:829
> 0x400dcf11: esphome::haier::HonClimate::status_handler_(unsigned char, unsigned char, unsigned char const*, unsigned int) at /config/.esphome/build/haier/src/esphome/components/haier/hon_climate.cpp:180
> 0x40173f8a: haier_protocol::HandlerError std::__invoke_impl(std::__invoke_memfun_deref, haier_protocol::HandlerError (esphome::haier::HonClimate&::*)(unsigned char, unsigned char, unsigned char const*, unsigned int), esphome::haier::HonClimate*&, unsigned char, unsigned char, unsigned char const*, unsigned int&&) at /config/.esphome/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/invoke.h:73
> 0x40173f8a: std::__invoke_result::type std::__invoke(haier_protocol::HandlerError (esphome::haier::HonClimate&::*)(unsigned char, unsigned char, unsigned char const*, unsigned int), esphome::haier::HonClimate*&, unsigned char, unsigned char, unsigned char const*, unsigned int) at /config/.esphome/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/invoke.h:95
> 0x40173f8a: haier_protocol::HandlerError std::_Bind, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>)::*)(unsigned char, unsigned char, unsigned char const*, unsigned int)>::__call(std::tuple&&, std::_Index_tuple<((unsigned int)0, (unsigned int)1, (unsigned int)2, (unsigned int)3, (unsigned int)4)...>) at /config/.esphome/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/functional:400
> 0x40173f8a: haier_protocol::HandlerError std::_Bind, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>)::*)(unsigned char, unsigned char, unsigned char const*, unsigned int)>::operator()(unsigned char, unsigned char, unsigned char const*, unsigned int&&) at /config/.esphome/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/functional:484
> 0x40173f8a: std::_Function_handler, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>)::*)(unsigned char, unsigned char, unsigned char const*, unsigned int)> >::_M_invoke(std::_Any_data const&, unsigned char&&, unsigned char&&, unsigned char const*&&, unsigned int&&) at /config/.esphome/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:282

paveldn, 2023-08-06:
OK. Thanks for the detailed info. It looks like I need to kill sending normal control commands completely. Will prepare you new version.

paveldn, 2023-09-14:
@knathan1987, I know it has been a long time, but I finally changed and fixed some potential problems on AD71S2SM3FA_test branch. Can you give it another try?
Please keep in mind that you need to use  `control_packet_size: 18` option for climate section like in `wemos_d1_mini.yaml` file on the branch.

paveldn, 2023-10-26:
The issue is fixed on the latest dev branch.
Please let me know if you will try it

paveldn, 2023-11-20:
Fixed on dev branch

closed 2023-11-20 by paveldn