paveldn/haier-esphome #21: When sending a command, the air conditioner emits 16-17 signals and only then performs an action.

Dmitry3650, 2023-07-31:
Until the last update, my 3 Haier air conditioners on ESP 32 worked perfectly! But after the latest ESP Home update, the air conditioner stopped working.
1) After flashing the ESP 32 with a new code, the air conditioner still does not work and gives an error 
[W][haier.protocol:013]: Answer handler error, msg=61, answ=03, err=7
2) I tried to change hOn to smartair2 and the air conditioner worked. But each time a command is sent, the air conditioner emits 16-17 beeps, as if I pressed the same button 17 times on the remote control and only then performs the action. I tried to edit the code according to the manual but the problem with sound signals could not be solved ... Do you have any ideas for solving such a problem?

Dmitry3650, 2023-07-31:
INFO Reading configuration /config/esphome/cabinet-haier.yaml...
INFO Detected log level for Haier protocol: DEBUG
INFO Starting log output from cabinet-haier.local using esphome API
INFO Successfully connected to cabinet-haier.local
[11:40:39][I][app:102]: ESPHome version 2023.7.0 compiled on Jul 31 2023, 11:38:10
[11:40:39][C][wifi:543]: WiFi:
[11:40:39][C][wifi:379]:   Local MAC: E0:5A:1B:A2:5A:50
[11:40:39][C][wifi:380]:   SSID: [redacted]
[11:40:39][C][wifi:381]:   IP Address: 192.168.1.107
[11:40:39][C][wifi:383]:   BSSID: [redacted]
[11:40:39][C][wifi:384]:   Hostname: 'cabinet-haier'
[11:40:39][C][wifi:386]:   Signal strength: -45 dB ▂▄▆█
[11:40:39][C][wifi:390]:   Channel: 5
[11:40:39][C][wifi:391]:   Subnet: 255.255.255.0
[11:40:39][C][wifi:392]:   Gateway: 192.168.1.1
[11:40:39][C][wifi:393]:   DNS1: 192.168.1.1
[11:40:39][C][wifi:394]:   DNS2: 0.0.0.0
[11:40:39][C][logger:301]: Logger:
[11:40:39][C][logger:302]:   Level: DEBUG
[11:40:39][C][logger:303]:   Log Baud Rate: 115200
[11:40:39][C][logger:305]:   Hardware UART: UART0
[11:40:40][C][uart.arduino_esp32:124]: UART Bus 1:
[11:40:40][C][uart.arduino_esp32:125]:   TX Pin: GPIO17
[11:40:40][C][uart.arduino_esp32:126]:   RX Pin: GPIO16
[11:40:40][C][uart.arduino_esp32:128]:   RX Buffer Size: 256
[11:40:40][C][uart.arduino_esp32:130]:   Baud Rate: 9600 baud
[11:40:40][C][uart.arduino_esp32:131]:   Data Bits: 8
[11:40:40][C][uart.arduino_esp32:132]:   Parity: NONE
[11:40:40][C][uart.arduino_esp32:133]:   Stop bits: 1
[11:40:40][C][haier.climate:193]: Haier Climate 'Haier AC'
[11:40:40][C][haier.climate:195]:   Device communication status: established
[11:40:40][C][haier.climate:071]:   Protocol version: smartAir2
[11:40:40][C][captive_portal:088]: Captive Portal:
[11:40:40][C][mdns:112]: mDNS:
[11:40:40][C][mdns:113]:   Hostname: cabinet-haier
[11:40:40][C][ota:093]: Over-The-Air Updates:
[11:40:40][C][ota:094]:   Address: cabinet-haier.local:3232
[11:40:40][C][api:138]: API Server:
[11:40:40][C][api:139]:   Address: cabinet-haier.local:6053
[11:40:40][C][api:141]:   Using noise encryption: YES
[11:40:44][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[11:40:44][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1C 00 37 00 7F 00 00 00 00 00 01 00 01 00 00 00 00 00 00 00 00 00 00
[11:40:44][D][haier.climate:439]: HVAC Mode = 0x1
[11:40:44][D][haier.climate:441]: Fan speed Status = 0x1
[11:40:44][D][haier.climate:443]: Horizontal Swing Status = 0x0
[11:40:44][D][haier.climate:445]: Vertical Swing Status = 0x0
[11:40:44][D][haier.climate:447]: Set Point Status = 0x0
[11:40:49][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[11:40:49][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1C 00 37 00 7F 00 00 00 00 00 01 00 01 00 00 00 00 00 00 00 00 00 00
[11:40:49][D][haier.climate:439]: HVAC Mode = 0x1
[11:40:49][D][haier.climate:441]: Fan speed Status = 0x1
[11:40:49][D][haier.climate:443]: Horizontal Swing Status = 0x0
[11:40:49][D][haier.climate:445]: Vertical Swing Status = 0x0
[11:40:49][D][haier.climate:447]: Set Point Status = 0x0
[11:40:54][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[11:40:54][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1C 00 37 00 7F 00 00 00 00 00 01 00 01 00 00 00 00 00 00 00 00 00 00
[11:40:54][D][haier.climate:439]: HVAC Mode = 0x1
[11:40:54][D][haier.climate:441]: Fan speed Status = 0x1
[11:40:54][D][haier.climate:443]: Horizontal Swing Status = 0x0
[11:40:54][D][haier.climate:445]: Vertical Swing Status = 0x0
[11:40:54][D][haier.climate:447]: Set Point Status = 0x0

paveldn, 2023-07-31:
Hi @Dmitry3650.

Which branch are you using, is this master or dev? 
I need logs where you triggering the problem. In the log that you provided nothing happens, just esp asking status. 
Also, I need your config.

Dmitry3650, 2023-07-31:
captive_portal:
    
uart:
  baud_rate: 9600
  tx_pin: 17
  rx_pin: 16
  id: ac_port

climate:
  - platform: haier
    id: haier_ac
    protocol: smartair2
    name: Haier AC
    uart_id: ac_port
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:
    - 'OFF'
    - AUTO
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    supported_swing_modes:
    - 'OFF'
    - VERTICAL
    - HORIZONTAL
    - BOTH

Dmitry3650, 2023-07-31:
[D][climate:010]: 'Haier AC' - Setting
[14:40:08][D][climate:014]:   Mode: COOL
[14:40:08][D][Control:265]: Control call
[14:40:08][I][haier.climate:128]: Control packet sent
[14:40:08][D][haier.protocol:019]: Sending frame: type 01, data: 4D 5F 00 1C 00 38 00 00 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:09][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:09][W][haier.climate:170]: Answer timeout for command 01, phase WAITING_CONTROL_ANSWER
[14:40:09][W][haier.protocol:013]: Unsupported message received: type 02 data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:09][W][haier.protocol:013]: Message handler error, msg=02, err=1
[14:40:09][I][haier.climate:128]: Control packet sent
[14:40:09][D][haier.protocol:019]: Sending frame: type 01, data: 4D 5F 00 1C 00 38 00 00 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:09][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:09][W][haier.climate:170]: Answer timeout for command 01, phase WAITING_CONTROL_ANSWER
[14:40:09][W][haier.protocol:013]: Unsupported message received: type 02 data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:09][W][haier.protocol:013]: Message handler error, msg=02, err=1
[14:40:09][I][haier.climate:128]: Control packet sent
[14:40:09][D][haier.protocol:019]: Sending frame: type 01, data: 4D 5F 00 1C 00 38 00 00 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:09][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:09][W][haier.climate:170]: Answer timeout for command 01, phase WAITING_CONTROL_ANSWER
[14:40:10][W][haier.protocol:013]: Unsupported message received: type 02 data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:10][W][haier.protocol:013]: Message handler error, msg=02, err=1
[14:40:10][I][haier.climate:128]: Control packet sent
[14:40:10][D][haier.protocol:019]: Sending frame: type 01, data: 4D 5F 00 1C 00 38 00 00 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:10][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:10][W][haier.climate:170]: Answer timeout for command 01, phase WAITING_CONTROL_ANSWER
[14:40:10][W][haier.protocol:013]: Unsupported message received: type 02 data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:10][W][haier.protocol:013]: Message handler error, msg=02, err=1
[14:40:10][I][haier.climate:128]: Control packet sent
[14:40:10][D][haier.protocol:019]: Sending frame: type 01, data: 4D 5F 00 1C 00 38 00 00 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:10][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:10][W][haier.climate:170]: Answer timeout for command 01, phase WAITING_CONTROL_ANSWER
[14:40:10][W][haier.protocol:013]: Unsupported message received: type 02 data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:10][W][haier.protocol:013]: Message handler error, msg=02, err=1
[14:40:11][I][haier.climate:128]: Control packet sent
[14:40:11][D][haier.protocol:019]: Sending frame: type 01, data: 4D 5F 00 1C 00 38 00 00 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:11][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:11][W][haier.climate:170]: Answer timeout for command 01, phase WAITING_CONTROL_ANSWER
[14:40:11][W][haier.protocol:013]: Unsupported message received: type 02 data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:11][W][haier.protocol:013]: Message handler error, msg=02, err=1
[14:40:11][I][haier.climate:128]: Control packet sent
[14:40:11][D][haier.protocol:019]: Sending frame: type 01, data: 4D 5F 00 1C 00 38 00 00 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:11][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:11][W][haier.climate:170]: Answer timeout for command 01, phase WAITING_CONTROL_ANSWER
[14:40:11][W][haier.protocol:013]: Unsupported message received: type 02 data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:11][W][haier.protocol:013]: Message handler error, msg=02, err=1
[14:40:11][I][haier.climate:128]: Control packet sent
[14:40:11][D][haier.protocol:019]: Sending frame: type 01, data: 4D 5F 00 1C 00 38 00 00 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:12][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:12][W][haier.climate:170]: Answer timeout for command 01, phase WAITING_CONTROL_ANSWER
[14:40:12][W][haier.protocol:013]: Unsupported message received: type 02 data: 6D 01 00 1C 00 38 00 7F 00 00 00 00 00 01 00 01 00 00 00 01 00 00 00 00 00 00
[14:40:12][W][haier.protocol:013]: Message handler error, msg=02, err=1
[14:40:12][I][haier.climate:128]: Control packet sent
[... cut ...]
[14:40:17][I][haier.climate:437]: HVAC values changed
[14:40:17][I][haier.climate:439]: HVAC Mode = 0x1
[14:40:17][I][haier.climate:441]: Fan speed Status = 0x1
[14:40:17][I][haier.climate:443]: Horizontal Swing Status = 0x0
[14:40:17][I][haier.climate:445]: Vertical Swing Status = 0x0
[14:40:17][I][haier.climate:447]: Set Point Status = 0x0
[14:40:17][W][component:204]: Component haier.climate took a long time for an operation (0.08 s).
[14:40:17][W][component:205]: Components should block for at most 20-30ms.

Dmitry3650, 2023-07-31:
Did I send you what you asked for?

Dmitry3650, 2023-07-31:
I don't understand which branch master or dev are talking about..

paveldn, 2023-07-31:
Can you provide me full config not just climate? You can replace all keys, IPs, MAC addresses with XXXX.
I  am asking because it is possible that other components interfere like ble client etc.
The last log you provided is OK I know what is happening now just don't know yet why.

Dmitry3650, 2023-07-31:
sorry for the long answer!

esphome:
  name: "cabinet-haier"
  friendly_name: cabinet-haier

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "хххххххххххххххххххххххххххххххххххххххххх"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "хххххххххххххххххххххх"
    password: "ххххххххххххххххххххххх"

captive_portal:
    
uart:
  baud_rate: 9600
  tx_pin: 17
  rx_pin: 16
  id: ac_port

climate:
  - platform: haier
    id: haier_ac
    protocol: smartair2
    name: Haier AC
    uart_id: ac_port
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:
    - 'OFF'
    - AUTO
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    supported_swing_modes:
    - 'OFF'
    - VERTICAL
    - HORIZONTAL
    - BOTH

Dmitry3650, 2023-07-31:
[image]

paveldn, 2023-07-31:
OK, now it is clear. I thought you were using my component all the time but you used the previous implementation (which was a completely different code in 2023.6.0). I suppose you have Candy, isn't it? They answer a little bit slowly. 
There is no fix in some repo yet but you can use the solution from my dev branch.
Add external_components section to your config:
```
external_components:
  source: github://paveldn/haier-esphome@dev
```
It will use haier component from the dev branch of my repo not from esphome.
After that add field answer_timeout, like this:
```
climate:
  - platform: haier
    ... 
    answer_timeout: "200ms"
    ....
```
Start with 200ms and if it will not help increase to 300ms
Let me know if it worked.

Dmitry3650, 2023-07-31:
[image]

Thanks for the answer. What is candy? is this addon from HACS? I pasted your code but it swears for an error. If you don't mind, then you can explain in more detail what kind of candy? and how to insert your last code sent to me so that the system accepts it. Thank you in advance!

paveldn, 2023-07-31:
Candy is Candy AC something like this: https://www.flipkart.com/air-conditioners/candy~brand/pr?sid=j9e%2Cabm%2Cc54
Some of them Haier compatible. If you don;t know what is Candy I suppose you don't have one :) Just ignore my comment about Candy
Try to use this config anyway (esphome dashboard know nothing about code on my branch( just ignore errors, save and install it

Dmitry3650, 2023-07-31:
It helped me!!!! You are the best! Thanks, I struggled with this for a long time.

Dmitry3650, 2023-07-31:
Here is my working config, in case anyone needs it. Thanks paveldn.

esphome:
  name: "cabinet-haier"
  friendly_name: cabinet-haier

external_components:
  source: github://paveldn/haier-esphome@dev

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "nqMnevJufvBF4JIjqq/756XCbgpFMGQMDq0aih0CXaM="

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-A25A50"
    password: "oL7ukUM7FWiN"

captive_portal:
    
uart:
  baud_rate: 9600
  tx_pin: 17
  rx_pin: 16
  id: ac_port

climate:
  - platform: haier
    answer_timeout: "200ms"
    id: haier_ac
    protocol: smartair2
    name: Haier AC
    uart_id: ac_port
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:
    - 'OFF'
    - AUTO
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    supported_swing_modes:
    - 'OFF'
    - VERTICAL
    - HORIZONTAL
    - BOTH

Dmitry3650, 2023-07-31:
Here is my working config, in case anyone needs it. Thanks paveldn.
[image]

closed 2023-07-31 by Dmitry3650