paveldn/haier-esphome #22: Some trouble with Swing mode on AS12NB1HRA and AS12TB3HRA

sergeybelozorov, 2023-08-02:
Hi!

When I try to change swing mode on my conditioners, I have unexpected result.

Some modes are confused with each other.

And I have 2 modes not implemented in your integration: vertical down, vertical up.

Can you fix the current modes and add new ones?

This is result from debug mode on uart:

Mode: COOL
Fan Mode: LOW
Preset: NONE
Swing Mode: OFF
Current Temperature: 25.00°C
Target Temperature: 23.00°C

FF:FF:22:00:00:00:00:00:00:02:6D:01:00:19:00:3C:00:7F:00:00:00:00:00:01:00:02:00:00:00:01:00:20:00:00:00:07:91  - Real swing mode: OFF          - Swing Mode ESPHome: OFF
FF:FF:22:00:00:00:00:00:00:02:6D:01:00:19:00:3C:00:7F:00:00:00:00:00:01:00:02:00:02:00:01:00:28:00:00:00:07:9B  - Real swing mode: Horiz auto
FF:FF:22:00:00:00:00:00:00:02:6D:01:00:19:00:3C:00:7F:00:00:00:00:00:01:00:02:00:00:00:01:00:28:00:00:00:07:99  - Real swing mode: Horiz Off
FF:FF:22:00:00:00:00:00:00:02:6D:01:00:19:00:3C:00:7F:00:00:00:00:00:01:00:02:00:00:00:01:00:30:00:00:00:07:A1  - Real swing mode: **Vert Down**    - Swing Mode ESPHome: **VERTICAL**
FF:FF:22:00:00:00:00:00:00:02:6D:01:00:19:00:3C:00:7F:00:00:00:00:00:01:00:02:00:01:00:01:00:20:00:00:00:07:92  - Real swing mode: **Vert Auto**    - Swing Mode ESPHome: **BOTH**
FF:FF:22:00:00:00:00:00:00:02:6D:01:00:19:00:3C:00:7F:00:00:00:00:00:01:00:02:00:00:00:01:00:28:00:00:00:07:99  - Real swing mode: **Vert Up**      - Swing Mode ESPHome: **HORIZONTAL**

paveldn, 2023-08-02:
Hi, 

I don't think a serial protocol supports it: line Real swing mode  Horiz Off exactly the same as Real swing mode  Vert Up. Also, you are not the first one who asked to support it and all previous checks came to the same conclusions - not supported by serial protocol.

sergeybelozorov, 2023-08-03:
1. When I turn on any of the above modes, the device gets a new set of bytes, i.e. protocol explicitly supports it.
2. I can make a video that shows that when I check vertical swing mode, the air conditioner simply lowers the curtain down. This set of bytes: FF:FF:22:00:00:00:00:00:00:02:6D:01:00:19:00:3C:00:7F:00:00:00:00:00:01:00:02:00:00:00:01:00:30:00:00:00:07:A1 - Real swing mode: Vert Down - Swing Mode ESPHome: VERTICAL
Same with all other Swing modes.

paveldn, 2023-08-03:
@sergeybelozorov 
I am not doubting that AC supports it. I just say that the UART protocol not supporting it. Some features are available only through an IR remote (or wired remote in some cases). Those remotes not controlled by ECP have a separate channel. It should be possible to decode IR packets and simulate remote commands by IR transmitter (or by soldering extra wire to IR remote receiver on board) but it is a huge amount of work. Also, there will be no way to get feedback from AC. I have plans to try it but it will take a lot of time so don't expect it soon.

sergeybelozorov, 2023-08-03:
Ok, thanks!
But all set of bytes in my first message this is feedback from AC. I set Swing mode via remote and ESP logged this unique set of bytes

closed 2023-08-06 by paveldn