paveldn/haier-esphome #24: Climate Default Mode

taqee, 2023-08-17:
I have Haier AC split that works SmartAir2 protocol and everything is working perfectly except for the default mode

when I turn off the AC and turn it back on it changes the mode from the previous state to Heat no matter what was the state it will change to Heat 
I previously used MiguelAngelLV and it was working just fine and always use Auto Mode but when I changed to HaierESP this started to show on my ACs

the config is :

`
uart:
  baud_rate: 9600
  tx_pin: 1
  rx_pin: 3
  id: ac_port

climate:
  - platform: haier
    id: haier_ac
    protocol: smartAir2
    name: Bedroom AC
    uart_id: ac_port
    wifi_signal: true          
    display: true  
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:
    - 'OFF'
    - AUTO
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    supported_swing_modes:
    - 'OFF'
    - VERTICAL
    - HORIZONTAL
    - BOTH
    supported_presets:        
    - BOOST
    - COMFORT

  `

paveldn, 2023-08-17:
It changes to heat because the Home assistant tells it to do so. Not sure how it worked for Miguel's code probably there are some special things for the climate template. 
If you want to turn on the climate in the same mode as it was before turning it off I would suggest you follow way:
1.  add this to your API section:

```
api:
....
  services:
    - service: turn_on
      then:
      - climate.haier.power_on: haier_ac
    - service: turn_off
      then:
      - climate.haier.power_off: haier_ac
```
2. In home assistant you will get services: ESPHome: haier_ac_turn_off and ESPHome: haier_ac_turn_off use them instead of standard Home assistant climate_on climate_off

paveldn, 2023-09-15:
With the latest changes in ESPHome instead of "Heat" mode by default "Heat/Cool" mode will be turned on (it is like auto mode before).

closed 2023-09-15 by paveldn