paveldn/haier-esphome #26: ESPHome 2023.8.2 (SmartAir2 protocol) Update Error

Krzysztonek, 2023-08-25:
When trying to do the update of my Haier AC device, I got this:
```
INFO ESPHome 2023.8.2
INFO Reading configuration /config/esphome/smartair2-salon.yaml...
INFO Detected log level for Haier protocol: DEBUG
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 1023, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 1010, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 390, in command_compile
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 185, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 197, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 626, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 177, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/haier/climate.py", line 410, in to_code
    await climate.register_climate(var, config)
  File "/esphome/esphome/components/climate/__init__.py", line 344, in register_climate
    await setup_climate_core_(var, config)
  File "/esphome/esphome/components/climate/__init__.py", line 237, in setup_climate_core_
    visual[CONF_TEMPERATURE_STEP][CONF_TARGET_TEMPERATURE],
TypeError: tuple indices must be integers or slices, not str
```

paveldn, 2023-08-25:
Can you please provide me whole climate section of your config?

Krzysztonek, 2023-08-25:
This is my climate section:

```

climate:
  - platform: haier
    id: haier_climate
    protocol: smartAir2
    name: AC Salon
    uart_id: ac_port

switch:
  - platform: template
    id: haier_climate_display_switch
    name: AC Salon Wyświetlacz
    icon: mdi:led-on
    entity_category: config
    restore_mode: "ALWAYS_OFF"
    lambda: |-
      return id(haier_climate).get_display_state();
    turn_on_action:
      climate.haier.display_on: haier_climate
    turn_off_action:
      climate.haier.display_off: haier_climate
  - platform: template
    id: haier_climate_health_mode
    name: AC Salon Tryb HEALTH
    icon: mdi:leaf
    restore_mode: "ALWAYS_OFF"
    lambda: |-
      return id(haier_climate).get_health_mode();
    turn_on_action:
      climate.haier.health_on: haier_climate
    turn_off_action:
      climate.haier.health_off: haier_climate
  - platform: restart
    name: Restart AC Salon

sensor:
  - platform: wifi_signal
    name: AC Salon WiFi Signal
    update_interval: 30s

```

paveldn, 2023-08-26:
Ok. You are right there is a problem. I will fix it asap but for now you can add following section to your climate
''''
climate: 
  - platform: haier 
    id: haier_climate 
    protocol: smartAir2 
    name: AC Salon uart_id: ac_port
    visual: 
      temperature_step: 
        target_temperature: 1 
        current_temperature: 0.5
'''

paveldn, 2023-09-01:
Fixed on dev and master and created a pull request for ESPHome.

closed 2023-09-01 by paveldn