paveldn/haier-esphome #28: Unsupported message received with AS25S2SJ1FA

sinitsyn-alexander, 2023-09-15:
Hello.
I have an AS25S2SJ1FA indoor unit with an ESP32-for-Haier module. I did not flash this module; I took LOLIN D1 mini for experiments.
I tried the master and AD71S2SM3FA_test branches, but I get an error in the log.
It looks like my unit is not supported. I think it is returning a message that is too long.

```[18:44:54][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[18:44:55][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[18:44:55][W][haier.climate:205]: Answer timeout for command 61, phase WAITING_INIT_1_ANSWER
[18:44:55][W][component:204]: Component haier.climate took a long time for an operation (0.16 s).
[18:44:55][W][component:205]: Components should block for at most 20-30ms.
[18:44:55][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[18:44:55][W][haier.protocol:013]: Message handler error, msg=62, err=1
[18:44:55][W][component:204]: Component haier.climate took a long time for an operation (0.27 s).
[18:44:55][W][component:205]: Components should block for at most 20-30ms.
[18:45:04][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[18:45:05][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[18:45:05][W][haier.climate:205]: Answer timeout for command 61, phase WAITING_INIT_1_ANSWER
[18:45:05][W][component:204]: Component haier.climate took a long time for an operation (0.16 s).
[18:45:05][W][component:205]: Components should block for at most 20-30ms.
[18:45:05][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[18:45:05][W][haier.protocol:013]: Message handler error, msg=62, err=1
[18:45:05][W][component:204]: Component haier.climate took a long time for an operation (0.27 s).
[18:45:05][W][component:205]: Components should block for at most 20-30ms.
[18:45:14][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[18:45:15][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[18:45:15][W][haier.climate:205]: Answer timeout for command 61, phase WAITING_INIT_1_ANSWER
[18:45:15][W][component:204]: Component haier.climate took a long time for an operation (0.16 s).
[18:45:15][W][component:205]: Components should block for at most 20-30ms.
[18:45:15][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[18:45:15][W][haier.protocol:013]: Message handler error, msg=62, err=1
[18:45:15][W][component:204]: Component haier.climate took a long time for an operation (0.27 s).
[18:45:15][W][component:205]: Components should block for at most 20-30ms.```

paveldn, 2023-09-15:
Hi @sinitsyn-alexander 
Don't use AD71S2SM3FA_test branch it is for experiments with one very specific model just use master or dev. Add answer_timeout option to your climate config. Start with value 200ms if it will not help try 300ms:
```
climate:
  - platform: haier
    ...
    answer_timeout: 200ms
    ...
```

sinitsyn-alexander, 2023-09-21:
Thanks, it works!

closed 2023-09-21 by sinitsyn-alexander