paveldn/haier-esphome #29: Does not work with scenes

Marko298, 2023-09-27:
Hello, I'm facing an issue where I can't use Home Assistant scenes with Air Conditioners. When I attempt to apply a scene that includes multiple changes (such as Swing, Temperature, and Mode), it fails and produces the following log:

```
[15:15:44][D][climate:011]: 'Office AC' - Setting
[15:15:44][D][climate:015]:   Mode: COOL
[15:15:44][D][Control:300]: Control call
[15:15:44][I][haier.climate:393]: Control packet sent
[15:15:44][D][haier.protocol:019]: Sending frame: type 01, data: 60 01 05 02 22 00 02 01 00 06 00 00
[15:15:44][D][climate:011]: 'Office AC' - Setting
[15:15:44][D][climate:040]:   Target Temperature: 21.00
[15:15:44][D][Control:300]: Control call
[15:15:44][W][haier.climate:306]: Overriding old valid settings before they were applied!
[15:15:44][D][climate:011]: 'Office AC' - Setting
[15:15:44][D][climate:033]:   Preset: NONE
[15:15:44][D][Control:300]: Control call
[15:15:44][W][haier.climate:306]: Overriding old valid settings before they were applied!
[15:15:44][D][climate:011]: 'Office AC' - Setting
[15:15:44][D][climate:037]:   Swing: OFF
[15:15:44][D][Control:300]: Control call
[15:15:44][W][haier.climate:306]: Overriding old valid settings before they were applied!
```

It works if I change the parameters using automation with a 500ms delay between each step.

Is there a way to batch these changes and send them to the Air Conditioner simultaneously? Alternatively, can we add a command queue so that all commands are executed sequentially instead of being ignored?

paveldn, 2023-09-27:
Hi @Marko298,

I didn't think somebody would be fast enough to send a new command before the previous process. You are the first one who complains 😄 . Can't send commands faster. The serial protocol is a dance, there are strict timings. Queue is a good idea but all these commands can be sent with one packet because packet setting AC state in one go. Anyway, I will think about it. Thanks for reporting.

Marko298, 2023-09-28:
As an idea, we can introduce intentional delay 100-200ms and batch all state changes to be sent as a one command

Marko298, 2023-09-28:
If there is something I can help with please let me know. I have experience in programming but mostly in JS.
I can help with testing/debugging if needed

paveldn, 2023-09-28:
I made experiments and found the root cause so the biggest delay when applying settings is waiting for AC's answer. So if you send commands fast most likely ESP module will apply the first one and all others will come when ESP is waiting for the answer for the first command. They kind of merged together but once confirmation that the first command processed was received from AC, they dropped. So I made a fix with some kind of "double buffering": there is a current settings that we trying to apply now and there is a next settings that will have all incoming new settings "merged" together. Once confirmation is received next settings are copied to the current one and another command is applied. Tested it and it worked for me. I pushed the fix on de branch please test them and let me know. You will get a warning that settings coming to fast but it will work anyway.

Marko298, 2023-09-29:
will test, thanks!

Marko298, 2023-09-29:
Works like a charm! Thanks a lot!

The only side effect I noticed is a double beep sound, but I don't care

Marko298, 2023-09-29:
I wanted to take a moment to express my sincere gratitude for your contribution. It has been an invaluable tool that saved me a significant amount of time. I truly appreciate your help and quick responses. Thank you for your hard work and for making my life, so much easier.

paveldn, 2023-09-29:
Thanks for testing and for your kind words. Happy I could help.

closed 2023-09-29 by paveldn