paveldn/haier-esphome #30: New generation ESP32

sinclairfr, 2023-10-10:
Hello, 

Thanks for your hard work and thank you for my 2 Haier AC flawlessly working with your haier-esphome.
I bought a third AC unit, and I ran into some troubles.
I have a new generation of ESP32-Haier devices that has encryption enabled I have the  boot loop with `error rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)`.
I have a regular Aliexpress ESP32 device that I will use instead. I will cut the wire from the Haier AC but I just need to now the pinout I need to use in order to solder the wire to the ESP32 pins.

Can you pleas help me?

Many thanks,

paveldn, 2023-10-11:
Hello @sinclairfr,
You can see the pinout here https://raw.githubusercontent.com/paveldn/haier-esphome/master/img/ESP32_back.jpg to communicate with AC MCU_RX MCU_TX pins used, 5v and GND to power ESP32. Keep in mind that usually ESP works with 3.3v signals, not 5v (data on TX RX encoded with 3.3v signals, not with 5v). so theoretically you will need a logic converter like this https://botland.store/raspberry-pi-gpio-extensions/1513-logic-level-converter-33v-5v-i2c-uart-spi-5904422330125.html . Potentially you can connect directly without a converter but you always will have a risk that the ESP board can stop working after some time (no risk for AC).  So it is up to you. I have been using M5stamp C3 for one of my ACs for 2 years already without any problems.
That is how it looks in my case:
[image]
Port config for this:
```
uart:
  baud_rate: 9600
  tx_pin: 19
  rx_pin: 18
  id: ${uart_id}  
```
Pinout for m5stamp:
https://static-cdn.m5stack.com/resource/docs/products/core/stamp_c3/stamp_c3_09.webp

Let me know if you need more info.

Good luck!

paveldn, 2023-10-12:
Glad it helped. 
I will keep this issue open for some time in case you have other questions.