paveldn/haier-esphome #33: Support for "freeze-protection" mode with Haier AW09TE1VHA

catlee, 2023-11-15:
I have an indoor AW09TE1VHA which has a freeze protection mode where it keeps the temperature to at least 50F / 10C.

Is there a way to get this working with haier-esphome? The only way I can figure out to enable this mode right now is via the IR remote control, via the "extra function" button.

If it helps, here are some logs with this function enabled. Previously the unit was enabled in heating mode set to 22C.

```
[11:53:18][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[11:53:18][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 17 00 2C 00 00 00 00 00 00 00 02 00 02 00 03 00 01 00 00 00 00 00 06
[11:53:18][D][haier.climate:535]: HVAC Mode = 0x2
[11:53:18][D][haier.climate:537]: Fan speed Status = 0x2
[11:53:18][D][haier.climate:539]: Horizontal Swing Status = 0x0
[11:53:18][D][haier.climate:541]: Vertical Swing Status = 0x0
[11:53:18][D][haier.climate:543]: Set Point Status = 0x6
[11:53:20][D][haier.climate:128]: WiFi signal is: -43dBm => 66%
[11:53:20][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 42
[11:53:20][D][haier.protocol:019]: Frame found: type 05, data: <empty>
[11:53:23][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[11:53:23][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 17 00 2C 00 00 00 00 00 00 00 02 00 02 00 03 00 01 00 00 00 00 00 06
[11:53:23][D][haier.climate:535]: HVAC Mode = 0x2
[11:53:23][D][haier.climate:537]: Fan speed Status = 0x2
[11:53:23][D][haier.climate:539]: Horizontal Swing Status = 0x0
[11:53:23][D][haier.climate:541]: Vertical Swing Status = 0x0
[11:53:23][D][haier.climate:543]: Set Point Status = 0x6
[11:53:28][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[11:53:28][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 17 00 2C 00 00 00 00 00 00 00 02 00 02 00 03 00 01 00 00 00 00 00 06
[11:53:28][D][haier.climate:535]: HVAC Mode = 0x2
[11:53:28][D][haier.climate:537]: Fan speed Status = 0x2
[11:53:28][D][haier.climate:539]: Horizontal Swing Status = 0x0
[11:53:28][D][haier.climate:541]: Vertical Swing Status = 0x0
[11:53:28][D][haier.climate:543]: Set Point Status = 0x6
[11:53:30][D][haier.climate:128]: WiFi signal is: -44dBm => 65%
[11:53:30][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 41
[11:53:30][D][haier.protocol:019]: Frame found: type 05, data: <empty>
[11:53:33][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[11:53:33][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 17 00 2C 00 00 00 00 00 00 00 02 00 02 00 03 00 01 00 00 00 00 00 06
[11:53:33][D][haier.climate:535]: HVAC Mode = 0x2
[11:53:33][D][haier.climate:537]: Fan speed Status = 0x2
[11:53:33][D][haier.climate:539]: Horizontal Swing Status = 0x0
[11:53:33][D][haier.climate:541]: Vertical Swing Status = 0x0
[11:53:33][D][haier.climate:543]: Set Point Status = 0x6
[11:53:38][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[11:53:38][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 17 00 2D 00 00 00 00 00 00 00 02 00 02 00 03 00 01 00 00 00 00 00 06
[11:53:38][D][haier.climate:535]: HVAC Mode = 0x2
[11:53:38][D][haier.climate:537]: Fan speed Status = 0x2
[11:53:38][D][haier.climate:539]: Horizontal Swing Status = 0x0
[11:53:38][D][haier.climate:541]: Vertical Swing Status = 0x0
[11:53:38][D][haier.climate:543]: Set Point Status = 0x6
[11:53:40][D][haier.climate:128]: WiFi signal is: -46dBm => 64%
[11:53:40][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 40
[11:53:40][D][haier.protocol:019]: Frame found: type 05, data: <empty>
```

paveldn, 2023-11-15:
Hi, 

Yes, it should be quite straightforward. I thought nobody needed it. Will add it soon.

catlee, 2023-11-15:
Thank you! Let me know if I can help in any way.

paveldn, 2023-11-16:
Hi @catlee,

Can you, please, test it? 
Implementation is on the top of the dev branch. I added it as an "Away" preset. It can be enabled only in Heat mode. So first you need to enable Heat mode, then set the "Away" preset. If you switch mode or turn off AC - "Away" mode will be disabled.

catlee, 2023-11-16:
Hi @paveldn 

This doesn't seem to work just yet.

When I set the preset to "Away", it immediately switches back to no preset selected.

```
[16:32:26][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:32:26][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 15 00 28 00 04 00 00 00 00 00 02 00 02 00 03 00 01 00 00 00 00 00 05
[16:32:26][D][haier.climate:536]: HVAC Mode = 0x2
[16:32:26][D][haier.climate:537]: Fan speed Status = 0x2
[16:32:26][D][haier.climate:538]: Horizontal Swing Status = 0x0
[16:32:26][D][haier.climate:539]: Vertical Swing Status = 0x0
[16:32:26][D][haier.climate:540]: Set Point Status = 0x5
[16:32:28][D][haier.climate:118]: WiFi signal is: -34dBm => 73%
[16:32:28][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 49
[16:32:28][D][climate:011]: 'Haier AC' - Setting
[16:32:28][D][climate:033]:   Preset: AWAY
[16:32:28][D][Control:327]: Control call
[16:32:28][D][haier.protocol:019]: Frame found: type 05, data: <empty>
[16:32:28][I][haier.climate:178]: Sending control packet
[16:32:28][D][haier.protocol:019]: Sending frame: type 01, data: 4D 5F 00 15 00 28 00 00 00 00 00 00 00 02 00 02 00 03 00 41 00 00 00 00 00 05
[16:32:28][D][haier.protocol:019]: Frame found: type 03, data: 00 00
[16:32:28][W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
[16:32:31][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:32:31][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 15 00 28 00 06 00 00 00 00 00 02 00 02 00 03 00 01 00 00 00 00 00 05
[16:32:31][D][haier.climate:536]: HVAC Mode = 0x2
[16:32:31][D][haier.climate:537]: Fan speed Status = 0x2
[16:32:31][D][haier.climate:538]: Horizontal Swing Status = 0x0
[16:32:31][D][haier.climate:539]: Vertical Swing Status = 0x0
[16:32:31][D][haier.climate:540]: Set Point Status = 0x5
```

paveldn, 2023-11-16:
I just checked again I don't have AC that supports SmartAir2 but I do have KZW-W002 and AC simulator so I can play with the smartAir2 app and see what it is doing when you change something on your phone. And the smartAir2 app doing exactly what I am doing. There is a chance that your AC doesn't support it over serial protocol only by remote. Or your AC uses different bits.
In your logs, I see that AC rejecting the setting a proper bit ( Frame found: type 03, data: 00 00 - that is an error)
I will need your help I need you to take logs when you enable 10 degree function from remote but I need you to include at least one status frame before you enable this function. In this case, I can compare and see what is changed. If nothing (except temp and humidity) then your AC doesn't support it over serial. If there will be changes I can make you new versions.

catlee, 2023-11-17:
```
# Started in 10-degree mode:
[17:12:49][D][haier.climate:541]: Fan speed Status = 0x2
[17:12:49][D][haier.climate:542]: Horizontal Swing Status = 0x0
[17:12:49][D][haier.climate:543]: Vertical Swing Status = 0x0
[17:12:49][D][haier.climate:544]: Set Point Status = 0x5
[17:12:50][D][haier.climate:118]: WiFi signal is: -35dBm => 72%
[17:12:50][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 48
[17:12:51][D][haier.protocol:019]: Frame found: type 05, data: <empty>
[17:12:54][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:12:54][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 12 00 26 00 00 00 00 00 00 00 02
[17:12:54][D][haier.climate:540]: HVAC Mode = 0x2
[17:12:54][D][haier.climate:541]: Fan speed Status = 0x2
[17:12:54][D][haier.climate:542]: Horizontal Swing Status = 0x0
[17:12:54][D][haier.climate:543]: Vertical Swing Status = 0x0
[17:12:54][D][haier.climate:544]: Set Point Status = 0x5
[17:12:59][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:12:59][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 12 00 26 00 00 00 00 00 00 00 02
[17:12:59][D][haier.climate:540]: HVAC Mode = 0x2
[17:12:59][D][haier.climate:541]: Fan speed Status = 0x2
[17:12:59][D][haier.climate:542]: Horizontal Swing Status = 0x0
[17:12:59][D][haier.climate:543]: Vertical Swing Status = 0x0
[17:12:59][D][haier.climate:544]: Set Point Status = 0x5
[17:13:00][D][haier.climate:118]: WiFi signal is: -36dBm => 71%
[17:13:00][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 47
[17:13:01][D][haier.protocol:019]: Frame found: type 05, data: <empty>
[17:13:03][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:13:04][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 12 00 26 00 00 00 00 00 00 00 02
[17:13:04][D][haier.climate:540]: HVAC Mode = 0x2
[17:13:04][D][haier.climate:541]: Fan speed Status = 0x2
[17:13:04][D][haier.climate:542]: Horizontal Swing Status = 0x0
[17:13:04][D][haier.climate:543]: Vertical Swing Status = 0x0
[17:13:04][D][haier.climate:544]: Set Point Status = 0x5
[17:13:09][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:13:09][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 12 00 26 00 00 00 00 00 00 00 02
[17:13:09][D][haier.climate:540]: HVAC Mode = 0x2
[17:13:09][D][haier.climate:541]: Fan speed Status = 0x2
[17:13:09][D][haier.climate:542]: Horizontal Swing Status = 0x0
[17:13:09][D][haier.climate:543]: Vertical Swing Status = 0x0
[17:13:09][D][haier.climate:544]: Set Point Status = 0x5

# Use IR remote to set HEAT mode

[17:13:11][D][haier.climate:118]: WiFi signal is: -35dBm => 72%
[17:13:11][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 48
[17:13:11][D][haier.protocol:019]: Frame found: type 05, data: <empty>
[17:13:14][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:13:14][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 12 00 26 00 00 00 00 00 00 00 02
[17:13:14][D][haier.climate:540]: HVAC Mode = 0x2
[17:13:14][D][haier.climate:541]: Fan speed Status = 0x2
[17:13:14][D][haier.climate:542]: Horizontal Swing Status = 0x0
[17:13:14][D][haier.climate:543]: Vertical Swing Status = 0x0
[17:13:14][D][haier.climate:544]: Set Point Status = 0x5
[17:13:19][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:13:19][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 12 00 26 00 00 00 00 00 00 00 02 00 02 00 03 00 01 00 00 00 00 00 05
[17:13:19][D][haier.climate:540]: HVAC Mode = 0x2
[17:13:19][D][haier.climate:541]: Fan speed Status = 0x2
[17:13:19][D][haier.climate:542]: Horizontal Swing Status = 0x0
[17:13:19][D][haier.climate:543]: Vertical Swing Status = 0x0
[17:13:19][D][haier.climate:544]: Set Point Status = 0x5
[17:13:21][D][haier.climate:118]: WiFi signal is: -35dBm => 72%
[17:13:21][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 48
[17:13:21][D][haier.protocol:019]: Frame found: type 05, data: <empty>
[17:13:24][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:13:24][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 12 00 26 00 00 00 00 00 00 00 02 00 02 00 03 00 01 00 00 00 00 00 05
[17:13:24][D][haier.climate:540]: HVAC Mode = 0x2
[17:13:24][D][haier.climate:541]: Fan speed Status = 0x2
[17:13:24][D][haier.climate:542]: Horizontal Swing Status = 0x0
[17:13:24][D][haier.climate:543]: Vertical Swing Status = 0x0
[17:13:24][D][haier.climate:544]: Set Point Status = 0x5
[... 3 lines cut ...]
[17:13:29][D][haier.climate:541]: Fan speed Status = 0x2
[17:13:29][D][haier.climate:542]: Horizontal Swing Status = 0x0
[17:13:29][D][haier.climate:543]: Vertical Swing Status = 0x0
[17:13:29][D][haier.climate:544]: Set Point Status = 0x5

# Use IR remote to restore 10 degree mode

[17:13:31][D][haier.climate:118]: WiFi signal is: -35dBm => 72%
[17:13:31][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 48
[17:13:31][D][haier.protocol:019]: Frame found: type 05, data: <empty>
[17:13:34][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:13:34][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 12 00 26 00 02 00 00 00 00 00 02 00 02 00 03 00 01 00 00 00 00 00 05
[17:13:34][D][haier.climate:540]: HVAC Mode = 0x2
[17:13:34][D][haier.climate:541]: Fan speed Status = 0x2
[17:13:34][D][haier.climate:542]: Horizontal Swing Status = 0x0
[17:13:34][D][haier.climate:543]: Vertical Swing Status = 0x0
[... cut ...]
[17:13:39][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:13:39][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 12 00 26 00 00 00 00 00 00 00 02 00 02 00 03 00 01 00 00 00 00 00 05
[17:13:39][D][haier.climate:540]: HVAC Mode = 0x2
[17:13:39][D][haier.climate:541]: Fan speed Status = 0x2
[17:13:39][D][haier.climate:542]: Horizontal Swing Status = 0x0
[17:13:39][D][haier.climate:543]: Vertical Swing Status = 0x0
[17:13:39][D][haier.climate:544]: Set Point Status = 0x5
```

paveldn, 2023-11-19:
@catlee Unfortunately I don't see a change of bits when you enable/disable 10 freeze protection mode. 
It means it is controlled by IR remote but not by serial protocol. I am afraid I can't help you here. 
Potentially it should be possible to add by integration of IR control but, there will be no feedback, and still need to figure out IR commands for your AC.

catlee, 2023-11-20:
@paveldn ok, thank you so much for looking into this! I'll explore the IR option for now.

paveldn, 2023-11-20:
OK sorry can't help

closed 2023-11-20 by paveldn