paveldn/haier-esphome #36: Health and 10C frost protection mode not added

TheHawk1337, 2024-01-18:
Hi,

I'm quite a noob with this ESP Home stuff but managed to get my haier AC working (Hon version). Many thanks for that.

But I did notice that the "health" (Uv Sterilizing) and "10C frost protection" modes are not there. Is this a simple config change in the ESP yaml or is this not supported by the Climate feature?

Thanks!

paveldn, 2024-01-18:
Hi @TheHawk1337,
As for Health mode, there is no way to do it through the standard climate component so you will need to use a separate switch like this:
```
switch:
  - platform: template
    id: climate_health_mode
    name: Haier AC health mode
    icon: mdi:leaf
    restore_mode: RESTORE_DEFAULT_OFF
    lambda: |-
      return id(<your_climate_id>).get_health_mode();
    turn_on_action:
      climate.haier.health_on: <your_climate_id>
    turn_off_action:
      climate.haier.health_off: <your_climate_id>
```
And 10C frost protection has been implemented but it has not been released yet you will need to wait until the next ESPHome release.  It is implemented as AWAY preset. Keep in mind you can enable it only if Heat mode is on.

paveldn, 2024-01-25:
No reply, suppose issue fixed

closed 2024-01-25 by paveldn