paveldn/haier-esphome #38: Smartair2 worked for a few hours and stopped working.

iku777, 2024-01-24:
Hi.

I have installed a Wemos D1 mini yesterday and it worked. Yesterday night turned it off and it seems this morning it's not responding.
It shows as off in HA and I get:
"08:44:31][I][haier.climate:095]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[08:44:37][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[08:44:39][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[08:44:41][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[08:44:41][I][haier.climate:095]: Answer timeout for command 61, phase SENDING_INIT_1
[08:44:41][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[08:44:43][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[08:44:45][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01".

I have tried with another Wemos D1 mini with same result.

Here is my config:

esphome:
  name: wemos-d1-living-2
  friendly_name: Wemos D1-Living 2

esp8266:
  board: d1_mini

# Enable Home Assistant API
api:
  encryption:
    key: <Removed>
  services:
    - service: turn_on
      then:
      - climate.haier.power_on: haier_ac
    - service: turn_off
      then:
      - climate.haier.power_off: haier_ac
      
ota:
  password: <Removed>

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Wemos-D1-Living-2"
    password: "5Jobh1Qp4rhx"

captive_portal:

#Mini

substitutions:
  device_name: Haier AC
  device_id: haier_climate
  uart_id: ac_port
  send_wifi: "true"

uart:
  - id: ${uart_id}
    baud_rate: 9600
    # You can also use pins 15 (TX) and 13 (RX)
    tx_pin: 1
    rx_pin: 3

logger:
  level: DEBUG
  baud_rate: 0

web_server:

external_components:
  - source: github://esphome/esphome@dev
    components: [ haier ]

climate:
  - platform: haier
    id: haier_ac
    protocol: smartAir2
    name: Haier AC 
    uart_id: ac_port
    wifi_signal: true           # Optional, default true, enables WiFI signal transmission from ESP to AC
    display: true               # Optional, default true, can be used to turn off LED display
    answer_timeout: 200ms       # Optional, request answer timeout, can be used to increase the timeout
                                # for some ACs that have longer answer delays
    visual:                     # Optional, you can use it to limit min and max temperatures in UI (not working for remote!)
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:            # Optional, can be used to disable some modes if you don't need them
    - 'OFF'
    #- AUTO
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    supported_presets:          # Optional, can be used to disable some presets if your AC does not support it
      - BOOST
      - COMFORT
    supported_swing_modes:      # Optional, can be used to disable some swing modes if your AC does not support it
    - 'OFF'
    - VERTICAL
    - HORIZONTAL
    - BOTH

Could you please help in understanding if there is something wrong in my config?

paveldn, 2024-01-24:
Hi @iku777 
Your AC just stopped answering at all. Can you try to disconnect your AC from power for 30 seconds and power it back? It will cause the reboot of the internal board. Let me know if it helped.

iku777, 2024-01-24:
Hi @paveldn 
Thanks for quick reply and sorry for not expressing myself properly.
The AC unit itself works, the issue is my Wemos D1 cannot control the AC unit anymore.
It is still online, I still see it in ESPHome and Home Assistant, but when I change the status of the AC unit through HA or ESPHome from "Off" to "Heat", nothing happens. It reverts back to Off. I looked at the log in ESPHome and found this error, which I thought could help understand if there is something wrong in my config. It worked yesterday but doesnt anymore.

08:44:31][I][haier.climate:095]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[08:44:37][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[08:44:39][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[08:44:41][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[08:44:41][I][haier.climate:095]: Answer timeout for command 61, phase SENDING_INIT_1
[08:44:41][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[08:44:43][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[08:44:45][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01".

paveldn, 2024-01-24:
@iku777 
No, it looks like I don't explain it properly. Your AC in Home Assistant is online because your Wemos is working but it can't communicate with AC's control board. From logs I see that the board not answering. Requests from Wemos' side are OK. There is a couple of scenarios but I would say the biggest chance is that the control board somehow got into state that it is not listening to ESP. So unpowering and powering the AC back should restart the AC's board and restore communication. Just try it.

iku777, 2024-01-25:
Thanks for the guidance that the issue was on communication between the Wemos and AC. I disconnected the Wemos, redid the USB wires and it seems to be working fine since then. Didn't know if the problem was in the config.,

closed 2024-01-25 by iku777