paveldn/haier-esphome #39: Answer handler error, msg=01, answ=03, err=7

sunshine-hass, 2024-02-04:
The air conditioner is functioning but cannot be controlled

INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/55.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected log level for Haier protocol: DEBUG
INFO Starting log output from 192.168.5.200 using esphome API
INFO Successfully connected to haier @ 192.168.5.200 in 0.007s
INFO Successful handshake with haier @ 192.168.5.200 in 0.094s
[16:51:26][I][app:102]: ESPHome version 2023.12.9 compiled on Feb  4 2024, 12:21:24
[16:51:26][C][logger:439]: Logger:
[16:51:26][C][logger:440]:   Level: DEBUG
[16:51:26][C][logger:441]:   Log Baud Rate: 115200
[16:51:26][C][logger:443]:   Hardware UART: UART0
[16:51:26][C][uart.arduino_esp32:137]: UART Bus 1:
[16:51:26][C][uart.arduino_esp32:138]:   TX Pin: GPIO17
[16:51:26][C][uart.arduino_esp32:139]:   RX Pin: GPIO5
[16:51:26][C][uart.arduino_esp32:141]:   RX Buffer Size: 256
[16:51:26][C][uart.arduino_esp32:143]:   Baud Rate: 9600 baud
[16:51:26][C][uart.arduino_esp32:144]:   Data Bits: 8
[16:51:26][C][uart.arduino_esp32:145]:   Parity: NONE
[16:51:26][C][uart.arduino_esp32:146]:   Stop bits: 1
[16:51:26][C][haier.climate:239]: Haier Climate 'Haier AC'
[16:51:26][C][haier.climate:240]:   Device communication status: established
[16:51:26][C][haier.climate:298]:   Protocol version: hOn
[16:51:26][C][haier.climate:299]:   Control method: 1
[16:51:26][C][haier.climate:301]:   Device protocol version: E++2.18
[16:51:26][C][haier.climate:302]:   Device software version: 20220902
[16:51:27][C][haier.climate:303]:   Device hardware version: 00000000
[16:51:27][C][haier.climate:304]:   Device name: U-BAC
[16:51:27][C][haier.climate:310]:   Device features: crc
[16:51:27][C][haier.climate:311]:   Active alarms: 00 00 00 00 00 00 00 00
[16:51:27][C][ethernet:222]: Ethernet:
[16:51:27][C][ethernet:354]:   IP Address: 192.168.5.200
[16:51:27][C][ethernet:355]:   Hostname: 'haier'
[16:51:27][C][ethernet:356]:   Subnet: 255.255.255.0
[16:51:27][C][ethernet:357]:   Gateway: 192.168.5.2
[16:51:27][C][ethernet:362]:   DNS1: 192.168.5.2
[16:51:27][C][ethernet:363]:   DNS2: 0.0.0.0
[16:51:27][C][ethernet:383]:   MAC Address: A0:A3:B3:23:B0:7B
[16:51:27][C][ethernet:388]:   Is Full Duplex: YES
[16:51:27][C][ethernet:393]:   Link Speed: 100
[16:51:27][C][ethernet:225]:   Power Pin: 16
[16:51:27][C][ethernet:227]:   MDC Pin: 23
[16:51:27][C][ethernet:228]:   MDIO Pin: 18
[16:51:27][C][ethernet:229]:   Type: LAN8720
[16:51:27][C][ethernet:230]:   PHY addr: 1
[16:51:27][C][mdns:115]: mDNS:
[16:51:27][C][mdns:116]:   Hostname: haier
[16:51:27][C][ota:097]: Over-The-Air Updates:
[16:51:27][C][ota:098]:   Address: haier.local:3232
[16:51:27][C][ota:101]:   Using Password.
[16:51:27][C][api:139]: API Server:
[16:51:27][C][api:140]:   Address: haier.local:6053
[16:51:27][C][api:142]:   Using noise encryption: YES
[16:51:30][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:51:30][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 07 02 47 00 00 03 14 00 00 00 00 00 00 00 00 07 01 23 2A 00 55 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[16:51:30][D][haier.climate:775]: HVAC Mode = 0x2
[16:51:30][D][haier.climate:776]: Fan speed Status = 0x7
[16:51:30][D][haier.climate:777]: Horizontal Swing Status = 0x0
[16:51:30][D][haier.climate:778]: Vertical Swing Status = 0x2
[16:51:30][D][haier.climate:779]: Set Point Status = 0x7
[16:51:35][D][climate:011]: 'Haier AC' - Setting
[16:51:35][D][climate:015]:   Mode: FAN_ONLY
[16:51:35][D][Control:327]: Control call
[16:51:35][W][haier.climate:333]: New settings come faster then processed!
[16:51:35][I][haier.climate:396]: Sending control packet, queue size 1
[16:51:35][D][haier.protocol:019]: Sending frame: type 01, data: 60 01 08 06 C4 00 02 83 14 00 00 00
[16:51:35][D][haier.protocol:019]: Frame found: type 03, data: 00 01
[16:51:35][W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
[16:51:37][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:51:37][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 07 02 47 00 00 03 14 00 00 00 00 00 00 00 00 07 01 23 2A 00 55 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[16:51:37][D][haier.climate:775]: HVAC Mode = 0x2
[16:51:37][D][haier.climate:776]: Fan speed Status = 0x7
[16:51:37][D][haier.climate:777]: Horizontal Swing Status = 0x0
[16:51:37][D][haier.climate:778]: Vertical Swing Status = 0x2
[16:51:37][D][haier.climate:779]: Set Point Status = 0x7
[16:51:38][D][climate:011]: 'Haier AC' - Setting
[16:51:38][D][climate:040]:   Target Temperature: 27.00
[16:51:38][D][Control:327]: Control call
[16:51:38][W][haier.climate:333]: New settings come faster then processed!
[16:51:38][I][haier.climate:396]: Sending control packet, queue size 1
[16:51:38][D][haier.protocol:019]: Sending frame: type 01, data: 60 01 08 06 C4 00 02 83 14 00 00 00
[16:51:38][D][haier.protocol:019]: Frame found: type 03, data: 00 01
[16:51:38][W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
[16:51:42][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:51:42][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 07 02 47 00 00 03 14 00 00 00 00 00 00 00 00 07 01 23 2A 00 55 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[16:51:42][D][haier.climate:775]: HVAC Mode = 0x2
[16:51:42][D][haier.climate:776]: Fan speed Status = 0x7
[16:51:42][D][haier.climate:777]: Horizontal Swing Status = 0x0
[16:51:42][D][haier.climate:778]: Vertical Swing Status = 0x2
[16:51:42][D][haier.climate:779]: Set Point Status = 0x7
[16:51:47][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:51:47][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 07 02 47 00 00 03 14 00 00 00 00 00 00 00 00 07 01 23 2A 00 55 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[16:51:47][D][haier.climate:775]: HVAC Mode = 0x2
[16:51:47][D][haier.climate:776]: Fan speed Status = 0x7
[16:51:47][D][haier.climate:777]: Horizontal Swing Status = 0x0
[16:51:47][D][haier.climate:778]: Vertical Swing Status = 0x2
[16:51:47][D][haier.climate:779]: Set Point Status = 0x7

uart:
  tx_pin: 17
  rx_pin: 5
  baud_rate: 9600
  id: ac_port

climate:
  - platform: haier
    id: haier_ac
    protocol: hOn
    name: Haier AC
    uart_id: ac_port
    # wifi_signal: true
    beeper: true
    display: true
    outdoor_temperature:
      name: Haier AC outdoor temperature
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:
    - 'OFF'
    - HEAT_COOL
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    supported_swing_modes:
    - 'OFF'
    - VERTICAL
    - HORIZONTAL
    - BOTH

sunshine-hass, 2024-02-04:
Hello!
Is it because the protocol is not supported or because my code configuration is wrong?

sunshine-hass, 2024-02-07:
Hello!
Thank you very much for your reply.
After adding the option, you can use most functions. However, after using the FAN_ONLY mode, Answer handler error, msg=01, answ=03, err=7 appears. The module must be restarted before it can be controlled normally. Now check whether there are any configuration problems.

paveldn, 2024-02-07:
@sunshine-hass ,
I'm glad it helped. 
Usually, answer 03 from AC means error. Keep in mind that this way of controlling AC using SET_SINGLE_PARAMETER is not very common. So not everything is 100% tested. Can you make logs of the situation for me?

sunshine-hass, 2024-02-08:
I need to provide useful information

sunshine-hass, 2024-02-08:
[logs_haier2_logs.txt](https://github.com/paveldn/haier-esphome/files/14211545/logs_haier2_logs.txt)
control_packet_size:18
When set to the above parameters, the indoor temperature will be displayed
When using the Haier Smart APP, the indoor temperature will be displayed.
Can the error be fixed based on the above log?

paveldn, 2024-02-09:
Thanks. Now more or less clear what is happening. Your AC is answering 03 (which most likely indicates that it is not a supported feature) to command 0x19 which is a quiet mode setting. Quiet mode should be disabled in "Fan only" mode. Not clear why this command was sent multiple times it should be only one time. Was this just a one-time attempt or did you try to do it multiple times? I see every time after "Answer handler error" "Control call", messages that it indicates that somebody outside of the component causing some changes but I don't know if it is you, Home Assistant, or ESPHome. Will see what I can do. I will come to you when I will have something to test.

sunshine-hass, 2024-02-09:
As long as there is error support, it cannot be controlled normally. The multiple errors you see are because when the error was triggered for the first time, I tried to use other functions, which caused multiple errors.

paveldn, 2024-02-10:
Can you try this branch: https://github.com/paveldn/haier-esphome/tree/no_silent_test ?
you will need to add the following config to the haier climate section:
```
  supported_presets:
    - BOOST
    - SLEEP
```

Let me know if it helps. It will remove the ECO preset from your climate (which is anyway not supported) and suppress the command that causes problems.

sunshine-hass, 2024-02-12:
Thank you very much for your reply.
1. Now this configuration function is basically normal, but the following prompt appears whether it needs to be repaired.
[17:15:40][W][component:214]: Component haier.climate took a long time for an operation (0.05 s).
[17:15:40][W][component:215]: Components should block for at most 20-30ms.
2. However, setting the temperature when using FAN_ONLY mode will still cause Answer handler error, msg=01, answ=03, err=7. The temperature cannot be set on the remote control when using FAN_ONLY. Is this possible?
3. The wind speed status is out of sync with when using the remote control. Neither low nor high can respond to the status controlled by the remote control. Can it be repaired?
fan_modes:
   -auto
   -low
   -medium
   -high
The configuration is now as follows:
external_components:
  - source:
      type: git
      url: https://github.com/paveldn/haier-esphome
      ref: no_silent_test
    components: [haier]
climate:
  - platform: haier
    id: haier_ac
    protocol: hOn
    name: Haier AC999 
    uart_id: ac_port
    # wifi_signal: true          # Optional, default true, enables WiFI signal transmission from ESP to AC
    beeper: true            # false Optional, default true, disables beep on commands from ESP
    display: true              # Optional, default true, can be used to turn off LED display
    control_method: SET_SINGLE_PARAMETER
    # control_packet_size: 18
    # control_method: SET_GROUP_PARAMETERS  
    answer_timeout: 200ms      # Optional, request answer timeout, can be used to increase the timeout
                                # for some ACs that have longer answer delays
    visual:                     # Optional, you can use it to limit min and max temperatures in UI (not working for remote!)
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:            # Optional, can be used to disable some modes if you don't need them
    - 'OFF'
    - HEAT_COOL
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    supported_presets:          # Optional, can be used to disable some presets if your AC does not support it
    # - BOOST
    # - SLEEP
    supported_swing_modes:      # Optional, can be used to disable some swing modes if your AC does not support it
    # - 'OFF'
    # - VERTICAL
    # - HORIZONTAL
    # - BOTH

paveldn, 2024-02-12:
Hi @sunshine-hass 

> Component haier.climate took a long time for an operation (0.05 s).

Sorry, can't fix this. I optimized my component as much as possible. My investigation shows that most of the delay is coming from 2 things: 
1. logging system - the more you log the slower your component. So I would suggest setting the logging level to WARN once you finish configuring your AC. It should ease the problem but not fix it completely.
2. parent component function calls. When there is a change of state for AC my component calls the parent component (which is Climate) to update status. That is causing all updates of web UI, publishing state to the home assistant, etc. This call is very slow and most likely that is the reason of warn messages you see. I can't do anything here it is not just a problem of my component but most of the components.

> However, setting the temperature when using FAN_ONLY mode will still cause Answer handler error, msg=01, answ=03, err=7. The temperature cannot be set on the remote control when using FAN_ONLY. Is this possible?

This one is easy to fix, will do it later today.

> The wind speed status is out of sync with when using the remote control. Neither low nor high can respond to the status controlled by the remote control. Can it be repaired?

That one is strange. Maybe your AC not updating the status correctly after using the remote. I will need logs from you. Just make sure you have at least 20 seconds of logs before you use the remote and 20 seconds after. And let me know which mode was before the change and which after.

paveldn, 2024-02-12:
Please, check the latest commit on https://github.com/paveldn/haier-esphome/tree/no_silent_test I removed setting temp in fan only mode.

paveldn, 2024-03-04:
No feedback I assume the issue is solved. The solution is merged in the dev branch and will be included in next pull request to ESPHome

d22cva, 2024-06-05:
Hello! I have the same problem as in the first post. When I select any action I get "Answer handler error, msg=01, answ=03, err=7". The module from the air conditioner kit is as pictured in the description on esphome.io in the "Haier Climate" section. I use a Wemos D1 mini board. When loading the firmware for hOn, I see the status of the compressor when I control the remote control. I conclude that the board was assembled correctly. Some parameters are displayed, but I'm not sure what's correct. I tried flashing the smartair2 protocol, but didn’t see any parameters.

On hOn firmware:
- I tried to swap Rx and Tx - no luck.
- I tried to add control_method: SET_SINGLE_PARAMETER in the config - no success.
- I tried adding control_packet_size: 18 - no success.
- I tried firmware for Wemos D1 mini from the DEV and EXPEREMETAL branches - no success.

Judging by the posts above, the problem has been resolved. Maybe I'm doing something wrong? I would appreciate your help!
I am writing in a language that is not my native one! Thanks for understanding!

d22cva, 2024-06-05:
> Никаких отзывов, думаю, проблема решена. Решение добавлено в ветку разработки и будет включено в следующий запрос на включение в ESPHome.

Hello! I have the same problem as in the first post. When I select any action I get "Answer handler error, msg=01, answ=03, err=7". The module from the air conditioner kit is as pictured in the description on esphome.io in the "Haier Climate" section. I use a Wemos D1 mini board. When loading the firmware for hOn, I see the status of the compressor when I control the remote control. I conclude that the board was assembled correctly. Some parameters are displayed, but I'm not sure what's correct. I tried flashing the smartair2 protocol, but didn’t see any parameters.

On hOn firmware:

I tried to swap Rx and Tx - no success.
I tried to add control_method: SET_SINGLE_PARAMETER in the config - no success.
I tried adding control_packet_size: 18 - no success.
I tried firmware for Wemos D1 mini from the DEV and EXPEREMETAL branches - no success.
Judging by the posts above, the problem has been resolved. Maybe I'm doing something wrong? I would appreciate your help!
I am writing in a language that is not my native one! Thanks for understanding!

paveldn, 2024-06-05:
Hi @d22cva,

Я говорю по русски так что можете использовать его. Чтобы помочь вам мне для начала нужен лог вашего устройства. При этом попробуйте что-то поменять из веб интерфейса (или Home assistant). Какая у вас модель кондиционера?

d22cva, 2024-06-05:
> Hi @d22cva,
> 
> Я говорю по русски так что можете использовать его. Чтобы помочь вам мне для начала нужен лог вашего устройства. При этом попробуйте что-то поменять из веб интерфейса (или Home assistant). Какая у вас модель кондиционера?
 
Замечательно! Модель кондиционера Casarte  внутренний блок CAS35MW1/R3-W, наружный 1U35MW1/R3. Родной Wi-Fi модуль ESP32-for-Haier_v1.1 работает через приложение EVO. Я думаю это протокол hOn. Родной модуль не стал перешивать так как кондиционер на гарантии. Пробовал протокол smartair2, но на нём по моему мнению вообще связи нет с кондиционером. На протоколе hOn получаю хоть какие-то значения. Было замечено если с пульта включить, например, охлаждение , то параметр "Haier AC Compressor Status" изменяет своё значение на ON и наоборот если выключить с пульта. На действия в веб-интерфейсе не реагирует, вернее в логе веб-интерфейса получаю сообщение "Answer handler error, msg=01, answ=03, err=7"  .Подскажи как правильно снять лог? Достаточно будет скопировать примитивно из веб-интерфейса esphome?

d22cva, 2024-06-05:
Прилагаю логи, которые смог скопировать с веб-интерфейса. Во всех случаях параметр control_method был убран из конфига. Если как-то поможет значение Compressor Status соответствовало действительности только при control_packet_size по умолчанию. В остальных случаях этот параметр не менялся. Некоторые параметры что-то отображали, но они были далеко от действительности.

[control_packet_size_12_Action_in_Web.txt](https://github.com/user-attachments/files/15594558/control_packet_size_12_Action_in_Web.txt)
[control_packet_size_12_Remote_action_OFF.txt](https://github.com/user-attachments/files/15594559/control_packet_size_12_Remote_action_OFF.txt)
[control_packet_size_12_Start.txt](https://github.com/user-attachments/files/15594560/control_packet_size_12_Start.txt)
[control_packet_size_14_Action_in_Web.txt](https://github.com/user-attachments/files/15594561/control_packet_size_14_Action_in_Web.txt)
[control_packet_size_14_Remote_action_ON.txt](https://github.com/user-attachments/files/15594562/control_packet_size_14_Remote_action_ON.txt)
[control_packet_size_14_Start.txt](https://github.com/user-attachments/files/15594563/control_packet_size_14_Start.txt)
[default_parametrs_Action.txt](https://github.com/user-attachments/files/15594564/default_parametrs_Action.txt)
[default_parametrs_Remote_action_ON.txt](https://github.com/user-attachments/files/15594565/default_parametrs_Remote_action_ON.txt)
[default_parametrs_Start.txt](https://github.com/user-attachments/files/15594566/default_parametrs_Start.txt)

paveldn, 2024-06-05:
Это что-то совсем новое. Я таких пакетов еще не видел. Протокол тот же что и hOn но пакет статуса совсем другой. Тут нужно копать с самого начала. Боюсь что простыми подстройками тут не обойдешся. 
Тут все зависит насколько вы полны решимости добавить его в Home Assistant. Потому что тут нужно єкспериментировать и анализировать, анализировать и єксперементировать. Анализировать я могу и сам но вот делать эксперименты прийдется вам.

d22cva, 2024-06-06:
Готов попытаться!

paveldn, 2024-06-06:
> Готов попытаться!

Возможно все не так плохо как я думал вгачале. Похоже паект статуса просто имеет вначале странный блрк состоящий почти полностью из нулей. Если его пропустить по крайней мере чтение состояния долэно работать. А вот что делать с управлкнием прийжктся разбираться. Я немного приболел поэтому сегодня не могу но постараюсь завтра написать инструкцию как подготовить ESP к экспериментам. А пока можете показать свою конфигурацию? Только часть насчет кондиционера.

d22cva, 2024-06-06:
Поправляйтесь! Буду пытаться помочь чем смогу. Единственное могу этим заниматься не всегда из-за своей работы.
Мой конфиг по большому счёту аналогичен вашему из репозитория master. Оставил прошивку на которой меняется статус Compressor Status. Ещё хочу отметить, может кому-то поможет, что на прошивках из ветки dev и experemental почему-то зависает esp при включенных компонентах self_cleaning_button и steri_cleaning_button, т.е. в веб-интерфейсе отображается 2-5 параметра и лог не идёт. Веб-интерфейс не отзывчив и будто зависает. Стоит отключить именно self_cleaning_button и steri_cleaning_button как веб-интерфейс начинает работать как положено. На прошивке из ветки master на такое поведение я наткнулся только 1 раз и для лучшей работы на всякий случай отключил эти компоненты. 

**haier_hon.yaml**
```
climate:
  - platform: haier
    id: ${device_id}
    protocol: hOn
    name: ${device_name}
    uart_id: ${uart_id}
    wifi_signal: ${send_wifi}   # Optional, default true, enables WiFI signal transmission from ESP to AC
    #answer_timeout: 200ms      # TEST. Default 200ms
    #control_packet_size: 14    # TEST. Default 10
    #control_method: SET_SINGLE_PARAMETER           # TEST. Default SET_GROUP_PARAMETERS 
    beeper: true                # Optional, default true, disables beep on commands from ESP
    visual:                     # Optional, you can use it to limit min and max temperatures in UI (not working for remote!)
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 
        target_temperature: 1
        current_temperature: 0.5
    supported_modes:            # Optional, can be used to disable some modes if you don't need them
      - 'OFF'                     # always available
      - HEAT_COOL                 # always available
      - COOL
      - HEAT
      - DRY
      - FAN_ONLY
    supported_swing_modes:      # Optional, can be used to disable some swing modes if your AC does not support it
      - 'OFF'
      - VERTICAL
      - HORIZONTAL
      - BOTH
    supported_presets:          # Optional, can be used to disable some presets if your AC does not support it
      - ECO
      - BOOST
      - SLEEP
    # Next two triggers calling Home Assistant services to notify user about alarms triggered by AC
    # this functionality is disabled by default. To enable it check "Allow the device to make Home Assistant service calls"
    # checkbox in Home Assistant's device settings
    on_alarm_start:
      then:
        - homeassistant.service:
            service: logbook.log
            data:
              domain: climate
              name: ${device_name}
            data_template:
              message: "Alarm activated ({{ alarm_code }}): {{alarm_message}}" 
            variables:
              alarm_message: !lambda "return message;"
              alarm_code: !lambda "return code;"
        - homeassistant.service:
            service: notify.persistent_notification
            data:
              title: "${device_name}: alarm activated"
            data_template:
              message: "Code: {{ alarm_code }}, message: \"{{ alarm_message }}\""
            variables:
              alarm_message: !lambda "return message;"
              alarm_code: !lambda "return code;"
    on_alarm_end:
      then:
        - homeassistant.service:
            service: logbook.log
            data:
              domain: climate
              name: ${device_name}
            data_template:
              message: "Alarm deactivated ({{ alarm_code }}): {{alarm_message}}" 
            variables:
              alarm_message: !lambda "return message;"
              alarm_code: !lambda "return code;"

```

paveldn, 2024-06-07:
ОК. Давайте начнем. Для начала мне нужно чтобы вы могли посылать любые команды кондиционеру. Для этого в секцию api вашей конфигурации нужно добавить такой сервис (если его еще нет, если есть пропускаете добавление):
```
api:
  services:
    - service: send_custom_message
      variables:
        message_type: int
        message_data: int[]
      then:
        - lambda: |-
            if ((message_type < 0) || (message_type > 255)) {
              ESP_LOGE("send_custom_command", "Wrong custom message type (should be byte)");
              return;  
            }
            if (message_data.size() == 0) {
              id(${device_id}).send_custom_command(haier_protocol::HaierMessage((haier_protocol::FrameType) message_type));
            } else {
              size_t sz = message_data.size();
              std::unique_ptr<uint8_t[]> msg_buffer(new uint8_t [sz]); 
              for (unsigned int i = 0; i < sz; i++)
                msg_buffer[i] = message_data[i];
              id(${device_id}).send_custom_command(haier_protocol::HaierMessage((haier_protocol::FrameType) message_type, msg_buffer.get(), sz));
            }
```
После этого у вас в Home Assistant должен появится сервис: <иля_кондиционера>_send_custom_message:
[image]
Для пробы пошлите команду:
```
message_type: 97
message_data: [0x00, 0x07]
```
И пришлите мне ответ.

d22cva, 2024-06-07:
Ответ:
```
01:55:31	[I]	[haier.climate:955]	
Fan speed Status = 0x0
01:55:31	[I]	[haier.climate:956]	
Horizontal Swing Status = 0x0
01:55:31	[I]	[haier.climate:957]	
Vertical Swing Status = 0x4
01:55:32	[I]	[haier.climate:958]	
Set Point Status = 0x20
01:55:32	[D]	[haier.protocol:019]	
Sending frame: type 61, data: 00 07
01:55:32	[D]	[haier.protocol:019]	
Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 32 30 31 31 32 31 30 31 F1 00 00 32 30 30 38 32 30 30 31 01 55 2D 41 43 00 00 00 00 00 04
01:55:32	[W]	[haier.protocol:013]	
Answer handler error, msg=61, answ=62, err=2
01:55:36	[D]	[haier.protocol:019]	
Sending frame: type 01, data: 4D FE
01:55:36	[D]	[haier.protocol:019]	
Frame found: type 02, data: 7D 01 20 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 08 06 C2 00 06 00 00 00 00 00 35 3F 54 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5C 55 54 00 54 00 00 00 00 00 01 E0
01:55:36	[D]	[climate:396]	
'Casarte AC' - Sending state:
01:55:36	[D]	[climate:399]	
  Mode: OFF
01:55:36	[D]	[climate:410]	
  Preset: NONE
01:55:36	[D]	[climate:416]	
  Swing Mode: OFF
01:55:36	[D]	[climate:419]	
  Current Temperature: 0.00°C
01:55:36	[D]	[climate:425]	
  Target Temperature: 48.00°C
01:55:36	[I]	[haier.climate:951]	
HVAC values changed
01:55:36	[I]	[haier.climate:954]	
HVAC Mode = 0x0
```

paveldn, 2024-06-07:
ОК. Работает!
Теперь проверим мою теорию:
```
message_type: 1
message_data: [0x60, 0x01, 0x08, 0x06, 0xC2, 0x00, 0x06, 0x01, 0x00, 0x00, 0x00, 0x00, 0x35, 0x3F, 0x54, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]
```
Мне нужен ответ от кондиционера. И если все ОК кондиционер должен включится

d22cva, 2024-06-07:
> Мне нужен ответ от кондиционера. И если все ОК кондиционер должен включится

Он включился!!!!!!!

Ответ:
```
02:35:19	[I]	[haier.climate:956]	
Horizontal Swing Status = 0x0
02:35:19	[I]	[haier.climate:957]	
Vertical Swing Status = 0x4
02:35:19	[I]	[haier.climate:958]	
Set Point Status = 0x20
02:35:20	[D]	[haier.protocol:019]	
Sending frame: type 01, data: 60 01 08 06 C2 00 06 01 00 00 00 00 35 3F 54 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
02:35:20	[D]	[haier.protocol:019]	
Frame found: type 02, data: 6D 01 20 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 08 06 C2 00 06 01 00 00 00 00 35 3C 50 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
02:35:20	[D]	[climate:396]	
'Casarte AC' - Sending state:
02:35:20	[D]	[climate:399]	
  Mode: OFF
02:35:20	[D]	[climate:410]	
  Preset: NONE
02:35:20	[D]	[climate:416]	
  Swing Mode: OFF
02:35:20	[D]	[climate:419]	
  Current Temperature: 0.00°C
02:35:20	[D]	[climate:425]	
  Target Temperature: 48.00°C
02:35:20	[I]	[haier.climate:951]	
HVAC values changed
02:35:20	[I]	[haier.climate:954]	
HVAC Mode = 0x0
02:35:20	[I]	[haier.climate:955]	
Fan speed Status = 0x0
02:35:20	[I]	[haier.climate:956]	
Horizontal Swing Status = 0x0
02:35:20	[I]	[haier.climate:957]	
```

paveldn, 2024-06-07:
Отлично. Тогда в общем-то все понятно нужно просто игнорировать первые 38 байт пакета. А все остальное как для стандартного протокола. Я постараюсь за выходные сделать версию которая должна работать с вашим кондиционером. Я напишу когда будет готово для тестирования.

paveldn, 2024-06-08:
@d22cva,
Попробуйте последнюю версию на ветке experimental
Нужно добавить параметр:
```
  status_message_header_size: 38
```
 Дайте знать или заработало

d22cva, 2024-06-09:
Только из выключенного состояния кондиционер включается в любой режим(обогрев, холод, вентиляция) и после этого на любые изменения режимов получаю Answer handler error, msg=01, answ=03, err=7. Выключить возможно только с пульта и после этого из веб-интерфейса невозможно включить вновь до тех пор пока не перезапустишь D1 mini.
Есть положительные моменты. Стали правильно отображаться параметры:
- текущая температура в помещении (совпадает с значением на дисплее кондиционера)
- статус текущей работы выключен/обогрев/охлаждение/осушение
- температура на улице

Вот что получаю когда в веб-интерфейсе режиме "Обогрев" выбираю режим OFF. 

```
17:56:35	[D]	[select:015]	
'Casarte AC airflow vertical': Sending state Up (index 2)
17:56:35	[D]	[haier.protocol:019]	
Sending frame: type FC, data: <empty>
17:56:35	[D]	[haier.protocol:019]	
Frame found: type FD, data: 00 00 00 00 00 00
17:56:35	[D]	[haier.protocol:019]	
Answer delay 35ms
17:56:35	[D]	[select:015]	
'Casarte AC airflow horizontal': Sending state Center (index 2)
17:56:37	[D]	[haier.climate:119]	
WiFi signal is: -51dBm => 60%
17:56:37	[D]	[haier.protocol:019]	
Sending frame: type F7, data: 00 00 00 3C
17:56:37	[D]	[haier.protocol:019]	
Frame found: type 05, data: <empty>
17:56:37	[D]	[haier.protocol:019]	
Answer delay 34ms
17:56:39	[D]	[haier.protocol:019]	
Sending frame: type 01, data: 4D FE
17:56:39	[D]	[haier.protocol:019]	
Frame found: type 02, data: 7D 01 20 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 04 85 00 02 01 00 00 00 00 30 3E 55 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 59 57 55 00 55 00 00 00 00 10 01 E0
17:56:39	[D]	[haier.protocol:019]	
Answer delay 138ms
17:56:39	[D]	[haier.climate:991]	
HVAC Mode = 0x4
17:56:40	[D]	[haier.climate:992]	
Fan speed Status = 0x5
17:56:40	[D]	[haier.climate:993]	
Horizontal Swing Status = 0x0
17:56:40	[D]	[haier.climate:994]	
Vertical Swing Status = 0x4
17:56:40	[D]	[haier.climate:995]	
Set Point Status = 0x0
17:56:40	[D]	[select:015]	
'Casarte AC airflow vertical': Sending state Up (index 2)
17:56:40	[D]	[select:015]	
'Casarte AC airflow horizontal': Sending state Center (index 2)
17:56:41	[D]	[climate:011]	
'Casarte AC' - Setting
17:56:41	[D]	[climate:015]	
  Mode: OFF
17:56:41	[D]	[Control:333]	
Control call
17:56:41	[W]	[haier.climate:339]	
New settings come faster then processed!
17:56:41	[I]	[haier.climate:410]	
Sending control packet, queue size 1
17:56:41	[D]	[haier.protocol:019]	
Sending frame: type 01, data: 60 01 08 64 00 00 02 00 00 00 00 00
17:56:41	[D]	[haier.protocol:019]	
Frame found: type 03, data: 00 00
17:56:41	[D]	[haier.protocol:019]	
Answer delay 51ms
17:56:41	[W]	[haier.protocol:013]	
Answer handler error, msg=01, answ=03, err=7
17:56:44	[D]	[haier.protocol:019]	
Sending frame: type 01, data: 4D 01
17:56:44	[D]	[haier.protocol:019]	
Frame found: type 02, data: 6D 01 20 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 04 85 00 02 01 00 00 00 00 30 3E 55 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
17:56:44	[D]	[haier.protocol:019]	
Answer delay 116ms
17:56:44	[D]	[haier.climate:991]	
HVAC Mode = 0x4
17:56:44	[D]	[haier.climate:992]	
Fan speed Status = 0x5
17:56:45	[D]	[haier.climate:993]	
Horizontal Swing Status = 0x0
17:56:45	[D]	[haier.climate:994]	
Vertical Swing Status = 0x4
17:56:45	[D]	[haier.climate:995]	
Set Point Status = 0x0
17:56:45	[D]	[select:015]	
'Casarte AC airflow vertical': Sending state Up (index 2)
17:56:45	[D]	[select:015]	
'Casarte AC airflow horizontal': Sending state Center (index 2)
```

[image]

Моя конфигурация:

d22cva, 2024-06-09:
Подскажите как можно собирать лог с веб-интерфейса в файл чтоб при тестировании не отвлекаться на копирование постоянно бегущей информации в логе?

paveldn, 2024-06-09:
Прошу прощения, забыл упомянуть нужно еще control_packet_size: 12
Про логи напишу вечером, сейчас с телефона.

d22cva, 2024-06-09:
```
status_message_header_size: 38
control_packet_size: 12
```
Добавил параметры и всё равно не удается управлять, но теперь на мой взгляд все параметры читаются корректно. 
[image]

Если перезапускать ESP , то первый раз параметр применяется, т.е. было включено охлаждение , делаю рестарт ESP и теперь выбираю FAN ONLY и этот режим включается, но последующий выбор режима работы дает "Answer handler error, msg=01, answ=03, err=7".

paveldn, 2024-06-09:
ОК. Спасибо. Думаю понял в чем проблема. Нужно кое-что переработать. Я дам знать когда будет готова следующая версия.

Насчет логов. Самій простой способ это из ESPHome Dashboard. Нужно в дашбоарде открыть логи устройства перед началом эксперимента, вот так:
[image]
А затем после окончания можно нажать  Download logs:
[image]
Файл будет содержать весь лог.

paveldn, 2024-06-10:
@d22cva 
Я думаю поправил. Можете проверить?
Настройки те же 
```
status_message_header_size: 38
control_packet_size: 12
```

d22cva, 2024-06-11:
@paveldn , победа! Я в восторге! Включаются все режимы (холод/обогрев/осушение/вентиляция/авто) как в Home Assistant так в веб-интерфейсе. Почти все параметры читаются. Обратил внимание, что значение текущей температуры внутри помещения теперь значение 40, на предыдущих прошивках показывало корректно. Мне неловко просить помочь решить вопрос с значением температуры внутри помещения. В остальном проблем не заметил. Главное, что теперь работает управление режимами! Благодарен за вашу помощь!

paveldn, 2024-06-11:
@d22cva 
Рад что заработало. Насчет текущей температуры я посмотрю видно где-то неудачно смещение сделал.

paveldn, 2024-06-12:
@d22cva ,
Попробуйте, пожалуйста, поставить
```
control_packet_size: 11
```
Это скорее всего порушит некоторые сенсоры но мне важно знать или управление работает. И текущая температура должна заработать

d22cva, 2024-06-12:
@paveldn 
С параметром control_packet_size: 11 текущая температура стала похожа на действительную. Включение/выключение работает. Вы правы, некоторые сенсоры стали показывать не реальные значения. Досконально не было возможности проверить, если необходимо могу проверить всё завтра.

paveldn, 2024-06-12:
@d22cva ,
Да потестируйте или все работает что бы небыло сюрпризов. Покоже у вас не толькотнестандартный размер пакета управления но и нестандартный размер пакета сенсоров. Я внесу изменентя и все должно работать.

d22cva, 2024-06-12:
@paveldn 
> P. S. Кстати ваш кондиционер первый с протоколом Haier который меряет расход энергии из тех что я видел. У всех до этого были всегда 0

Сам был удивлен. Замерял ваттметров и значения близки. 

По поводу нестандартности. В родном приложении EVO помимо показаний текущей температуры было значение текущей влажности. Возможно из-за значений влажности смещено значение температуры.

paveldn, 2024-06-12:
Влажность и сейчас поддерживается, пример:
https://github.com/paveldn/haier-esphome/blob/master/configs/sensor/humidity.yaml
с `control_packet_size: 11` должна работать без проблем

paveldn, 2024-06-12:
Проверьте обновленный код на ветке experimentalб конфигурация должна быть:
```
    status_message_header_size: 38
    control_packet_size: 12
    sensors_packet_size: 23
```
Кстати а управлять влажностью на вашем кондиционере можно? В протоколе есть возможность устанавливать целевую влажность но пока еще не видел кондиционера который бы это поддерживал. Было бы интересно попробовать.

d22cva, 2024-06-13:
```
    status_message_header_size: 38
    control_packet_size: 12
    sensors_packet_size: 23
```
C этими параметрами в логе циклично крутятся :
```
[19:15:19][W][haier.climate:771]: Unexpected message size 74 (expexted >= 75)
[19:15:19][W][haier.climate:169]: Error 4 while parsing Status packet
[19:15:19][W][haier.protocol:013]: Answer handler error, msg=01, answ=02, err=4
```

paveldn, 2024-06-13:
ОК, тогда используйте 
```
sensors_packet_size: 22
```
Дополнительные сенсоры не будут работать (кроме температуры и влажности) но вы сможете проверить управление. А я еще раз посмотрю что я сделал не так.

d22cva, 2024-06-13:
@paveldn 
```
    status_message_header_size: 38
    **control_packet_size: 11**
    sensors_packet_size: 23
```
[image]

При такой конфигурации управление есть. Показания текущей температуры отличаются от значения на дисплее кондиционера на 1.5-2 градуса. Влажность 92% , а этого быть не может.

С конфигом 
```
    status_message_header_size: 38
    **control_packet_size: 11**
    sensors_packet_size: 23
```

Наблюдаю сообщение в логе :
```
[20:03:06][W][haier.climate:771]: Unexpected message size 74 (expexted >= 85)
[20:03:06][W][haier.climate:169]: Error 4 while parsing Status packet
[20:03:06][W][haier.protocol:013]: Answer handler error, msg=01, answ=02, err=4
```
Отклик на включение/выключение режимов только 1 раз, потом реакции нет, только издается звук на полученную команду, но действия нет. Перезагружаю ESP и снова реагирует 1 раз на любое действие. Было замечено , что при загрузке  ESP первые пару секунд корректно отображается текущая температура и влажность, и очень похоже что остальные параметры, но спустя буквально 2-3 секунды почти все сенсоры принимают значение NA или 0.

d22cva, 2024-06-13:
@paveldn 
[logs_casarte_logs.txt](https://github.com/user-attachments/files/15825837/logs_casarte_logs.txt)
22:28 == 24.5 
22:39 == 25 
22:43 == 25.5

Если недостаточно значений , то могу провести эксперимент вновь.

paveldn, 2024-06-13:
@d22cva,
Странно как-то получается. Можете попробовать:
```
    status_message_header_size: 38
    control_packet_size: 10
    sensors_packet_size: 24
```
?

d22cva, 2024-06-13:
@paveldn ,

d22cva, 2024-06-13:
@paveldn ,
```
    status_message_header_size: 38
    control_packet_size: 10
    sensors_packet_size: 24
```
Досконально проверить нету сейчас возможности. Попробовал включить, выключить и мне кажется это то как должно быть. В субботу проверю все режимы и функции.

Результат:
[image]

d22cva, 2024-06-15:
Сегодня проверил всё что было доступно. Всё работает! Все показания сенсоров показываются правильно. Конкретно текущую температуру(в помещении) и влажность сверял с показаниями родного модуля в его приложении EVO и они одинаковые! Так же сверил температуру с обычного кондиционера Haier в другой комнате и всё совпало. 
Приятно было, что теперь есть возможность смотреть энергопотребление. Есть у меня бытовой ваттметр и может кому будет полезна следующая информация. Показания мощности сошлись с ваттметром. Конечно было расхождение, но это в пределах погрешности. Значения потребления тока на 0.2-0.3 А меньше чем показывал ваттметр. В целях статистических данных погрешность особого значения не имеет. Обнаружил особенность температуры с датчика уличного блока. Оказывается данные с него обновляются только когда кондиционер работает и не важно в каком режиме вентиляция или обогрев, после выключения отображаются последние данные. Изначально думал задействовать уличный датчик температуры для статистики погоды, но видимо проще сделать отдельное устройство для этого. 

Строки конфига решили окончательно все проблемы :
```
    status_message_header_size: 38
    control_packet_size: 10
    sensors_packet_size: 24
```

> P. S. Кстати ваш кондиционер первый с протоколом Haier который меряет расход энергии из тех что я видел. У всех до этого были всегда 0

В приложении кондиционера EVO такого не обнаружил и в инструкции тоже. Видимо это задел на будущее или вообще для очень дорогих моделей. Если понадобится такой функционал, то проще это осуществить через автоматизацию Home Assistant.
Приложение EVO оказалось очень неудобным и нет официальной возможности интеграций в системы умного дома. К тому же оно постоянно требует входа в аккаунт! Бред! ESPHome было моё спасение.

Спасибо Вам за терпение! Я благодарен Вам за помощь!

paveldn, 2024-06-16:
@d22cva ,
Спасибо за тестирование. Рад что все заработало. Я тоже начал с того что не смог нормально работать родным приложением (hOn в моем случае) и решил сделать собственное ESPHome решение.  Я подготовлю пулл реквест для ESPHome. Обычно это долгое дело и может занимать месяц-два но после этого вы сможете использовать компонент напрямую с ESPHome не используя мой репозиторий.
Удачи вам с умным домом!

closed 2024-06-17 by paveldn