paveldn/haier-esphome #40: Haier AS25S2SF2FA-2 with M5Stamp C3

Alloc186, 2024-03-26:
HI, I cant make it work. My unit is Haier AS25S2SF2FA-2m I have USB dongle, works OK with SmartAir2 APP. 
I bought M5Stamp C3 board + USB A to USB C adapter. Changed code to tx_pin: 19 rx_pin: 18.

When I put code to ESP Home ota (I am newbie in ESP world), it doesn't work.

Here is my code:

`esphome:
  name: klima
  friendly_name: klima

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

web_server:

external_components:
  - source: github://esphome/esphome@dev
    components: [ haier ]

logger:
  hardware_uart: UART1

uart:
  baud_rate: 9600
  tx_pin: 19
  rx_pin: 18
  id: ac_port  

climate:
  - platform: haier
    id: haier_ac
    protocol: hOn
    name: Haier AC 
    uart_id: ac_port
    wifi_signal: true           # Optional, default true, enables WiFI signal transmission from ESP to AC
    beeper: true                # Optional, default true, disables beep on commands from ESP
    display: true               # Optional, default true, can be used to turn off LED display
    answer_timeout: 200ms      # Optional, request answer timeout, can be used to increase the timeout
                                # for some ACs that have longer answer delays

    visual:                     # Optional, you can use it to limit min and max temperatures in UI (not working for remote!)
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:            # Optional, can be used to disable some modes if you don't need them
      - 'OFF'
      - COOL
      - HEAT
      - DRY
      - FAN_ONLY
    supported_presets:          # Optional, can be used to disable some presets if your AC does not support it
      - AWAY
      - ECO
      - BOOST
      - SLEEP
    supported_swing_modes:      # Optional, can be used to disable some (or all) swing modes if your AC does not support it
      - 'OFF'
      - VERTICAL
      - HORIZONTAL
      - BOTH
    on_alarm_start:
      then:
        - logger.log:
            level: WARN
            format: "Alarm activated. Code: %d. Message: \"%s\""
            args: [ code, message]
    on_alarm_end:
      then:
        - logger.log:
            level: INFO
            format: "Alarm deactivated. Code: %d. Message: \"%s\""
            args: [ code, message]`

[image]

[image]

[image]

Thanks for any help ;/

paveldn, 2024-03-26:
Hi

I am afraid it will not work like this. You can't use a USB from your M5Stamp because Haeir AC doesn't have a real USB. It is just a USB connector that is used for the serial port. And M5Stamp has a real USB. To make it work you will need to solder USB wire to pins yourself.

Alloc186, 2024-03-26:
Ok, that's what I thought, but I was hoping to avoid soldering. So I'll try to get free unused USB cable, cut it and feed it directly to the board.

Alloc186, 2024-04-03:
Hi I just figured it out without soldering! I used M5 Stack [AtomS3U]
(https://shop.m5stack.com/products/atoms3u)
[image]

[image]

Then set GPIO like this:
uart:
  baud_rate: 9600
  tx_pin: 19
  rx_pin: 20
  id: ac_port

paveldn, 2024-04-03:
Oh, cool I didn't know on these boards they connected USB data directly to the GPIO. Thanks for the info.

Alloc186, 2024-04-03:
It even fits into place under original cover .) Thanks for your amazing work!

Alloc186, 2024-04-12:
One more question I have successfully managed to control vertical and horizontal airflow, with select command. But does the device also communicate back? Like if I set the air flow using the position of the slats with the controller, will this setting also be reflected in the Home assistant?

paveldn, 2024-04-12:
You can write a template sensor (or some other component) using public functions of haier climate:
get_vertical_airflow
get_horizontal_airflow 
Should be something like this:
```
text_sensor:
  - platform: template
    name: ${device_name} Vertical airflow
    lambda: |-
      auto vflow = id(${device_id}).get_horizontal_airflow();
      switch(vflow) {
        case esphome::haier::AirflowHorizontalDirection::MAX_LEFT:
            return {"Max left"};
        case esphome::haier::AirflowHorizontalDirection::LEFT:
            return {"Left"};
        case esphome::haier::AirflowHorizontalDirection::CENTER:
            return {"Center"};
        case esphome::haier::AirflowHorizontalDirection::RIGHT:
            return {"Right"};
        case esphome::haier::AirflowHorizontalDirection::MAX_RIGHT:
            return {"Max right"};
        default:
            return {"Unknown"};
      }
    update_interval: 10s
```
And similar for vertical

Alloc186, 2024-04-13:
Hmm, sensor works for setting from ESP but not reporting change made by remote control. 
[image]

paveldn, 2024-04-14:
@Alloc186 I will need logs from you to check. Can you get logs for me with the Debug level? I will need 10 - 15 seconds before you make changes with the remote and 10 - 15 seconds after. And a comment on what you did with the remote.

Alloc186, 2024-04-15:
OK, here is my LOG file, just turned ON in heating mode, tried all states for horizontal swing from left to right position, tried vertical swing on and off then turned OFF unit.
[logs_klimatizace-c_logs (1).txt](https://github.com/paveldn/haier-esphome/files/14976955/logs_klimatizace-c_logs.1.txt)

paveldn, 2024-04-16:
@Alloc186  Thanks. Yes, something is not right. Modes not detected correctly. Not sure why. I will need to dig deeper into this. Thanks for putting my attention on this.

paveldn, 2024-04-18:
@Alloc186  OK I know where the problem is. Your AC is using values for the vertical and horizontal positions that are not used in my integration so when an unknown value is received my app will always use "Center" (for both vertical and horizontal). I want to rework a little bit away how my integration works with this. I will let you know once I have something to test.

Alloc186, 2024-04-20:
Super, let me know! You are the best ;) For info: my indoor AC unit is as25s2sf2fa-2, may have different settings option than yours.

paveldn, 2024-04-24:
@paveldn
I made some changes to the experimental branch. Can you please test them?
I found a couple of new modes but because my ACs don't support horizontal swing I can't test what they doing. Can you please, also check what is effect of these settings is? I called them "Test 01" and "Test 02" in horizontal swing select.

Alloc186, 2024-04-24:
Yes,  i would loved to, but I don't know how to switch to experimental version. I am noob in ESP world :)

paveldn, 2024-04-24:
No problem Everybody starts with something. 
You need to modify your config in a part that configures AC. Easiest way to use premaid config from my repo. So you will need to have something like this:
```
# You can use your own names and IDs
substitutions:
  device_name: Haier AC
  device_id: haier_climate
  uart_id: ac_port
  send_wifi: "true"

esphome:
  name: haier

# Here use your board settings
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y

#Use your own wifi settings
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

#Here use your setup UART pins
uart:
  baud_rate: 9600
  tx_pin: 17
  rx_pin: 16
  id: ${uart_id}

logger:
  level: DEBUG

# Use my component from experimental branch
external_components:
  - source: github://paveldn/haier-esphome@experimental
    refresh: 0s

# Use premaid config for hOn AC from experimental branch
packages:
  remote_package:
    url: https://github.com/paveldn/haier-esphome
    ref: experimental
    refresh: 0s
    files:
      - .haier-hon-base.yaml

```

Alloc186, 2024-04-24:
Thanks for your help ;) IT WORKS!

OK, so Left from remote is left on ESP, WOW!
Max left=> Max left
Center=> Center
Right=> Right
Max right=> Max right
Swinging shows as AUTO
Test01 and Test02, just keeps in center position, nothing extra.

For horizontal:
Hmm it only shows Max Up, if I stop motion it in any position with remote (change from swinging mode). It was in heating mode...

With extra function: Health Down from remote=Health Down on ESP
With extra function: Health Up from remote=Health Up on ESP
With extra function: Down from remote=Down on ESP

Manually switched form ESP webpage:
Health UP=> 100% all up
Max Up=> 95% up
Up=>70%
Center=>50%
Down=> 20%
Max down=> 40% (strange, it should be bellow down, visa versa)
Health Down=>0% all down
Swinging shows as AUTO

So horizontal control works 100%, vertical acts strange...

paveldn, 2024-04-25:
> I will swap Max Down <=> Down, and remove extra not needed options.

Done. You can update your ESP. It will automatically pull changes.
I will prepare a pull request for ESPHome but usually, it takes a month or two for changes to be included in the official ESPHome so don't expect it will be there soon. Maybe I will also include some other changes I planned. Will see.

Alloc186, 2024-04-25:
You are the best thank you! Send me PayPal address to donate your work ;)

paveldn, 2024-04-26:
Hi @Alloc186 
Thanks, I would appreciate it. [My PayPal](https://www.paypal.com/paypalme/truemark?country.x=NL&locale.x=en_US) And glad I could help 👍

Alloc186, 2024-04-26:
10 EURO just arrived to you, enjoy your beer mate, thank you. 👍😊.

closed 2024-04-26 by paveldn

zmstr, 2024-05-16:
Sorry to hog your issue @Alloc186 but this issue made me buy an AtomS3U but I'm unable to get it working. Something with “Configuration does not match the platform of the connected device. Expected an ESP32 device”. Would you mind sharing your full config?

paveldn, 2024-05-17:
@zmstr  Hi, I just recently got my AtomS3U and haven't tried it yet with Haier. I will make a config and test it later today to make sure it is working.

Alloc186, 2024-05-17:
> Sorry to hog your issue @Alloc186 but this issue made me buy an AtomS3U but I'm unable to get it working. Something with “Configuration does not match the platform of the connected device. Expected an ESP32 device”. Would you mind sharing your full config?

Hello, here is my config:

substitutions:
  name: "klimatizace-a"
  friendly_name: Klimatizace A

  # Přidáno
  device_name: Haier AC
  device_id: haier_climate
  uart_id: ac_port
  send_wifi: "true"

# Přidáno
# Component from the experimental branch
external_components:
  - source: github://paveldn/haier-esphome@experimental
    refresh: 0s

# Premaid config for hOn AC from the experimental branch
packages:
  remote_package:
    url: https://github.com/paveldn/haier-esphome
    ref: experimental
    refresh: 0s
    files:
      - .haier-hon-base.yaml

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/firmware/esphome-web/esp32s3.yaml@v2
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

uart:
  baud_rate: 9600
  tx_pin: 19
  rx_pin: 20
  id: ac_port

paveldn, 2024-05-17:
@zmstr my version of config. Pretty much the same as @Alloc186 version just a little bit more compact:
```
substitutions:
  device_name: Haier AC
  device_id: haier_climate
  uart_id: ac_port
  send_wifi: "true"

esphome:
  name: haier
  name_add_mac_suffix: true
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

uart:
  baud_rate: 9600
  tx_pin: 19
  rx_pin: 20
  id: ${uart_id}

web_server:

logger:
  level: DEBUG

packages:
  remote_package:
    url: https://github.com/paveldn/haier-esphome
    refresh: 0s
    files:
      - .haier-hon-base.yaml
```

Keep in mind that AtomS3U has much more than I used there is an IR transmitter, RGB LED, and even a microphone. You can find some  configs for this device here: https://github.com/esphome/feature-requests/issues/2332

zmstr, 2024-05-20:
Awesome, thanks for sharing your config guys. I got it to work last night and it looks pretty slick in the AC instead of a soldered USB male connector with a dangling ESP8266 😊

Alloc186, 2024-05-25:
Hello, I maybe found a bug in dev version - Switching mode to heating does not affect unit setting, unit just stays in cooling mode :/.

paveldn, 2024-05-26:
Hi,

I need a log from you. First with failed attempts to switch from cooling to heating with my component. Also, logs when switching in the same situation with it remote.