paveldn/haier-esphome #44: Fan mode

uKL, 2024-06-18:
Is it expected that fan mode and swing mode are not visible with esp32-haier-module.yaml?

paveldn, 2024-06-18:
@uKL You need to provide more information: visible where? Home Assistant? Web Interface? What kind of config you are using?
Usually, fan and swing modes will not be visible in the web interface just in Home Assistant.

uKL, 2024-06-19:
Web Interface. This is my config
```

substitutions:
  device_name: Haier AC
  device_id: haier_climate
  uart_id: ac_port
  send_wifi: "true"

esphome:
  name: haier
  name_add_mac_suffix: true
  platformio_options:
    upload_speed: 115200

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:
  port: 8080
  local: true
  auth:
    username: admin
    password: !secret api_pwd

uart:
  baud_rate: 9600
  tx_pin: 17
  rx_pin: 16
  id: ${uart_id}

logger:
  level: DEBUG

packages:
  local_haier: !include configs/external_components/local_haier.yaml
  haier_base: !include .haier-hon-base.yaml 

```

I'm gonna use it with MQTT in the long run, without the Home Assistant :)

This is what I see in the Web Intetrface.
[image]

Is it possible to add fan and swing like the manual direction control?

uKL, 2024-06-19:
Well, I'll answer myself :) It is available via MQTT so it is possible to add it to the Web UI.

closed 2024-07-02 by paveldn