paveldn/haier-esphome #46: i need help to intall it :)

lakusz1github, 2024-07-24:
Hi

I'm a little noob, and probably i made some stupid mistake, which i not see it.

I use D1 mini, but i have SmartAir2 version of climate, so i "merge" smartair2-esp8266.yaml and wemos_d1_mini.yaml and put it on my d1_mini device, i also put file .haier-smartair2-base.yaml and folder configs to HA\esphome\*

i get all the time "Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST"

i also add to file \configs\climate\haier_smartair2.yaml
```
climate:
  - platform: haier
    ...
    answer_timeout: 200ms
    ...
```

from https://github.com/paveldn/haier-esphome/issues/28 

how install this integration properly in my HA ?

paveldn, 2024-07-24:
Hi. First of all can you provide your full configuration (of course replace all passwords, keys and IPs with xx)? Also please check that UART port config is correct (same pins as in old working version of integration)

lakusz1github, 2024-07-25:
```
substitutions:
  device_name: Haier AC
  device_id: haier_climate
  uart_id: ac_port
  send_wifi: "true"

esphome:
  name: esp-haier
#  name_add_mac_suffix: true
  friendly_name: esp-Haier

esp8266:
  board: d1_mini_lite

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  platform: esphome
  password: 

web_server:
  port: 8080
 
wifi:
  ssid: 
  password: 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Haier Fallback Hotspot"
    password: 

captive_portal:
 
uart:
  - id: ${uart_id}
    baud_rate: 9600
    # You can also use pins 15 (TX) and 13 (RX)
    tx_pin: 1
    rx_pin: 3

packages:
#  haier_base: !include .haier-hon-base.yaml
  haier_base: !include .haier-smartair2-base.yaml
```

paveldn, 2024-07-26:
OK. The config looks OK. Can you provide logs from devices 30-40 seconds will be enough.

lakusz1github, 2024-07-26:
```
[14:33:00][C][api:142]:   Using noise encryption: YES
[14:33:00][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[14:33:00][W][component:237]: Component haier.climate took a long time for an operation (51 ms).
[14:33:01][W][component:238]: Components should block for at most 30 ms.
[14:33:01][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[14:33:01][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[14:33:01][W][haier.protocol:013]: Message handler error, msg=62, err=1
[14:33:01][W][component:237]: Component haier.climate took a long time for an operation (449 ms).
[14:33:01][W][component:238]: Components should block for at most 30 ms.
[14:33:02][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[14:33:02][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[14:33:03][I][haier.climate:096]: Answer timeout for command 61, phase SENDING_INIT_1
[14:33:03][W][component:237]: Component haier.climate took a long time for an operation (148 ms).
[14:33:03][W][component:238]: Components should block for at most 30 ms.
[14:33:03][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[14:33:03][W][haier.protocol:013]: Message handler error, msg=62, err=1
[14:33:03][W][component:237]: Component haier.climate took a long time for an operation (279 ms).
[14:33:03][W][component:238]: Components should block for at most 30 ms.
[14:33:03][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[14:33:03][W][component:237]: Component haier.climate took a long time for an operation (87 ms).
[14:33:03][W][component:238]: Components should block for at most 30 ms.
[14:33:05][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[14:33:07][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[14:33:07][I][haier.climate:096]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[14:33:13][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[14:33:13][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[14:33:14][W][component:237]: Component haier.climate took a long time for an operation (57 ms).
[14:33:14][W][component:238]: Components should block for at most 30 ms.
[14:33:14][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[14:33:14][W][haier.protocol:013]: Message handler error, msg=62, err=1
[14:33:14][W][component:237]: Component haier.climate took a long time for an operation (279 ms).
[14:33:14][W][component:238]: Components should block for at most 30 ms.
[14:33:15][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[14:33:15][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[14:33:16][W][component:237]: Component haier.climate took a long time for an operation (57 ms).
[14:33:16][W][component:238]: Components should block for at most 30 ms.
[14:33:16][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[14:33:16][W][haier.protocol:013]: Message handler error, msg=62, err=1
[14:33:16][W][component:237]: Component haier.climate took a long time for an operation (279 ms).
[14:33:16][W][component:238]: Components should block for at most 30 ms.
[14:33:17][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[14:33:17][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[14:33:18][I][haier.climate:096]: Answer timeout for command 61, phase SENDING_INIT_1
[14:33:18][W][component:237]: Component haier.climate took a long time for an operation (148 ms).
[14:33:18][W][component:238]: Components should block for at most 30 ms.
[14:33:18][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[14:33:18][W][haier.protocol:013]: Message handler error, msg=62, err=1
[14:33:18][W][component:237]: Component haier.climate took a long time for an operation (278 ms).
[14:33:18][W][component:238]: Components should block for at most 30 ms.
[14:33:18][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[14:33:18][W][component:237]: Component haier.climate took a long time for an operation (87 ms).
[14:33:18][W][component:238]: Components should block for at most 30 ms.
[14:33:20][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[14:33:22][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[14:33:22][I][haier.climate:096]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[... cut ...]
[14:33:30][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[14:33:31][W][component:237]: Component haier.climate took a long time for an operation (57 ms).
[14:33:31][W][component:238]: Components should block for at most 30 ms.
[14:33:31][W][haier.protocol:013]: Unsupported message received: type 62 data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
```

paveldn, 2024-07-26:
OK Let me promote to you my frequently asked questions 😄 : https://github.com/paveldn/haier-esphome/blob/master/docs/faq.rst#q-what-does-the-unsupported-message-received-type-xx-message-in-the-logs-indicate
In your case, it looks like your device is starving. It is too overloaded. Try to remove:
- AP and captive_portal
- web_server
- API encryption
Try disabling them one by one and after each disable try if it starts working better.

lakusz1github, 2024-07-27:
Thx for advisorys, but not help. i turn off AP, CP and WS. now i set time out for 300ms, and logs look lie that

```
[12:50:56][I][haier.climate:096]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[12:51:02][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[12:51:02][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[12:51:02][D][haier.protocol:019]: Answer delay 206ms
[12:51:03][W][component:237]: Component haier.climate took a long time for an operation (114 ms).
[12:51:03][W][component:238]: Components should block for at most 30 ms.
[12:51:04][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[12:51:06][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[12:51:08][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[12:51:08][I][haier.climate:096]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[12:51:14][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[12:51:14][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[12:51:14][D][haier.protocol:019]: Answer delay 209ms
[12:51:15][W][component:237]: Component haier.climate took a long time for an operation (115 ms).
[12:51:15][W][component:238]: Components should block for at most 30 ms.
[12:51:16][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[12:51:18][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[12:51:20][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[12:51:21][I][haier.climate:096]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[12:51:26][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[12:51:26][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[12:51:26][D][haier.protocol:019]: Answer delay 206ms
[12:51:27][W][component:237]: Component haier.climate took a long time for an operation (115 ms).
[12:51:27][W][component:238]: Components should block for at most 30 ms.
[12:51:28][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[12:51:30][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[12:51:32][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[12:51:33][I][haier.climate:096]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
```

If i disable API, how HA can manage the divecie, so i leave it now.

paveldn, 2024-07-27:
> Answer delay 206ms
It is a huge delay. I would advise you to use ESP32 in your case which has much more power

lakusz1github, 2024-07-31:
Hi @paveldn 

i found ESP32 c3, but i have different layout, i have pin 20 and 21 (RX and TX), and in yours sample in yaml file:
  tx_pin: 19
  rx_pin: 18

should i just replace to:
  tx_pin: 21
  rx_pin: 20 ?

paveldn, 2024-07-31:
Hi @lakusz1github 
My examples are just for reference you can also use other pins. 20 and 21 should be OK

lakusz1github, 2024-08-04:
Hi i get esp32-c3, and in the loop:

```
[16:59:01][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[16:59:01][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[16:59:01][D][haier.protocol:019]: Answer delay 83ms
[16:59:01][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:59:03][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:59:05][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:59:05][I][haier.climate:097]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[16:59:11][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[16:59:13][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[16:59:13][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[16:59:13][D][haier.protocol:019]: Answer delay 83ms
[16:59:13][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:59:15][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:59:17][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[16:59:17][I][haier.climate:097]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
```

paveldn, 2024-08-04:
@lakusz1github it looksblike your ac is using hon protocol. Try to add 

protocol: hon

 to your config

paveldn, 2024-09-02:
No feedback. I suppose issue fixed

closed 2024-09-02 by paveldn