paveldn/haier-esphome #49: Feature request: Status sensor for Display and Health mode

GroteGehaktBal, 2024-08-21:
Thank you for this amazing integration, It is working perfectly and I recently added buttons for the turn on and of Display and Health actions.
I would like to be able to see the current status of these so I can use a switch in home assistant to turn it on/off. I want the switch to update according to the current state. Would this be possible?

paveldn, 2024-08-21:
Hi @GroteGehaktBal ,
You are welcome. 
Unfortunately, I don't have a lot of time recently. I am actually implementing exactly what you need right now. It will be special switches for the Haier platform. But it will take some time until it will be finished. But you can easily use template switches for this. Check these examples:
https://github.com/paveldn/haier-esphome/blob/master/configs/switch/display.yaml
https://github.com/paveldn/haier-esphome/blob/master/configs/switch/health_mode.yaml

GroteGehaktBal, 2024-08-21:
Thank you so much for the fast reply. That will solve my issue!

closed 2024-08-21 by GroteGehaktBal