paveldn/haier-esphome #50: Get no response on UART

balloob, 2024-08-25:
Hey Pavel,

Thanks for all your work on integrating Haier devices 😄 

I am trying to use the Haier component in ESPHome with an [`SLWF-01`](https://smartlight.me/smart-home-devices/wifi-devices/wifi-dongle-air-conditioners-midea-idea-electrolux-for-home-assistant) connected to a Haier split unit model `AS25S2SF2FA-1` (manuf Nov. 2019).

I am using the SmartAir2 protocol but I am unable to get any messages back. Now I am 99% sure it's my fault, but even with the help of Jesse, I wasn't able to figure it out 😞 

```yaml
logger:
  baud_rate: 0

uart:
  tx_pin: 12
  rx_pin: 14
  baud_rate: 9600
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, ' ');

climate:
  - platform: haier
    id: haier_ac
    protocol: smartair2
    name: "${friendly_name}"
```

Now I once had a soldered version using [this custom component](https://github.com/MiguelAngelLV/esphaier) that I remember did work partially. I tried mimicking the message it sends (it's very similar), but also that gives me no response:

```yaml
button:
  - platform: uart
    name: Do the poll
    id: midea_display_toggle
    data: [255, 255, 10, 0, 0, 0, 0, 0, 1, 1, 77, 1, 90]
```

Can you see at a glance what's going wrong? 

Thanks,
Paulus

paveldn, 2024-08-26:
Hi @balloob 

You are welcome, and thank you for ESPhome and Home Assistant it allow us geeks to have a lot of fun with smart home 😄  
This kind of problems is most difficult because it may lay in hardware plain. I suppose you checked your wirings and sure that all is working. I have a way to check it which will require some advanced knowledge how to build code with CMake. I described it here: 
https://github.com/paveldn/haier-esphome/blob/master/docs/faq.rst#q-i-connected-the-esp-module-but-all-i-can-see-in-the-logs-is-answer-timeout-for-command-61-phase-sending_init_1-and-nothing-is-working-what-should-i-do I am talking about using simulator part. You will need TTL to USB convertor and PC with Windows or Linux with CMake and C++ compiler.  Unfortunately Haier AC (both hOn and smartAir) not sending anything by default just answering requests. With simulator you can check of your wiring is OK. Like simulator don't receive anything - you having problem with TX, if simulator sends answer but you don't receive it on ESP side you have a problem in RX.

Also have you tried using hOn protocol? Do you get any logs on ESP side?

balloob, 2024-09-03:
I did try the h0n protocol, but that didn't work. It's weird I get no responses.  Any chance there is a SmartAir protocol?

I had this issue at my parents in the Netherlands but I left now back to the States. Will close this issue as I have no further time to investigate. Thanks!

closed 2024-09-03 by balloob

paveldn, 2024-09-04:
Sorry, it didn't work. When you will be in the Netherlands next time let me know upfront. Maybe if we would have more time we will figure it out.

balloob, 2024-09-05:
It will be a while I think! Hopefully next summer

raineth, 2024-09-05:
I recently tried using an SLWF-01 for a different Haier unit and found that the TX/RX pins are backwards compared to the official Wi-Fi module.  The version I had (the newer one with a USB-C connector) has unidirectional level shifters, so switching the RX/TX pins in the configuration wasn't enough to get things working for me.  The board doesn't have a convenient way to swap TX/RX, so I ended up soldering a USB cable to a different ESP board.

paveldn, 2024-09-09:
> It will be a while I think! Hopefully next summer

Hello Paulus.

I actually may have a solution for you. I have some spare equipment and I can make a small device for you. I will create a sample configuration and test it on my side then I will share a configuration with you. You will change it as you want (but keep hardware compatible with the initial one) and build binaries. Binaries you will share with me and I will flash them. It is just binaries so no passwords or keys sharing. After that, I will l send the device by post to your parent's house. All your parents will need to do is install a USB plug into the AC USB socket. 

Let me know if you are OK with this.