paveldn/haier-esphome #51: Is "silent mode" supported?

wzieba, 2024-08-25:
hi there! Thank you a lot for your contribution. I successfully configured the controller and made my Haier devices much better comparing to the time I was using hOn API :)

I have a question about modes: is "silent mode", as described in hOn mobile app, supported?

<details><summary>Screen from hOn mobile app</summary>

[image]
</details>

In this part of source code, I see a mention of "quiet mode" (which I'm guessing is the same as "silent mode"). But, if I interpret the code below correctly, it's mapped to "eco" preset. Looking at the description from the mobile app, those two modes (eco and silent) are separate, so hence this question: is "silent mode" supported or planned to be?

https://github.com/paveldn/haier-esphome/blob/b07f7a67a36ddc92dc46ef02855f3521704d29bf/components/haier/hon_climate.cpp#L596-L602

paveldn, 2024-08-26:
Hi @wzieba,

As far as I remember silent mode is Night mode preset. Have you tried it?
[image]

wzieba, 2024-08-26:
Thanks - yes I've tried the Sleep mode, but this is something different from Silent (a.k.a Quiet) mode. Quite/Silent makes the AC quiter, but Sleep applies an algorithm of changing the temperature over time.

I've looked as some Haier manuals, e.g. here https://www.kokotas.gr/images/usrImage/4-07-2024-14-49-66868c3a2e23b.pdf and on page 5 there are details about each of these modes.

And this is how I think things are named between manual and the mobile app.

[image]

paveldn, 2024-08-27:
OK Can you please make a log for me from your AC using IR remote? 
Like in this instruction https://github.com/paveldn/haier-esphome/blob/master/docs/faq.rst#q-my-ac-has-a-cool-feature-that-is-not-supported-by-your-component-can-you-add-it
I will check if it is supported and if not will try to add it.

wzieba, 2024-08-29:
When trying to prepare logs, I've realized the following:

- My AC unit (Haier Pearl Plus), at least according to the manual, doesn't seem to have "ECO mode". This is contrary to the screenshot from the app I shared.
- Turning on "QUIET" on remote enables "ECO" preset on ESPHome.
- Turning on "SLEEP" on remote enables "SLEEP" preset on ESPHome.

So, overall, everything seems to be supported. 

---

I only wonder what signals are sent whenever user taps "ECO mode" vs. "Silent mode" on mobile app. But foor this to verify, I'd need to sniff the communication. You made an awesome work with [the documentation](https://github.com/paveldn/haier-esphome/blob/master/docs/sniffing_serial_communication.rst), maybe I'll follow this and check signals, but I don't have tools right now.

Anyway, it looks like all features are supported by haier-esphome. I'll close this issue for now, and if I'll find something useful sniffing the communication after mobile app turns on `silent`/`eco`, I'll follow up here.

closed 2024-08-29 by wzieba