paveldn/haier-esphome #52: Beginner questions, and MQTT

danielgindi, 2024-09-01:
Hi Pavel!

This project looks promising, thank you for investing your time in it!
I did try to hop on the SmartAir2 protocol with the Chinese servers, but after a while I realized it was not worth it as these servers are randomly unavailable, and I need something local.

If you don't mind, I have some very basic questions for lack of prior experience with esphome.
* Do I need to specifically install this plugin somehow? Or is a configuration in a clean installation of esphome enough? It's not very clear.
* Does this work with MQTT too?
* If I enable MQTT, how do I know what are the messages to communicate with this esp? I want to connect this to mqttthing in a homebridge.

Thank you!
Daniel

paveldn, 2024-09-01:
Hi Daniel!

Answers to your questions:

* You don't need to install anything additionally. Haier component is included in ESPHome so just need to write the proper configuration.
* I don't have anything special regarding MQTT in my component so all MQTT functionality is the same as for any other ESPHome climate. 
* To tell the truth, I am not using my climates with MQTT so my knowledge here is limited. But you can try to check here https://esphome.io/components/climate/index.html (MQTT options section). Configure topics for climate component and you should be able to use them for control.

danielgindi, 2024-09-03:
Thank you very much!
Trying this as we speak. Will let you know how it goes :-)