paveldn/haier-esphome #56: Configuration failed

ShifuSonny, 2024-09-15:
Hello, I’m not very experienced with ESPHome, so I apologize if this question seems trivial.
I tried to upload the max-hon.yaml file to understand what my AC system can expose, removing only the API section from the basic configuration and all the automation part.
The script compiles, but I can no longer find the ESP8266 in any way, nor can I see the log.

Could it be because I’m powering it via USB? I wanted to test everything to ensure it was connected correctly to Home Assistant before wiring it to the AC unit, but since I can’t find it, I see no entities.
Where am I going wrong?

Thank you in advance for your work and for any response!

paveldn, 2024-09-15:
Hi. 
So let me get it straight. You are using ESP8266, you flashed it with configuration with my max hon example and kept it connected to the USB but can't get logs from it. Is this right?
Can you please check if the board is connected to your wifi (in your router)?
Are you sure you configured your board correctly? Can you please try the same config but remove all the related sections to check that the HW part of the config is OK?
Also, I would strongly recommend using ESP32 for any of them. ESP8266 is a really old and not very powerful board. I am not saying that the problem is because of ESP8266 but it might be.
Also, do you have a USB to TTL converter? If you have you can also connect to the UART port and check on the PC if the board is trying to talk to AC.

ShifuSonny, 2024-09-15:
Thank you for the quick response!
The issue was indeed with the Wi-Fi, even though I had checked it... 27 entities recognized! Tomorrow I’ll move on to the wiring, but I’m confident everything will work.

Thank you again for your work!

closed 2024-09-15 by ShifuSonny