paveldn/haier-esphome #57: msg=F7, answ=03. err=7

intmainvoidchar, 2024-09-17:
Здравствуйте, подскажите что можно сделать с такой проблемой?

[18:16:11][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 35
[18:16:11][D][haier.protocol:019]: Frame found: type 03, data: 00 00
[18:16:11][D][haier.protocol:019]: Answer delay 90ms
[18:16:11][W][haier.protocol:013]: Answer handler error, msg=F7, answ=03, err=7

плата esp32, конфиг максимальный из репозитория 1 к 1(smartair2), кондиционер AD09SS1ERA

```
uart:
  baud_rate: 9600
  tx_pin: 1
  rx_pin: 3
  id: haier_uart

climate:
  - platform: haier
    id: haier_ac
    protocol: smartAir2
    name: Haier SmartAir2 Climate
    uart_id: haier_uart
    alternative_swing_control: false
    wifi_signal: true
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:
      - 'OFF'
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
      - FAN_ONLY
    supported_swing_modes:
      - 'OFF'
      - VERTICAL
      - HORIZONTAL
      - BOTH
    supported_presets:
      - BOOST
      - COMFORT
      - AWAY

switch:
  - platform: template
    id: haier_ac_health_mode
    name: Haier SmartAir2 Climate health mode
    icon: mdi:leaf
    restore_mode: RESTORE_DEFAULT_OFF
    lambda: |-
      return id(haier_ac).get_health_mode();
    turn_on_action:
      climate.haier.health_on: haier_ac
    turn_off_action:
      climate.haier.health_off: haier_ac
  - platform: template
    id: haier_ac_display_switch
    name: Haier SmartAir2 Climate display
    icon: mdi:led-on
    entity_category: config
    restore_mode: RESTORE_DEFAULT_ON
    lambda: |-
      return id(haier_ac).get_display_state();
    turn_on_action:
      climate.haier.display_on: haier_ac
    turn_off_action:
      climate.haier.display_off: haier_ac
```

paveldn, 2024-09-17:
Добрый день. А все остальное работает?

Чтобы убрать сообщение об ошибке поменяйте значение wifi_signal с true на false.
Для silent mode используйте пресет COMFORT

intmainvoidchar, 2024-09-17:
> Добрый день. А все остальное работает?
> 
> Чтобы убрать сообщение об ошибке поменяйте значение wifi_signal с true на false. Для silent mode используйте пресет COMFORT

вроде работает, только задержка в районе 20-30 сек (меняю температуру в ХА, и через 20-30 сек вижу результат на проводном пульте кондиционера)

smartair2_climate_display — вот эта штука не выключается, за что она отвечает?

paveldn, 2024-09-17:
> через 20-30 сек вижу результат на проводном пульте кондиционера

Может быть особенность проводного пульта. Не знаю как часто он опрашивает кондиционер.

>smartair2_climate_display — вот эта штука не выключается, за что она отвечает?

У вас кассетный кондиционер и нет светящегося экрана. Это для выключения светящегося диплея на самом кондиционере. Такие обычно есть на сплит системах.

intmainvoidchar, 2024-09-18:
> Добрый день. А все остальное работает?
> 
> Чтобы убрать сообщение об ошибке поменяйте значение wifi_signal с true на false. Для silent mode используйте пресет COMFORT

ошибка ушла.

и последний вопрос, можно ли безболезненно сделать изменение температуры по пол градуса? сам кондиционер это умеет и пульт тоже

paveldn, 2024-09-18:
> и последний вопрос, можно ли безболезненно сделать изменение температуры по пол градуса? сам кондиционер это умеет и пульт тоже

Попробуйте вместо  `temperature_step: 1 °C ` поставить `temperature_step: 0.5 °C`
Должно работать

closed 2024-09-22 by intmainvoidchar