paveldn/haier-esphome #59: Wemos D1 mini UART fails mostly

Atlas553, 2024-10-14:
Good morning, 

last winter I configured my "Heiko" ac in my bathroom with esphome. Everything worked as it should until march. (last known running state) 

Now I run into the issue that the Wemos D1 gets no answer from the ac unit in most cases. I think the configuration is still okay because some times it works for a short time.  

I tinkered around with timeout and log settings but get now clear view on my error.

Are there any hints out there what I can try to figure out whats the problem is? 

Thanks a lot!

Log
```
Time	level	Tag	Message
13:09:04	[D]	[haier.protocol:019]	Sending frame: type 01, data: 4D 01
13:09:04	[D]	[uart_debug:158]	>>> "\xFF\xFF\n"
13:09:04	[D]	[uart_debug:158]	>>> "@\x00\x00\x00\x00\x00\x01M\x01\x99\xB3\xB4"
13:09:04	[D]	[uart_debug:158]	<<< "\xFF\xFF*@\x00\x00\x00\x00\x00\x02m\x01\x06\n"
13:09:04	[D]	[haier.protocol:019]	Frame found: type 02, data: 6D 01 06 0A 83 00 00 01 00 05 00 00 2F 00 4B 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
13:09:04	[D]	[haier.protocol:019]	Answer delay 64ms
13:09:04	[D]	[haier.climate:1000]	HVAC Mode = 0x4
13:09:04	[D]	[haier.climate:1001]	Fan speed Status = 0x3
13:09:04	[D]	[haier.climate:1002]	Horizontal Swing Status = 0x5
13:09:05	[D]	[haier.climate:1003]	Vertical Swing Status = 0xA
13:09:05	[D]	[haier.climate:1004]	Set Point Status = 0x6
13:09:05	[D]	[uart_debug:158]	<<< "\x83\x00\x00\x01\x00\x05\x00\x00/\x00K\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xF0\x01C"
13:11:00	[W]	[haier.climate:223]	Answer timeout for command FC, phase SENDING_UPDATE_SIGNAL_REQUEST
13:11:03	[D]	[haier.protocol:019]	Sending frame: type 01, data: 4D 01
13:11:03	[D]	[uart_debug:158]	>>> "\xFF\xFF\n"
13:11:03	[D]	[uart_debug:158]	>>> "@\x00\x00\x00\x00\x00\x01M\x01\x99\xB3\xB4"
13:11:03	[W]	[haier.climate:223]	Answer timeout for command 01, phase SENDING_STATUS_REQUEST
13:12:50	[D]	[haier.protocol:019]	Sending frame: type 61, data: 00 07
13:12:50	[D]	[uart_debug:158]	>>> "\xFF\xFF\n"
13:12:50	[D]	[uart_debug:158]	>>> "@\x00\x00\x00\x00\x00a\x00\a\xB2\xFFU\x01"
13:12:50	[W]	[haier.climate:223]	Answer timeout for command 61, phase SENDING_INIT_1
```

Esphome Config
```
esphome:
  name: klima-bad
  friendly_name: "Klima Bad"
  on_boot:
    priority: 200
    then:
      - lambda: |-        
          delay(15000);
esp8266:
  board: d1_mini

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG
  logs:
    bluetooth_proxy: WARN
    esp32_ble: WARN
    json: WARN
    sensor: DEBUG
    haier.protocol: DEBUG
    haier.climate: DEBUG
    climate: DEBUG
    web_server_idf: WARN
    mdns: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXX"

ota:
- platform: esphome
  password: "XXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Klima-Bad Fallback Hotspot"
    password: "XXXXX"

captive_portal:

web_server:
  port: 80
    
uart:
  baud_rate: 9600
  tx_pin: 1
  rx_pin: 3
  id: haier_uart
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

climate:
  - platform: haier
    protocol: HON
    id: haier_climate
    uart_id: haier_uart
    name: Klimaanlage Bad
    answer_timeout: 3000ms
    #control_packet_size: 18
    #control_method: SET_SINGLE_PARAMETER
    wifi_signal: true
    #beeper: false
    display: false
    supported_modes:
    - 'OFF'
    - HEAT_COOL
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    supported_swing_modes:
    - 'OFF'
    - VERTICAL

sensor:
  - platform: haier
    haier_id: haier_climate
    outdoor_temperature:
      name: Klimaanlage Bad outdoor temperature
    humidity:
      name: Haier AC indoor humidity

binary_sensor:
  - platform: haier
    haier_id: haier_climate
    defrost_status:
      name: Klimaanlage Bad defrost status
```

Timple, 2024-10-14:
I might have the same issue. After a powercycle I works again for a little while. So I don't think it's the wiring.

Mine is a Haier btw

paveldn, 2024-10-14:
@Atlas553 Try to set `wifi_signal: false` it looks like your AC stops responding when getting the command used to update the wifi signal. Also powercycle your AC.

@Timple,
I will need logs. In your case can be completely different reason.

Atlas553, 2024-10-14:
@paveldn 
Thanks for your quick reply. 

I tried it but sadly with the exact same result. The unit does not replied also after power cycle. This time without an initial reply.

paveldn, 2024-10-14:
@Atlas553 Can you get me more logs? At least a couple of minutes. One with normal communication and another one without.

Atlas553, 2024-10-14:
Maybe the wifi option helped. After a few minuites after restart the ac unit it seems there is a stable communication (until now). 

I hope thats it. Anyway the log with regular communication. 

```
Time	level	Tag	Message
18:25:16	[D]	[haier.climate:1004]	
Set Point Status = 0x7
18:25:16	[D]	[uart_debug:158]	
<<< "\x83\x00\x00\x01\x00\x05\x00\x00/\x00J\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00`pM\x00\x00\x00\x01\xFFU\x02 \x00X\xFAw\xF6"
18:25:21	[D]	[haier.protocol:019]	
Sending frame: type 01, data: 4D 01
18:25:21	[D]	[uart_debug:158]	
>>> "\xFF\xFF\n"
18:25:21	[D]	[uart_debug:158]	
>>> "@\x00\x00\x00\x00\x00\x01M\x01\x99\xB3\xB4"
18:25:21	[D]	[uart_debug:158]	
<<< "\xFF\xFF*@\x00\x00\x00\x00\x00\x02m\x01\a\n"
18:25:21	[D]	[haier.protocol:019]	
Frame found: type 02, data: 6D 01 07 0A 83 00 00 01 00 05 00 00 2F 00 4A 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
18:25:21	[D]	[haier.protocol:019]	
Answer delay 61ms
18:25:21	[D]	[haier.climate:1000]	
HVAC Mode = 0x4
18:25:21	[D]	[haier.climate:1001]	
Fan speed Status = 0x3
18:25:21	[D]	[haier.climate:1002]	
Horizontal Swing Status = 0x5
18:25:21	[D]	[haier.climate:1003]	
Vertical Swing Status = 0xA
18:25:21	[D]	[haier.climate:1004]	
Set Point Status = 0x7
18:25:21	[D]	[uart_debug:158]	
<<< "\x83\x00\x00\x01\x00\x05\x00\x00/\x00J\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xF0|"
18:25:26	[D]	[haier.protocol:019]	
Sending frame: type 01, data: 4D 01
18:25:26	[D]	[uart_debug:158]	
>>> "\xFF\xFF\n"
18:25:26	[D]	[uart_debug:158]	
>>> "@\x00\x00\x00\x00\x00\x01M\x01\x99\xB3\xB4"
18:25:26	[D]	[uart_debug:158]	
<<< "\xFF\xFF*@\x00\x00\x00\x00\x00\x02m\x01\a\n"
18:25:26	[D]	[haier.protocol:019]	
Frame found: type 02, data: 6D 01 07 0A 83 00 00 01 00 05 00 00 2F 00 4A 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
18:25:26	[D]	[haier.protocol:019]	
Answer delay 62ms
18:25:26	[D]	[haier.climate:1000]	
HVAC Mode = 0x4
18:25:26	[D]	[haier.climate:1001]	
Fan speed Status = 0x3
18:25:26	[D]	[haier.climate:1002]	
Horizontal Swing Status = 0x5
18:25:26	[D]	[haier.climate:1003]	
Vertical Swing Status = 0xA
18:25:26	[D]	[haier.climate:1004]	
Set Point Status = 0x7
18:25:26	[D]	[uart_debug:158]	
<<< "\x83\x00\x00\x01\x00\x05\x00\x00/\x00J\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xF0|"
18:25:31	[D]	[haier.protocol:019]	
Sending frame: type 01, data: 4D FE
18:25:31	[D]	[uart_debug:158]	
>>> "\xFF\xFF\n"
18:25:31	[D]	[uart_debug:158]	
>>> "@\x00\x00\x00\x00\x00\x01M\xFE\x96\xF3\xF4"
18:25:31	[D]	[uart_debug:158]	
<<< "\xFF\xFF8@\x00\x00\x00\x00\x00\x02}\x01\a\n"
18:25:31	[D]	[haier.protocol:019]	
Frame found: type 02, data: 7D 01 07 0A 83 00 00 01 00 05 00 00 2F 00 4A 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 60 6F 4D 00 00 00 01 FF 02 20 00 58
18:25:31	[D]	[haier.protocol:019]	
Answer delay 76ms
18:25:31	[D]	[haier.climate:1000]	
HVAC Mode = 0x4
18:25:31	[D]	[haier.climate:1001]	
Fan speed Status = 0x3
18:25:31	[D]	[haier.climate:1002]	
Horizontal Swing Status = 0x5
18:25:31	[D]	[haier.climate:1003]	
Vertical Swing Status = 0xA
18:25:31	[D]	[haier.climate:1004]	
Set Point Status = 0x7
18:25:31	[D]	[uart_debug:158]	
<<< "\x83\x00\x00\x01\x00\x05\x00\x00/\x00J\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00`oM\x00\x00\x00\x01\xFFU\x02 \x00X\xF9\xA3\x92"
18:25:36	[D]	[haier.protocol:019]	
Sending frame: type 01, data: 4D 01
18:25:36	[D]	[uart_debug:158]	
>>> "\xFF\xFF\n"
18:25:36	[D]	[uart_debug:158]	
>>> "@\x00\x00\x00\x00\x00\x01M\x01\x99\xB3\xB4"
18:25:36	[D]	[uart_debug:158]	
<<< "\xFF\xFF*@\x00\x00\x00\x00\x00\x02m\x01\a\n"
18:25:36	[D]	[haier.protocol:019]	
Frame found: type 02, data: 6D 01 07 0A 83 00 00 01 00 05 00 00 2F 00 4A 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
18:25:36	[D]	[haier.protocol:019]	
Answer delay 63ms
18:25:36	[D]	[haier.climate:1000]	
HVAC Mode = 0x4
18:25:36	[D]	[haier.climate:1001]	
Fan speed Status = 0x3

```

paveldn, 2024-10-14:
@Atlas553  Great, the answer delays look good. It looks like your ESP8266 can handle the load. Lety me know if the problem happens again.

Atlas553, 2024-10-15:
Sadly tonight the unit stopped to communicate.
[image]

it looks like there were communication issues before because the unit is constantly powered on and in heating mode since yesterday: 
[image]

Since then the logs looks like: 
```
Time | level | Tag | Message
-- | -- | -- | --
08:51:30 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:51:33 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:51:40 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:51:40 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:51:40 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:51:43 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:51:50 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:51:50 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:51:50 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:51:53 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:52:00 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:52:00 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:52:00 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:52:03 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:52:10 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:52:10 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:52:10 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:52:13 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:52:20 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:52:20 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:52:20 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:52:23 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:52:30 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:52:30 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:52:30 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:52:33 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:52:40 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:52:40 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:52:40 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:52:43 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:52:50 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:52:50 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:52:50 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:52:53 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:53:00 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:53:00 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:53:00 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:53:03 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:53:10 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:53:10 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:53:10 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:53:13 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:53:20 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:53:20 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:53:20 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:53:23 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1
08:53:30 | [D] | [haier.protocol:019] | Sending frame: type 61, data: 00 07
08:53:30 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
08:53:30 | [D] | [uart_debug:158] | >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
08:53:33 | [W] | [haier.climate:223] | Answer timeout for command 61, phase SENDING_INIT_1

```

paveldn, 2024-10-15:
@Atlas553,
What I see in the logs is that AC just not answering anymore. It is strange that you have this kind of problem after recent updates. Can you try ESP32 board instead of ESP8266? I have a feeling that ESP8266 just not able to handle recent ESPHome updates. More and more features introduced and ESP8266 is old and not powerful enough. Maybe the chip is overheated or something.

Atlas553, 2024-10-15:
@paveldn 

I tried with this one: 
[image]

But no luck so far. 
Config
```
substitutions:
  name: esphome-web-ca36f8
  friendly_name: Klima Bad ESP32

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

    
uart:
  baud_rate: 9600
  tx_pin: 1
  rx_pin: 3
  id: haier_uart
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

climate:
  - platform: haier
    protocol: HON
    id: haier_climate
    uart_id: haier_uart
    name: Klimaanlage Bad
    answer_timeout: 3000ms
    #control_packet_size: 18
    #control_method: SET_SINGLE_PARAMETER
    wifi_signal: false
    #beeper: false
    display: false
    supported_modes:
    - 'OFF'
    #- HEAT_COOL
    - COOL
    - HEAT
    - DRY
    #- FAN_ONLY
    #supported_swing_modes:
    #- 'OFF'
    #- VERTICAL

sensor:
  - platform: haier
    haier_id: haier_climate
    outdoor_temperature:
      name: Klimaanlage Bad outdoor temperature
    #humidity:
     # name: Haier AC indoor humidity

binary_sensor:
  - platform: haier
    haier_id: haier_climate
    defrost_status:
      name: Klimaanlage Bad defrost status
```

Log
```
INFO Successfully connected to esphome-web-ca36f8 @ 192.168.4.175 in 0.006s
INFO Successful handshake with esphome-web-ca36f8 @ 192.168.4.175 in 0.182s
[11:17:27][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[11:17:27][D][uart_debug:158]: >>> "\xFF\xFF\n"
[11:17:27][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[11:17:30][W][haier.climate:224]: Answer timeout for command 61, phase SENDING_INIT_1
[11:17:37][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[11:17:37][D][uart_debug:158]: >>> "\xFF\xFF\n"
[11:17:37][W][component:237]: Component haier.climate took a long time for an operation (145 ms).
[11:17:37][W][component:238]: Components should block for at most 30 ms.
[11:17:37][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[11:17:40][W][haier.climate:224]: Answer timeout for command 61, phase SENDING_INIT_1
[11:17:47][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[11:17:47][D][uart_debug:158]: >>> "\xFF\xFF\n"
[11:17:47][W][component:237]: Component haier.climate took a long time for an operation (140 ms).
[11:17:47][W][component:238]: Components should block for at most 30 ms.
[11:17:47][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[11:17:50][W][haier.climate:224]: Answer timeout for command 61, phase SENDING_INIT_1
[11:17:57][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[11:17:57][D][uart_debug:158]: >>> "\xFF\xFF\n"
[11:17:57][W][component:237]: Component haier.climate took a long time for an operation (144 ms).
[11:17:57][W][component:238]: Components should block for at most 30 ms.
[11:17:57][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[11:18:00][W][haier.climate:224]: Answer timeout for command 61, phase SENDING_INIT_1
[11:18:06][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[11:18:06][D][esp32.preferences:114]: Saving 1 preferences to flash...
[11:18:06][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[11:18:07][W][component:237]: Component safe_mode took a long time for an operation (142 ms).
[11:18:07][W][component:238]: Components should block for at most 30 ms.
[11:18:07][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[11:18:07][D][uart_debug:158]: >>> "\xFF\xFF\n"
[11:18:07][W][component:237]: Component haier.climate took a long time for an operation (139 ms).
[11:18:07][W][component:238]: Components should block for at most 30 ms.
[11:18:07][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[11:18:07][W][component:237]: Component uart took a long time for an operation (137 ms).
[11:18:07][W][component:238]: Components should block for at most 30 ms.
[11:18:10][W][haier.climate:224]: Answer timeout for command 61, phase SENDING_INIT_1
[11:18:17][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[11:18:17][D][uart_debug:158]: >>> "\xFF\xFF\n"
[11:18:17][W][component:237]: Component haier.climate took a long time for an operation (143 ms).
[11:18:17][W][component:238]: Components should block for at most 30 ms.
[11:18:17][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[11:18:20][W][haier.climate:224]: Answer timeout for command 61, phase SENDING_INIT_1
[11:18:27][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[11:18:27][D][uart_debug:158]: >>> "\xFF\xFF\n"
[11:18:27][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[11:18:30][W][haier.climate:224]: Answer timeout for command 61, phase SENDING_INIT_1
```

paveldn, 2024-10-15:
@Atlas553 I don't see AC answers at all. Are you sure the wiring is OK? Please, also powercycle AC. Sometimes AC gets into a strange state and stops answering (internal bug?).

Atlas553, 2024-10-22:
Thanks a lot for your help. 

I think it is correct. But yes - no answer so far. Powercycled a fey times now and wait a lot. 

Maybe you can find any wiring issue. 

Here is my wiring: 
[image]

[image]

Atlas553, 2024-10-22:
Okay. @paveldn 

A few minutes after taking the picture and wrote the post I got an answer from the ac. 

Wiring should be fine.  

```
Time | level | Tag | Message
-- | -- | -- | --
13:28:47 | [D] | [uart_debug:158] | <<< "\xFF\xFF8@\x00\x00\x00\x00\x00\x02}\x01\x05\n"
13:28:47 | [D] | [haier.protocol:019] | Frame found: type 02, data: 7D 01 05 0A 83 00 02 01 00 05 00 00 2C 00 4D 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 76 70 44 00 00 21 01 FF 02 25 00 58
13:28:47 | [D] | [haier.protocol:019] | Answer delay 116ms
13:28:47 | [D] | [climate:396] | 'Klimaanlage Bad' - Sending state:
13:28:47 | [D] | [climate:399] | Mode: HEAT
13:28:47 | [D] | [climate:404] | Fan Mode: LOW
13:28:47 | [D] | [climate:410] | Preset: NONE
13:28:47 | [D] | [climate:416] | Swing Mode: OFF
13:28:47 | [D] | [climate:419] | Current Temperature: 22.00°C
13:28:47 | [D] | [climate:425] | Target Temperature: 21.00°C
13:28:47 | [I] | [haier.climate:1053] | HVAC values changed
13:28:47 | [I] | [haier.climate:1056] | HVAC Mode = 0x4
13:28:47 | [I] | [haier.climate:1057] | Fan speed Status = 0x3
13:28:47 | [I] | [haier.climate:1058] | Horizontal Swing Status = 0x5
13:28:47 | [I] | [haier.climate:1059] | Vertical Swing Status = 0xA
13:28:47 | [I] | [haier.climate:1060] | Set Point Status = 0x5
13:28:47 | [W] | [component:237] | Component haier.climate took a long time for an operation (75 ms).
13:28:47 | [W] | [component:238] | Components should block for at most 30 ms.
13:28:47 | [D] | [uart_debug:158] | <<< "\x83\x00\x02\x01\x00\x05\x00\x00,\x00M\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00vpD\x00\x00!\x01\xFFU\x02%\x00X-UU"
13:28:52 | [D] | [haier.protocol:019] | Sending frame: type 01, data: 4D 01
13:28:52 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
13:28:52 | [D] | [uart_debug:158] | >>> "@\x00\x00\x00\x00\x00\x01M\x01\x99\xB3\xB4"
13:28:52 | [D] | [uart_debug:158] | <<< "\xFF\xFF*@\x00\x00\x00\x00\x00\x02m\x01\x05\n"
13:28:52 | [D] | [haier.protocol:019] | Frame found: type 02, data: 6D 01 05 0A 83 00 02 01 00 05 00 00 2C 00 4D 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
13:28:52 | [D] | [haier.protocol:019] | Answer delay 95ms
13:28:52 | [D] | [haier.climate:1056] | HVAC Mode = 0x4
13:28:52 | [D] | [haier.climate:1057] | Fan speed Status = 0x3
13:28:52 | [D] | [haier.climate:1058] | Horizontal Swing Status = 0x5
13:28:52 | [D] | [haier.climate:1059] | Vertical Swing Status = 0xA
13:28:52 | [D] | [haier.climate:1060] | Set Point Status = 0x5
13:28:52 | [D] | [uart_debug:158] | <<< "\x83\x00\x02\x01\x00\x05\x00\x00,\x00M\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xF0\x86\xBD"
13:28:57 | [D] | [haier.protocol:019] | Sending frame: type 01, data: 4D 01
13:28:57 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
13:28:57 | [D] | [uart_debug:158] | >>> "@\x00\x00\x00\x00\x00\x01M\x01\x99\xB3\xB4"
13:29:00 | [W] | [haier.climate:235] | Answer timeout for command 01, phase SENDING_STATUS_REQUEST
13:29:02 | [D] | [haier.protocol:019] | Sending frame: type 01, data: 4D FE
13:29:02 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
13:29:02 | [D] | [uart_debug:158] | >>> "@\x00\x00\x00\x00\x00\x01M\xFE\x96\xF3\xF4"
13:29:03 | [I] | [safe_mode:041] | Boot seems successful; resetting boot loop counter
13:29:03 | [D] | [esp32.preferences:114] | Saving 1 preferences to flash...
13:29:03 | [D] | [esp32.preferences:143] | Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
13:29:05 | [W] | [haier.climate:235] | Answer timeout for command 01, phase SENDING_STATUS_REQUEST
13:29:07 | [D] | [haier.protocol:019] | Sending frame: type 01, data: 4D 01
13:29:07 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
13:29:07 | [D] | [uart_debug:158] | >>> "@\x00\x00\x00\x00\x00\x01M\x01\x99\xB3\xB4"
13:29:10 | [W] | [haier.climate:235] | Answer timeout for command 01, phase SENDING_STATUS_REQUEST
13:29:12 | [D] | [haier.protocol:019] | Sending frame: type 01, data: 4D 01
13:29:12 | [D] | [uart_debug:158] | >>> "\xFF\xFF\n"
13:29:12 | [D] | [uart_debug:158] | >>> "@\x00\x00\x00\x00\x00\x01M\x01\x99\xB3\xB4"
13:29:15 | [W] | [haier.climate:235] | Answer timeout for command 01, phase SENDING_STATUS_REQUEST

```

Pr0mises, 2025-06-29:
> [@Atlas553](https://github.com/Atlas553), What I see in the logs is that AC just not answering anymore. It is strange that you have this kind of problem after recent updates. Can you try ESP32 board [...]

As I've successfully setup this project using an ESP32, I tried my wemos d1 mini due to the smaller form factor, which failed. I'm seeing the same issues with `[I][haier.climate:096]: Answer timeout for command 61, phase SENDING_INIT_1` and `[22:22:46][I][haier.climate:096]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST`.  Due to that, I think wemos d1 is not powerful enough for the new esphome versions...

michalkurzeja, 2025-07-30:
Hi, just 2 cents from me, as I'm also running into issues with the same board, albeit memory-related.

My board crashes on boot, usually when connecting to my wifi network. I have observed 2 errors:
- OOM (confirmed by enabling debug component)
- Exception type: Access to invalid address: STORE (wild pointer?)

Sometimes it would crash multiple times in a row with OOM, sometimes with wild pointer, sometimes a mix of both. I could not find any pattern here.

This is a fresh board, I haven't even connected it to my AC yet, I'm just trying to flash firmware and get it to start up :)

Here's a sample output with the wild pointer error:

```
[19:31:25][I][wifi:630]: Connected
[19:31:25][C][wifi:434]:   Local MAC: 60:01:94:66:91:83
[19:31:25][C][wifi:439]:   SSID: 'Valhalla'
[19:31:26][C][wifi:442]:   IP Address: 192.168.88.155
[19:31:26][C][wifi:446]:   BSSID: 18:E8:29:CA:58:6C
[19:31:26][C][wifi:446]:   Hostname: 'ac-bedroom'
[19:31:26][C][wifi:446]:   Signal strength: -59 dB ▂▄▆█
[19:31:26][V][wifi:453]:   Priority: 0.0
[19:31:26][C][wifi:455]:   Channel: 1
[19:31:26][C][wifi:455]:   Subnet: 255.255.255.0
[19:31:26][C][wifi:455]:   Gateway: 192.168.88.1
[19:31:26][C][wifi:455]:   DNS1: 192.168.88.59
[19:31:26][C][wifi:455]:   DNS2: 192.168.88.1
[19:31:26][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[19:31:26][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 61 00 07 72
[19:31:26][C][api:034]: Running setup
[19:31:26][W][component:279]: api set Warning flag: unspecified
[19:31:26][I][app:091]: setup() finished successfully!
[19:31:26][W][component:307]: wifi cleared Warning flag
[19:31:26][W][component:407]: wifi took a long time for an operation (60 ms)
[19:31:26][W][component:408]: Components should block for at most 30 ms
[19:31:27][W][haier.climate:234]: Answer timeout for command 61, phase SENDING_INIT_1
[19:31:27][W][component:407]: haier.climate took a long time for an operation (252 ms)
[19:31:27][W][component:408]: Components should block for at most 30 ms
[19:31:27][W][component:407]: esphome.ota took a long time for an operation (169 ms)
[19:31:27][W][component:408]: Components should block for at most 30 ms
[19:31:27][W][component:407]: safe_mode took a long time for an operation (168 ms)
[19:31:27][W][component:408]: Components should block for at most 30 ms
[19:31:27][W][component:407]: api took a long time for an operation (166 ms)
[19:31:27][W][component:408]: Components should block for at most 30 ms
[19:31:27][W][component:407]: mdns took a long time for an operation (159 ms)
[19:31:27][W][component:408]: Components should block for at most 30 ms
[19:31:27][W][component:407]: debug took a long time for an operation (160 ms)
[19:31:28][W][component:408]: Components should block for at most 30 ms
[19:31:28][I][app:149]: ESPHome version 2025.7.4 compiled on Jul 30 2025, 19:29:39
[19:31:28][C][wifi:613]: WiFi:
[19:31:28][C][wifi:434]:   Local MAC: 60:01:94:66:91:83
[19:31:28][C][wifi:439]:   SSID: 'Valhalla'
[19:31:28][C][wifi:442]:   IP Address: 192.168.88.155
[19:31:28][C][wifi:446]:   BSSID: 18:E8:29:CA:58:6C
[19:31:28][C][wifi:446]:   Hostname: 'ac-bedroom'
[19:31:28][C][wifi:446]:   Signal strength: -58 dB ▂▄▆█
[19:31:28][V][wifi:453]:   Priority: 0.0
[19:31:28][C][wifi:455]:   Channel: 1
[19:31:28][C][wifi:455]:   Subnet: 255.255.255.0
[19:31:28][C][wifi:455]:   Gateway: 192.168.88.1
[19:31:28][C][wifi:455]:   DNS1: 192.168.88.59
[19:31:28][C][wifi:455]:   DNS2: 192.168.88.1
[19:31:28][C][logger:246]: Logger:
[19:31:28][C][logger:246]:   Max Level: VERBOSE
[19:31:28][C][logger:246]:   Initial Level: VERBOSE
[19:31:28][C][logger:252]:   Log Baud Rate: 9600
[19:31:28][C][logger:252]:   Hardware UART: UART0
[19:31:28][C][uart.arduino_esp8266:118]: UART Bus:
[19:31:28][C][uart.arduino_esp8266:119]:   TX Pin: GPIO15
[19:31:28][C][uart.arduino_esp8266:120]:   RX Pin: GPIO13
[19:31:29][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[... 170 lines cut ...]
      # - configs/sensor/outdoor_coil_temperature.yaml
      # - configs/sensor/outdoor_defrost_temperature.yaml
      # - configs/sensor/outdoor_in_air_temperature.yaml
      # - configs/sensor/outdoor_out_air_temperature.yaml
      # - configs/sensor/power.yaml
      # - configs/sensor/compressor_frequency.yaml
      # - configs/sensor/compressor_current.yaml
      # - configs/sensor/expansion_valve_open_degree.yaml
      # # Diagnostic binary sensors
      # - configs/binary_sensor/outdoor_fan_status.yaml
      # - configs/binary_sensor/defrost_status.yaml
      # - configs/binary_sensor/compressor_status.yaml
      # - configs/binary_sensor/indoor_fan_status.yaml
      # - configs/binary_sensor/four_way_valve_status.yaml
      # - configs/binary_sensor/indoor_electric_heating_status.yaml
      # # Diagnostic text sensors
      # - configs/text_sensor/appliance_name.yaml
      # - configs/text_sensor/protocol_version.yaml
```

Initially, I have imported the entire `haier_hon_base.yaml` but the board kept rebooting (took me a while to figure that out, this is my first encounter with esphome and dev boards). So I started writing minimal config, with fewer components - that's why I have all the configs listed and commented out, I've been enabling them one by one to see when the board booted and when it crashlooped. `display.yaml` is about as far as I got - here I'm getting OOM on startup in a loop. Curiously, even if I comment out more components, it sometimes manages to boot and sometimes not.

This behaviour has been observed without the debugger, I have enabled it last to confirm that, indeed, the heap memory keeps shrinking during the boot process and, around the point of connecting to my wifi network, it runs out.

I've eventually also set up the UART bus on pins 13 and 15 because I thought that maybe logger interferes with how the board operates, but no dice (it's just a measure of my desperation to try and get this to work 😅).

michalkurzeja, 2025-08-01:
And just like for the OP, it works flawlessly on ESP32. Guess 8266 is both too slow and has too little memory for esphome :(