paveldn/haier-esphome #60: ESP 2024.10.0 update - The beeper option is deprecated, use beeper_on/beeper_off actions or beeper switch for a haier platform instead.

andremain, 2024-10-18:
As of the last ESP update i cannot update the firmware of my device stating:
The beeper option is deprecated, use beeper_on/beeper_off actions or beeper switch for a haier platform instead.

"INFO ESPHome 2024.10.0
INFO Reading configuration /config/esphome/bedroom-3-ac-esp32.yaml...
Failed config

climate.haier: [source /config/esphome/bedroom-3-ac-esp32.yaml:49]
  platform: haier
  id: haier_ac
  protocol: hOn
  name: Haier AC
  uart_id: ac_port
  wifi_signal: True
  
  The beeper option is deprecated, use beeper_on/beeper_off actions or beeper switch for a haier platform instead.
  beeper: True
  display: True
  visual: 
    min_temperature: 16 °C
    max_temperature: 30 °C
    temperature_step: 1 °C"

What do i need to do to fix it?

paveldn, 2024-10-18:
@andremain 

Sorry braking changes. The error message says exactly what should be done:
remove the beeper option from climate and use the beeper switch for Haier integration. 
You can see an example here (maximum configuration):
https://github.com/paveldn/haier-esphome/blob/master/docs/hon_example.rst

andremain, 2024-10-18:
IT worked.

Thanks a lot!

paveldn, 2024-10-18:
You are welcome.

closed 2024-10-18 by paveldn