paveldn/haier-esphome #61: Clarification about connecting an ESP32 board to Haier AC

nadimz, 2024-10-23:
Hi,

How to connect an esp32 board to the AC is explained in details in repo's homepage and has been discussed several times in issues. However, I still have some doubts. Apologies if they've been answered before.

I have a Haier AC that uses hOn protocol and plan to use a [seed studio xiao esp3c3](https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/) board with it.

There are two connection options I am considering that have been discussed in #41 :

1. Cutting the original cable coming out from the the AC and wiring it to the esp32 board.
2. Buying [this 4-pin cable](https://es.aliexpress.com/item/1005003374938897.html?gatewayAdapt=glo2esp), wiring it to the esp32 board from one end, and connecting the other end to the AC.

I prefer to go with option 2. In that case, does [this 4-pin cable](https://es.aliexpress.com/item/1005003374938897.html?gatewayAdapt=glo2esp) connect to the cable coming out from the AC? or does it connect to a board that isn't visible from outside?

Regardless of the option I choose, can the cable coming out from the AC be unplugged? I don't want to try to pull it a realize it's soldered and break it.

Thanks for the support!

paveldn, 2024-10-23:
Hi @nadimz,

To use 4 pin cable you will need to open AC, remove one screw, remove the panel, and get access to the AC board. There you can remove the original cable with the same 4 pins at the end (it should be connected to the cn34 socket on the board). You can use your 4-pin cable with this cn34 connector.
Here: https://comf.life/podklyuchaem-kondicioner-haier-flexis-super-match-as25s2sf1fa-g-v-umnyj-dom-wirenboard-cherez-shlyuz-onokom-hr-1-mb-b-b-apple-home-yandex-alisa-spruthub-i-spruthome.html you can see how this process looks like. Sorry for the Russian language. This is a procedure for the installation of some 3rd party hardware but getting to the cn34 plug will look the same. Photos should help you to get what to expect. The wires inside AC are a little bit messy but the procedure itself is not difficult.

nadimz, 2024-10-23:
@paveldn, this is great! Thank you very much for the link. The video makes it very clear now.

Does anyone know what's the name of the connector on [this 4-pin cable](https://es.aliexpress.com/item/1005003374938897.html?gatewayAdapt=glo2esp)? I want to try to source it locally but I have no idea how to refer to it.

paveldn, 2024-10-23:
@nadimz 

Glad it helped. The official name of 4 pin connector on the AC board side is Molex 5264 4p and on the ESP side (smaller one) JST SM04B-GHS-TB.

closed 2024-11-06 by paveldn

nadimz, 2024-11-08:
@paveldn, thanks again for helping out. Got the connector you suggested a few days ago and today I had time to build everything. It works great! No issues.

Just wanted to drop a picture here of the setup in case someone wants to go the similar path.

[image]