paveldn/haier-esphome #62: Configuring a `platform: haier` beeper switch causes esp8266 to crash

roobre, 2024-11-03:
Hey,

I recently updated esphome which brought the new version of the library with the native haier switches/buttons, which I think are great!

I'm however facing some trouble getting the new beeper swtich to work after the deprecation of `beeper: false`. When I try to call `climate.haier.beeper_off`, or just create a beeper switch as in the example:

```yaml
switch:
  - platform: haier
    haier_id: hvac # This is the id in my config
    beeper:
      name: Haier hOn Climate beeper
```

My esp8266 (d1_mini) crashes:

```
[11:30:26][I][app:062]: setup() finished successfully!
[11:30:26][W][component:170]: Component wifi cleared Warning flag
[11:30:26][W][component:157]: Component api set Warning flag: unspecified
[11:30:26][I][app:100]: ESPHome version 2024.10.2 compiled on Nov  3 2024, 11:30:03
[11:30:26][C][wifi:600]: WiFi:
# Redacted
[11:30:26][C][logger:185]: Logger:
[11:30:26][C][logger:186]:   Level: DEBUG
[11:30:26][C][logger:188]:   Log Baud Rate: 115200
[11:30:26][C][logger:189]:   Hardware UART: UART0
[11:30:26][C][uart.arduino_esp8266:118]: UART Bus:
[11:30:26][C][uart.arduino_esp8266:119]:   TX Pin: GPIO4
[11:30:26][C][uart.arduino_esp8266:120]:   RX Pin: GPIO5
[11:30:26][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[11:30:26][C][uart.arduino_esp8266:124]:   Baud Rate: 9600 baud
[11:30:26][C][uart.arduino_esp8266:125]:   Data Bits: 8
[11:30:26][C][uart.arduino_esp8266:126]:   Parity: NONE
[11:30:26][C][uart.arduino_esp8266:127]:   Stop bits: 1
[11:30:26][C][uart.arduino_esp8266:131]:   Using software serial
[11:30:26][C][haier.climate:256]: Haier Climate 'HVAC'
[11:30:26][C][haier.climate:257]:   Device communication status: none
[11:30:26][C][haier.climate:333]:   Protocol version: hOn
[11:30:26][C][haier.climate:334]:   Control method: 1
[11:30:26][C][template.select:065]: Template Select 'Vertical airflow'
[11:30:26][C][template.select:065]:   Icon: 'mdi:arrow-expand-vertical'
[11:30:26][C][template.select:066]:   Update Interval: 60.0s
[11:30:26][C][mdns:116]: mDNS:
[11:30:26][C][mdns:117]:   Hostname: living-room-ac
[11:30:26][C][esphome.ota:073]: Over-The-Air updates:
[11:30:26][C][esphome.ota:074]:   Address: 10.0.3.41:8266
[11:30:26][C][esphome.ota:075]:   Version: 2
[11:30:26][C][esphome.ota:078]:   Password configured
[11:30:26][C][safe_mode:018]: Safe Mode:
[11:30:26][C][safe_mode:019]:   Boot considered successful after 60 seconds
[11:30:26][C][safe_mode:021]:   Invoke after 10 boot attempts
[11:30:26][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[11:30:26][W][safe_mode:028]: Last reset occurred too quickly; safe mode will be invoked in 7 restarts
[11:30:26][C][api:140]: API Server:
[11:30:26][C][api:141]:   Address: 10.0.3.41:6053
[11:30:26][C][api:143]:   Using noise encryption: YES
[11:30:27][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[11:30:27][W][haier.climate:234]: Answer timeout for command 61, phase SENDING_INIT_1
[11:30:37][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[11:30:37][W][haier.climate:234]: Answer timeout for command 61, phase SENDING_INIT_1
[11:30:41][D][api:103]: Accepted 10.0.0.31
[11:30:41][W][api.connection:092]: : Socket operation failed: HANDSHAKESTATE_SETUP_FAILED errno=11
[11:30:41][W][component:170]: Component api cleared Warning flag
[11:30:41][W][component:157]: Component api set Warning flag: unspecified
[11:30:47][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[11:30:47][W][haier.climate:234]: Answer timeout for command 61, phase SENDING_INIT_1
[11:30:56][D][select:015]: 'Vertical airflow': Sending state Center (index 3)
[11:30:57][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[11:30:57][W][haier.climate:234]: Answer timeout for command 61, phase SENDING_INIT_1
[... 75 lines cut ...]
WARNING Decoded 0x4022378e: esphome::Application::loop()
[11:31:00]3fffff80:  0000a989 3fff7d5c 3fff9b54 3fff3b7c
[11:31:00]3fffff90:  3fffad6c 3fff8c34 3fff2234 3fff3b7c
[11:31:00]3fffffa0:  3fffdad0 00000000 3fff3b50 3fff3b7c
[11:31:00]3fffffb0:  3fffdad0 00000000 3fff3b50 40231cf8
WARNING Decoded 0x40231cf8: loop_wrapper() at core_esp8266_main.cpp
[11:31:00]3fffffc0:  feefeffe feefeffe 3fffdab0 40100669
WARNING Decoded 0x40100669: cont_wrapper
[11:31:00]<<<stack<<<
[11:31:00]
[11:31:00]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[11:31:00]
[11:31:00] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[11:31:00]
[11:31:00]load 0x4010f000, len 3424, room 16
[11:31:00]tail 0
[11:31:00]chksum 0x2e
[11:31:00]load 0x3fff20b8, len 40, room 8
[11:31:00]tail 0
[11:31:00]chksum 0x2b
[11:31:00]csum 0x2b
[11:31:00]v000813c0
```

In this case the error occurred after a few seconds. Is something wrong with the example?

roobre, 2024-11-03:
I'm not a CPP expert or anything close, but I can't help but notice that there is nothing related to haier in the trace. However I have verified that adding the haier switch snippet definitely caused this to happen, and vice versa. As the error seems to be in malloc, perhaps somewhere the haier code is looping and exhausting all available memory? Just a wild guess.

paveldn, 2024-11-03:
Hi @roobre,

That is strange. Technically those switches not doing anything it is just mark that every time when command will be sent in future bit that suppress sound feedback should be set to 1. 
I can't reproduce problem with my AC. But I am using ESP32. Can you try with ESP32? Through last releases I see that ESPHome uses more and more resources (providing more and more features of course) but for old ESP8266 it gets more and more difficult to keep up. When you adding switch it includes a lot of switch related code which push your device "over the edge". 

Also one of the thing you can try  is to remove everything related to Haier except plain AC itself and beeper switch. It should be something like this:
```
climate:
  - name: HVAC
    id: hvac
    platform: haier
    protocol: hon
    uart_id: hw0

switch:
  - platform: haier
    haier_id: hvac
    beeper:
      name: Beeper
```

Remove all other buttons sensors and selects. Try to reproduce same problem. THanks.

nadiamoe, 2025-03-09:
Hey @paveldn, thanks for your patience waiting for a response, it took me a while to get back to this.

I've replaced the ESP8266 (D1 Mini) with an ESP32-C3 (cheapo "SuperMini" board from Ali) and the latest version of everything seems to be running fine. So I guess this might have been due to 8266 not having enough memory after all.

Not sure what to do with the issue. The problem is solved for me, but I still think that the fact of versions newer than 2024.9.2 not being runnable on the 8266 stands. I'll leave that up to you.

Huge thanks again for the work you're doing here, this integration is invaluable to me! ❤

closed 2025-03-10 by paveldn