paveldn/haier-esphome #63: Issue with quiet mode.

Jeroen2006, 2024-11-12:
Hello,
I'm running a few Haier units running the hon protocol in my house. It has been working perfectly (thanks for the great work). 
I am running into one issue though. I cannot seem to get the quiet mode on the unit activated from my esphome controller. I have the switch configured as the examples I was able to find.
```
switch:
  - platform: haier
    quiet_mode:
      name: Quiet mode
```

But when I toggle the switch in homeassistant it doesn't seem to do anything. I had a look at the incoming packets from the airconditioner, and when i changed the state I wasn't able to see any bytes change. However, when I used the included remote and enabled the quiet mode from there I was able to see something change (I tried it a few times to make sure it wasn't anything else). When I used the remote I was able to hear the unit make less noise, but the switch in homeassistant did not change state.

Running in normal mode:
`Frame found: type 02, data: 6D 01 08 06 85 00 00 01 00 00 00 00 32 00 4D 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

Running in quiet mode:
`Frame found: type 02, data: 6D 01 08 06 85 00 00 11 00 00 00 00 32 00 4C 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

Also when I toggle the quiet mode switch in homeassistant I don't see the message 'Sending control packet, queue size 1' printed in the log, but when I change the display for example I do see that message,
Is this something I missed somewhere or is the quiet mode broken?

Here is my config for the climate device:
```
climate:
  - platform: haier
    id: haier_ac
    protocol: hon
    name: Haier hOn Climate
    uart_id: haier_uart
    wifi_signal: true
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 
        target_temperature: 1
        current_temperature: 0.5
    supported_modes:
      - 'OFF'
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
      - FAN_ONLY
    supported_swing_modes:
      - 'OFF'
      - VERTICAL
    supported_presets:
      - AWAY
      - BOOST
      - SLEEP
    on_status_message:
      then:
        - output.turn_on: haier_led
        - delay: 100ms
        - output.turn_off: haier_led
        - delay: 250ms
        - output.turn_on: haier_led
        - delay: 100ms
        - output.turn_off: haier_led
    on_alarm_start:
      then:
        - homeassistant.service:
            service: logbook.log
            data:
              domain: climate
              name: Haier hOn Climate
            data_template:
              message: "Alarm activated ({{ alarm_code }}): {{alarm_message}}" 
            variables:
              alarm_message: !lambda "return message;"
              alarm_code: !lambda "return code;"
        - homeassistant.service:
            service: notify.persistent_notification
            data:
              title: "Haier hOn Climate: alarm activated"
            data_template:
              message: "Code: {{ alarm_code }}, message: \"{{ alarm_message }}\""
            variables:
              alarm_message: !lambda "return message;"
              alarm_code: !lambda "return code;"
    on_alarm_end:
      then:
        - homeassistant.service:
            service: logbook.log
            data:
              domain: climate
              name: Haier hOn Climate
            data_template:
              message: "Alarm deactivated ({{ alarm_code }}): {{alarm_message}}" 
            variables:
              alarm_message: !lambda "return message;"
              alarm_code: !lambda "return code;"
```

Kind regards
Jeroen

paveldn, 2024-11-13:
Hi Jeroen,

Can you please record logs for me with 15-20 seconds of messages before and 15-20 seconds after action:

* Turning quiet mode ON by the home assistant
* Turning quiet mode OFF by the home assistant
* Turning quiet mode ON by IR remote
* Turning quiet mode OFF by IR remote

Jeroen2006, 2024-11-13:
Yes, no problem!
I turned on silent mode with the IR remote @ 17:11:59, and turned it off again @ 17:12:21

[haier-log-toggle-ha.txt](https://github.com/user-attachments/files/17735589/haier-log-toggle-ha.txt)
[haier-log-toggle-ir.txt](https://github.com/user-attachments/files/17735697/haier-log-toggle-ir.txt)

paveldn, 2024-11-14:
I see you are using it in FAN mode. Can you try using quiet in another mode, like heat or cool? I am blocking quiet mode for fan mode. It is not working with my AC, so I thought it was not allowed.

Jeroen2006, 2024-11-19:
Sorry, didn't notice my remote was still set to fan mode
[logs_haier-woonkamer_logs.txt](https://github.com/user-attachments/files/17819271/logs_haier-woonkamer_logs.txt)

paveldn, 2024-11-26:
@Jeroen2006  Sorry missed your message. I will check your logs today

paveldn, 2024-11-26:
@Jeroen2006 Yes you are right my fault will make a fix for quiet mode this weekend.

paveldn, 2024-12-01:
@Jeroen2006 I think I fixed it (together with other small issues). Can you please check it. You will need to use dev branch. 
Thanks.

Jeroen2006, 2024-12-01:
@paveldn It appears to be working. I can enable quiet mode from HASS and I see the switch state toggle if I turn it on/off from the IR remote. Thank you very much!

paveldn, 2024-12-01:
@Jeroen2006  Great! Thanks for checking. I will prepare pull request for esphome.

closed 2024-12-01 by paveldn