paveldn/haier-esphome #64: Boost and quiet mode separate from presets.

gammeflamme, 2024-12-20:
Hi my ac just switches preset to none when I try to apply a new one. I assume its cause it doesn't support presets but I would like to use the boost and quiet modes that the Ir remote supports. I have attached a summary of the log and the full log of the communication.
This is the Ir remote command order.

Turn boost on
Turn boost off
Turn quiet on
Turn quiet off

[ac log summary](https://github.com/user-attachments/files/18213815/ac.log.summary.txt)
[Haier ac log](https://github.com/user-attachments/files/18213816/Haier.ac.log.txt)

paveldn, 2024-12-26:
Hi @gammeflamme,

It is difficult to see clearly which bits changed. Can you make separate logs for every action (turn boost on, turn bust off, turn quiet on, turn quiet off)? 20–30 seconds each.

Also, can you tell me the model of your AC? I don't have smartAir2 AC to do experiments, so I will need your help.

gammeflamme, 2024-12-27:
The model is AS12TA2HRA but I'm unsure about separate logs. Do you want me to separate the log that's attached or do you want me to reset the ESP, send a signal and download the logs for every signal or is it just that I need to wait longer between Ir signals? 

The attached log has four [I] line blocks and each [I] Block appears right after an Ir signal is received.
The first [I] block is when I send the Turn on boost signal
the second is when I turn boost off
the third is turn on quiet
the fourth is when I turn quiet off.

The attached log summary contains the string after "Frame found: type 02, data:" and when there is an [I] block I add what ir command I sent to the summary. Based on how they look I assume that the frame found data before the [I] Block is when the Ir signal is received causing an [I] block to appear after it.

I'll gladly provide logs and tests but the ac is in a vacation home so that limits when I can send ir commands. I have remote access so testing code is fine most of the time.

What I mean with [I] Block

[18:35:16][I][haier.climate:537]: HVAC values changed
[18:35:16][I][haier.climate:540]: HVAC Mode = 0x2
[18:35:16][I][haier.climate:541]: Fan speed Status = 0x0
[18:35:16][I][haier.climate:542]: Horizontal Swing Status = 0x0
[18:35:16][I][haier.climate:543]: Vertical Swing Status = 0x0
[18:35:16][I][haier.climate:544]: Set Point Status = 0xE

Thanks for the fantastic work.

paveldn, 2024-12-30:
Hi @gammeflamme,

I am asking you to make separate logs for every event. From one common logs not very clear which bit is responsible for which mode. I want to have one log for turning boost mode on from IR remote, another for boost mode off, third for sleep mode on, forth for sleep off.
In this case it would be easier for me to detect changes.
All actions should be made from IR remote.

gammeflamme, 2025-01-06:
Hi! I think I understand, thank you for clarifying. Here is the same common log but separated by what IR signal I sent. I'm sorry for misunderstanding if that isn't what you meant. I'm unsure what state the ac should be in when sending the IR signal. 
All these logs are in heating mode, I don't remember the swing state but I think the temperature was set to 27C so that it wouldn't switch off heating in an attempt to make the logs clearer.

1. Boost and Quiet mode is off when sending Boost on 
[Boost mode on.txt](https://github.com/user-attachments/files/18318898/Boost.mode.on.txt)
2. Only Boost mode is on when sending Boost off
[Boost mode off.txt](https://github.com/user-attachments/files/18318900/Boost.mode.off.txt)
3. Boost and Quiet mode is off  when sending Quiet on
[Quiet mode on.txt](https://github.com/user-attachments/files/18318901/Quiet.mode.on.txt)
4. Only Quiet mode is on when sending Quiet off
[Quiet mode off.txt](https://github.com/user-attachments/files/18318905/Quiet.mode.off.txt)

paveldn, 2025-01-06:
Hi! Thanks. Unfortunately I will not have time to check logs today and tomorrow. I will try to do it on Wednesday.