paveldn/haier-esphome #65: Steri-Clean 56 not working on AS25S2SF2FA

Ot4eVash, 2024-12-24:
Good afternoon! Thank you for the excellent integration! I’ve encountered the following issue. The high-temperature cleaning mode "haier_hon_climate_start_56_c_steri_cleaning" is not working. According to the logs, the air conditioner starts the dehumidification mode and then turns off. Unfortunately, this mode is not available on the remote control and was previously only launched through the EVO app. Is it possible to resolve this issue somehow? The log is attached, the event occurred at [16:14:34].
 AC: AS25S2SF2FA - 2pc. (Multi-split) 
[logs_ac-bedroom_logs.txt](https://github.com/user-attachments/files/18239894/logs_ac-bedroom_logs.txt)

paveldn, 2024-12-26:
@Ot4eVash 
Everything looks OK. Esp sending correct command but for some reason AC resets it back.
Can you record what is happenig when you push steri clean in EVO app with orioginal ESP?
It looks like it doing something special in your case.

Ot4eVash, 2024-12-28:
Thank you for your reply! Sorry, I may have misunderstood. How can I capture logs from the EVO app? Or do you mean simply recording a video of what happens with the air conditioner when I send the command?

To explain in words: when I press "steri-clean" in the app, the indoor unit displays "CL" and the process starts.
With ESPhome, "CL" lights up for half a second, but then the air conditioner immediately turns off. According to Home Assistant, before shutting down, it seems to switch to "DRY" mode for some reason.

Ot4eVash, 2024-12-28:
> @Ot4eVash Everything looks OK. Esp sending correct command but for some reason AC resets it back. Can you record what is happenig when you push steri clean in EVO app with orioginal ESP? It looks like it doing something special in your case.

Thank you for your reply! Sorry, I may have misunderstood. How can I capture logs from the EVO app? Or do you mean simply recording a video of what happens with the air conditioner when I send the command?

To explain in words: when I press "steri-clean" in the app, the indoor unit displays "CL" and the process starts.
With ESPhome, "CL" lights up for half a second, but then the air conditioner immediately turns off. According to Home Assistant, before shutting down, it seems to switch to "DRY" mode for some reason.

paveldn, 2024-12-28:
> Thank you for your reply! Sorry, I may have misunderstood. How can I capture logs from the EVO app? Or do you mean simply recording a video of what happens with the air conditioner when I send the command?

I wm afraid it is not that easy. Here I have instruction https://github.com/paveldn/haier-esphome/blob/master/docs%2Fsniffing_serial_communication.rst

Ot4eVash, 2024-12-28:
> I wm afraid it is not that easy. Here I have instruction https://github.com/paveldn/haier-esphome/blob/master/docs%2Fsniffing_serial_communication.rst

Thank you! I’ll figure it out and share the results.