paveldn/haier-esphome #70: External temperature sensor?

Smandurlo, 2025-03-16:
Hi, 
I am using some wemos d1 mini hon protocol. Is it possible to use an external temperature sensor as reference for the ac?

Thank you

paveldn, 2025-03-17:
Hi @Smandurlo Unfortunately no. At least, it is not possible at the moment. Serial protocol for Wi-Fi module doesn't support this. 

Maybe it would be possible to do it using wired or IR remotes, but this is a completely different protocol and story.

Smandurlo, 2025-03-17:
hi @paveldn thank you so much for your reply. It is a pity :-( I have made controllers for mitsubishi, panasonic, lg and daikin and all of them accept external temperature sensors...

Never mind! Thank you for your work!

paveldn, 2025-03-18:
You are welcome. It was a disappointment for me as well. AC's like Midea (follow me feature https://esphome.io/components/climate/midea.html#midea-ac-follow-me-action ) have a way to use external temperature sensors through IR remote protocol. I saw mentions on the internet that Haier has models that also support it, but I didn't find remotes that support it.

closed 2025-03-18 by paveldn

Smandurlo, 2025-03-18:
> You are welcome. It was a disappointment for me as well. AC's like Midea (follow me feature https://esphome.io/components/climate/midea.html#midea-ac-follow-me-action ) have a way to use external [...]

out of curiosity, sending the temperature through IR will make the AC beep?