paveldn/haier-esphome #71: Away mode not working

codeprutser, 2025-03-18:
When i set away mode in home assistant the AC beeps but it is not set to 10 degrees Celcius. The preset is also reset to none. When i use the remote to set away mode (Extra function > 10c > Confirm) the preset in home assistant is correctly set to Away mode.

My Haier AC model is: AS25PBPHRA-PRE
haier-esphome Firmware: 2024.12.2

paveldn, 2025-03-20:
Hi @codeprutser  

Can you provide me with 2 logs? One when you enable it with IR remote, another one when you enable it with Home assistant. Please include at least 20 seconds of logs before action and 20 seconds after.

paveldn, 2025-03-25:
Hi @codeprutser,

Yes, it looks like something is wrong. Let me check the code, and I will get back to you.

codeprutser, 2025-03-28:
I now also have a different Haier Airco indoor unit for the living room that is connected to the same outdoor unit. This is model number: AS71S2SF1FA-MW3. Let me now if it useful to get logging from this unit.

paveldn, 2025-03-30:
Hi, sorry for long delay, couldn't find time to dig into the code. Unfortunately, I can't find a reason. In code, everything quite straight forward. The only thing required is a Heat mode on (which you have). I will make a test version with more logging to figure out what is happening.