paveldn/haier-esphome #72: FAN_ONLY not working with smartair2

thatdevsherry, 2025-03-18:
Hi, I'm using this module using smartair2 and everything works except "FAN_ONLY" mode and  "AWAY" preset. My remote doesn't have away preset so I believe it wouldn't work.

But fan mode is available in remote.

# Config

```yaml
esphome:
  name: living-room-ac
  friendly_name: living-room-ac

esp32:
  board: esp32dev

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "***="

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: "***"
  password: "***"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "****"
    password: "****"

uart:
  id: ac_port
  tx_pin: GPIO17  # Connect to AC RX
  rx_pin: GPIO16  # Connect to AC TX
  baud_rate: 9600

# Example configuration entry
climate:
  - platform: haier
    id: living_room_ac_haier
    protocol: smartair2
    alternative_swing_control: true
    name: Living room AC
    uart_id: ac_port
    wifi_signal: true
    display: true
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
    supported_modes:
      - 'OFF'
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
      - FAN_ONLY
    supported_swing_modes:
      - 'OFF'
      - VERTICAL
      - HORIZONTAL
      - BOTH
    supported_presets:
      # Commented out. Selecting this goes back to 'None'
      #- AWAY
      - BOOST
      - COMFORT
    on_status_message:
      then:
        - logger.log:
            level: INFO
            format: "New status message received, size=%d, subcmd=%02X%02X"
            args: [ 'data_size', 'data[0]', 'data[1]' ]

# Example configuration entry
switch:
  - platform: haier
    haier_id: living_room_ac_haier
    display:
      name: Haier display
```

# esphome logs

Using DEBUG logging.

```
[19:47:10][D][climate:011]: 'Living room AC' - Setting
[19:47:10][D][climate:015]:   Mode: FAN_ONLY
[19:47:10][D][Control:372]: Control call
[19:47:10][W][haier.climate:378]: New settings come faster then processed!
[19:47:10][I][haier.climate:182]: Sending control packet
[19:47:10][D][haier.protocol:019]: Sending frame: type 01, data: 4D 5F 00 18 00 37 00 00 00 00 00 00 00 03 00 D0 00 00 00 01 00 00 00 00 00 00
[19:47:10][D][uart_debug:114]: >>> FF:FF:22:00:00:00:00:00:00:01:4D:5F:00:18:00:37:00:00:00:00:00:00:00:03:00:D0:00:00:00:01:00:00:00:00:00:00:F2
[19:47:10][D][uart_debug:114]: <<< FF:FF:0A
[19:47:10][D][haier.protocol:019]: Frame found: type 03, data: 00 00
[19:47:10][D][haier.protocol:019]: Answer delay 146ms
[19:47:10][W][haier.protocol:013]: Answer handler error, msg=01, answ=03, err=7
[19:47:10][D][uart_debug:114]: <<< 00:00:00:00:00:00:03:00:00:0D
```

paveldn, 2025-03-20:
Hi @thatdevsherry .

Away and Fan-only depend on your settings and do not work the same way for different AC modes. My component sends commands to AC but it doesn't mean that AC will accept it. Try to switch FAN mode to low speed in COOL mode and then switch to Fan only.

thatdevsherry, 2025-03-20:
Makes sense. It's the AC that isn't accepting it.

Somehow just reuploading esphome on 2 AC made fan mode work.

As for the other one, I was able to make it work by turning on fan mode from IR remote. The AC then accepts fan mode from esphome after that.

closed 2025-03-20 by thatdevsherry