paveldn/haier-esphome #73: Completely New to ESPhome

noviardi1986, 2025-03-25:
Hi can anyone help me?
This is my  first project on esphome and still very confusing to used it. Usualy i'm using tasmota.
when i tried tests/esp8266-smartair2-wifi.yaml i got error

INFO Reading configuration esp8266-smartair2-wifi.yaml...
ERROR Unable to import component haier.switch:
Traceback (most recent call last):
  File "C:\Users\Qiang\AppData\Local\Programs\Python\Python38\lib\site-packages\esphome\loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "C:\Users\Qiang\AppData\Local\Programs\Python\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "..\components\haier\switch\__init__.py", line 5, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_DISPLAY' from 'esphome.const' (C:\Users\Qiang\AppData\Local\Programs\Python\Python38\lib\site-packages\esphome\const.py)
ERROR Unable to import component haier.switch:
Traceback (most recent call last):
  File "C:\Users\Qiang\AppData\Local\Programs\Python\Python38\lib\site-packages\esphome\loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "C:\Users\Qiang\AppData\Local\Programs\Python\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "..\components\haier\switch\__init__.py", line 5, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_DISPLAY' from 'esphome.const' (C:\Users\Qiang\AppData\Local\Programs\Python\Python38\lib\site-packages\esphome\const.py)
ERROR Unable to import component haier.climate:
Traceback (most recent call last):
  File "C:\Users\Qiang\AppData\Local\Programs\Python\Python38\lib\site-packages\esphome\loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "C:\Users\Qiang\AppData\Local\Programs\Python\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "..\components\haier\climate.py", line 7, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_DISPLAY' from 'esphome.const' (C:\Users\Qiang\AppData\Local\Programs\Python\Python38\lib\site-packages\esphome\const.py)
Failed config

switch.haier: [source ..\configs/switch/health_mode.yaml:2]

  Platform not found: 'switch.haier'.
  platform: haier
  health_mode:
    name: Haier AC health mode
switch.haier: [source ..\configs/switch/display.yaml:2]

  Platform not found: 'switch.haier'.
  platform: haier
  display:
    name: Haier AC display
api: [source ..\configs/api/service.turn_off.yaml:1]
  services:
    - service: turn_off
      then:
        -
          Unable to find action with the name 'climate.haier.power_off'.
          climate.haier.power_off: haier_climate
    - service: turn_on
      then:
        - climate.haier.power_on: haier_climate
climate.haier: [source ..\configs/climate/haier_smartair2.yaml:2]

  Platform not found: 'climate.haier'.
  platform: haier
  id: haier_climate
  protocol: smartAir2
  name: Haier AC
  uart_id: ac_port
  alternative_swing_control: False
  wifi_signal: true
  visual:
    min_temperature: 16 °C
    max_temperature: 30 °C
    temperature_step: 1 °C
  supported_modes:
    - OFF
    - HEAT_COOL
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
  supported_swing_modes:
    - OFF
    - VERTICAL
    - HORIZONTAL
    - BOTH
  supported_presets:
    - BOOST
    - COMFORT

paveldn, 2025-03-25:
Hi @noviardi1986 

Welcome to ESPHome world.!
What version of ESPHome are you using? Can you provide complete configuration (just remove keys and passwords)?

noviardi1986, 2025-03-25:
Hi @paveldn Thank for your reply. i'm using Version: 2022.9.4 in windows  pc. i tried to validate in HAOS  no error where found. i think  is the PC versin problem. The problem already solve, next step will be wiring to the  AC  unit. will tried in weekend..

paveldn, 2025-03-25:
Your esphome is too old. Haier component just don't exist in that version. Take at least 2024 versions

noviardi1986, 2025-03-26:
i'm installing in windows pc for test  purpose by followed the official website tutorial
https://esphome.io/guides/installing_esphome.html
maybe it is already abadoned for the esphome pip. in the home assistant addon the code are running perfect.
1 more question my AC are using MK-QTWIFI-15 that using Haismart App. can i  just connect the rx and tx directly to AC or i needc a logic converter..

paveldn, 2025-03-26:
Strange no pip should give you the latest version. I am using pip to install esphome for my home PC (both Windows and Linux). 
Please use

pip install -U esphome

It should pull the latest esphome with dependencies.

As for your AC, I can't give you a 100% correct answer without trying it, but it almost always works without a logic converter.

noviardi1986, 2025-04-12:
Thanks @paveldn

closed 2025-04-12 by noviardi1986

noviardi1986, 2025-04-13:
As for the esphome old version in my pc are caused by outdated pyhoon version.
https://github.com/esphome/esphome/discussions/5393