paveldn/haier-esphome #76: Connector Idea

noviardi1986, 2025-04-14:
If do not want to cut the unit side connector cable.
we can remove the original female connector from the haier module and making a converter from jst to female dupont connector 

[image]
[image]
[image]

paveldn, 2025-04-14:
@noviardi1986  

Great job. Your soldering skills are much better than mine. 😄 
In my home setup, I used a different approach: I made a conversion cable using https://www.aliexpress.com/item/1005006243133251.html?spm=a2g0o.order_list.order_list_main.349.6aa31802iYTgGP cable (for AC board side), added a USB type C cable to another, and used [M5Stamp C3U (white color)](https://shop.m5stack.com/products/m5stamp-c3u-mate-with-pin-headers) boards. This board allows to use USB connector as UART. Here is more info: https://github.com/paveldn/haier-esphome/blob/master/docs/usb_2_uart_boards.rst

noviardi1986, 2025-04-14:
Thank you for the compliment about soldering, compared to your contribution it was nothing. as if anyboby trying to make solder this connecto  be  awared of the  polaritie of the voltage input..

mardouwevisser, 2025-06-19:
Speaking of ideas, I use this module for my air conditioners: [https://www.tinytronics.nl/en/development-boards/microcontroller-boards/with-wi-fi/universal-air-conditioning-controller-esp32-s3](https://www.tinytronics.nl/en/development-boards/microcontroller-boards/with-wi-fi/universal-air-conditioning-controller-esp32-s3)
It is practically plug-and-play and it also comes with a build-in level converter!