paveldn/haier-esphome #77: is candy supported as ac brand?

borgqueenx, 2025-04-17:
I want to buy a new candy ac from 2024, and wonder if these are supported?

paveldn, 2025-04-17:
@borgqueenx  Sorry, I have no idea. Previously, there were models of Candy ACs that were produced by Haier, but it doesn't mean all of them are supported. I suppose the only way is to try.

borgqueenx, 2025-04-17:
I know they support hon, but hon only has a cloud api.
I would hate if my internet goes down that i cant control the ac. I suppose my chances are ok but stays a gamble then, thanks for the quick answer.

paveldn, 2025-04-17:
If they support hOn, chances that it will work are quite high. But there is always a chance that there can be a new version of the protocol or some additional protection.