paveldn/haier-esphome #78: Any guide to follow to a new esp board creation for this integration?

borgqueenx, 2025-04-19:
being quite new to esphome, the documentation is a bit complicated.
Im trying to find out how it would work. i see that a new esp module can be bought and used, and i see these little boards are crazy cheap.
i already have esphome in home assistant for precense sensors but required no messing around.
So how does this work exactly, is there a guide to follow?
Do i insert a usb cable in the AC unit, then have the 4 wires in there go to the rx, tx, ground and 3.3v positions? that seems pretty easily do-able, but what then? what code do i flash on the esphome?
I'm quite confused on it all after reading it a few times.

paveldn, 2025-04-19:
Hi @borgqueenx 
Setting up a module for a Haier AC is quite similar to creating any other ESPHome-based module. You'll need an ESP board, wiring to connect it to your AC, and a suitable ESPHome configuration.
Your AC should provide 5V power, ground, TX, and RX lines for UART communication. That's essentially all you need. The quickest way is to take the original cable, cut and solder the wires to the corresponding pins on your ESP board, create a configuration, build the firmware, flash the board, and connect it to the AC unit.
You can find example configurations here (depending on the protocol used):
[HON protocol example](https://github.com/paveldn/haier-esphome/blob/master/docs/hon_example.rst)
[SmartAir2 protocol example](https://github.com/paveldn/haier-esphome/blob/master/docs/smartair2_example.rst)

borgqueenx, 2025-04-19:
Thanks a lot, i am probably going to buy the haier geos r+ and try to get this working!