paveldn/haier-esphome #79: Tweeking control_packet_size and sensors_packed_size

noviardi1986, 2025-05-02:
Been playing around on both setting, i got off sensor reading particulary temperature and humidity
the yamlbelow
  - platform: haier
    id: matteo_ac_aqua
    protocol: hon
    name: Aqua AC
    uart_id: ac_port
    wifi_signal: true
    display: false
    answer_timeout: 400ms
    control_packet_size: 11
    sensors_packet_size: 22
    # control_packet_size: 12
    # sensors_packet_size: 22
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step:
        target_temperature: 1
        current_temperature: 0.5
    supported_modes:
      - 'OFF'
      - COOL
      - DRY
      - FAN_ONLY
    supported_swing_modes:
      - 'OFF'
      - VERTICAL
    supported_presets:
      - AWAY
      - BOOST
      - SLEEP
    on_status_message:
      then:
        - logger.log:
            level: INFO
            format: "New status message received, size=%d, subcmd=%02X%02X"
            args: [ 'data_size', 'data[0]', 'data[1]' ]
if i set control_packet_size size to 12 the temperature are reading correctly but humidity  0%
vice versa when i change the control_packet_size to 11, the humidity read Ok but the temperature read 0C and log  "[W][haier.climate:864]: HVAC error, code=0x80"
how to use control_packet_size correctly?

noviardi1986, 2025-05-02:
below some log when i set control_packet_size 12 and sensors_packet_size 22
tem normal, humidity  0%
[15:28:51][D][select:015]: 'Haier hOn Climate airflow horizontal': Sending state Auto (index 5)
[15:28:52][D][haier.protocol:019]: Sending frame: type 70, data: <empty>
[15:28:52][D][select:015]: 'Haier hOn Climate airflow vertical': Sending state Center (index 3)
[15:28:52][D][haier.protocol:019]: Frame found: type 71, data: 20 08 61 08 00 82 03 24 02 12 00 11 80 12 56 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40
[15:28:52][D][haier.protocol:019]: Answer delay 184ms
[15:28:54][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[15:28:54][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 05 0C 25 00 02 02 00 07 08 0C 00 00 3C 00 5B 80 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[15:28:54][D][haier.protocol:019]: Answer delay 197ms
[15:28:54][W][haier.climate:864]: HVAC error, code=0x5B
[15:28:54][D][sensor:093]: 'Haier hOn Climate outdoor temperature': Sending state -4.00000 °C with 0 decimals of accuracy
[15:28:54][D][sensor:093]: 'Haier hOn Climate Indoor Humidity': Sending state 0.00000 % with 0 decimals of accuracy
[15:28:54][D][climate:396]: 'Aqua AC' - Sending state:
[15:28:54][D][climate:399]:   Mode: OFF
[15:28:54][D][climate:404]:   Fan Mode: AUTO
[15:28:54][D][climate:410]:   Preset: NONE
[15:28:54][D][climate:416]:   Swing Mode: VERTICAL
[15:28:54][D][climate:419]:   Current Temperature: 0.00°C
[15:28:54][D][climate:425]:   Target Temperature: 21.00°C
[15:28:54][I][haier.climate:1068]: HVAC values changed
[15:28:54][I][haier.climate:1071]: HVAC Mode = 0x1
[15:28:54][I][haier.climate:1072]: Fan speed Status = 0x5
[15:28:54][I][haier.climate:1073]: Horizontal Swing Status = 0x7
[15:28:54][I][haier.climate:1074]: Vertical Swing Status = 0xC
[15:28:54][I][haier.climate:1075]: Set Point Status = 0x5
[15:28:54][I][main:370]: New status message received, size=36, subcmd=6D01
[15:28:54][I][haier.climate:220]: First HVAC status received
[15:28:54][W][component:239]: Component haier.climate took a long time for an operation (96 ms).
[15:28:54][W][component:240]: Components should block for at most 30 ms.
[15:28:56][D][haier.protocol:019]: Sending frame: type 73, data: <empty>
[15:28:56][D][haier.protocol:019]: Frame found: type 74, data: 0F 5A 00 00 00 00 00 00 00 00
[15:28:56][D][haier.protocol:019]: Answer delay 153ms
[15:28:56][D][select:015]: 'Haier hOn Climate airflow horizontal': Sending state Auto (index 5)
[15:28:56][I][haier.climate:449]: Sending control packet, queue size 1
[15:28:56][D][haier.protocol:019]: Sending frame: type 01, data: 60 01 05 0C 25 00 00 82 00 07 08 0C 00 00
[15:28:56][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 05 0C 25 00 02 02 00 07 08 0C 00 00 3C 00 5B 80 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[15:28:56][D][haier.protocol:019]: Answer delay 199ms
[15:28:56][D][sensor:093]: 'Haier hOn Climate outdoor temperature': Sending state 27.00000 °C with 0 decimals of accuracy
[15:28:56][D][climate:396]: 'Aqua AC' - Sending state:
[15:28:56][D][climate:399]:   Mode: OFF
[15:28:57][D][climate:404]:   Fan Mode: AUTO
[15:28:57][D][climate:410]:   Preset: NONE
[15:28:57][D][climate:416]:   Swing Mode: VERTICAL
[15:28:57][D][climate:419]:   Current Temperature: 30.00°C
[15:28:57][D][climate:425]:   Target Temperature: 21.00°C
[15:28:57][I][haier.climate:1068]: HVAC values changed
[15:28:57][I][haier.climate:1071]: HVAC Mode = 0x1
[15:28:57][I][haier.climate:1072]: Fan speed Status = 0x5
[15:28:57][I][haier.climate:1073]: Horizontal Swing Status = 0x7
[15:28:57][I][haier.climate:1074]: Vertical Swing Status = 0xC
[15:28:57][I][haier.climate:1075]: Set Point Status = 0x5
[15:28:57][I][main:370]: New status message received, size=36, subcmd=6D01
[15:28:57][D][select:015]: 'Haier hOn Climate airflow vertical': Sending state Auto (index 7)
[15:28:58][D][haier.protocol:019]: Sending frame: type FC, data: <empty>
[15:28:58][D][haier.protocol:019]: Frame found: type FD, data: 00 00 00 00 00 00
[15:28:58][D][haier.protocol:019]: Answer delay 151ms
[15:29:00][D][haier.climate:117]: WiFi signal is: -54dBm => 57%
[15:29:00][D][haier.protocol:019]: Sending frame: type F7, data: 00 00 00 39
[15:29:00][D][haier.protocol:019]: Frame found: type 05, data: <empty>
[15:29:00][D][haier.protocol:019]: Answer delay 152ms
[15:29:01][D][select:015]: 'Haier hOn Climate airflow horizontal': Sending state Auto (index 5)
[15:29:02][D][select:015]: 'Haier hOn Climate airflow vertical': Sending state Auto (index 7)
[15:29:02][D][haier.protocol:019]: Sending frame: type 01, data: 4D FE
[15:29:03][D][haier.protocol:019]: Frame found: type 02, data: 7D 01 05 0C 25 00 02 02 00 07 08 0C 00 00 3C 00 5B 80 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 62 6E 00 00 00 00 00 00 02 20 00 00
[15:29:03][D][haier.protocol:019]: Answer delay 204ms
[15:29:03][D][sensor:093]: 'Haier hOn Climate Indoor Coil Temperature': Sending state 29.00000 °C with 0 decimals of accuracy
[15:29:03][D][sensor:093]: 'Haier hOn Climate Outdoor Coil Temperature': Sending state -64.00000 °C with 0 decimals of accuracy
[15:29:03][D][sensor:093]: 'Haier hOn Climate Outdoor Defrost Temperature': Sending state -64.00000 °C with 0 decimals of accuracy
[15:29:03][D][sensor:093]: 'Haier hOn Climate Outdoor In Air Temperature': Sending state -64.00000 °C with 0 decimals of accuracy
[... 98 lines cut ...]
[15:29:36][D][haier.climate:1071]: HVAC Mode = 0x1
[... cut ...]
[15:29:41][D][haier.climate:1075]: Set Point Status = 0x5
[15:29:41][I][main:370]: New status message received, size=36, subcmd=6D01
[15:29:41][D][select:015]: 'Haier hOn Climate airflow horizontal': Sending state Auto (index 5)
[15:29:42][D][select:015]: 'Haier hOn Climate airflow vertical': Sending state Auto (index 7)
[15:29:42][D][haier.protocol:019]: Sending frame: type FC, data: <empty>
[15:29:43][D][haier.protocol:019]: Frame found: type FD, data: 00 00 00 00 00 00
[15:29:43][D][haier.protocol:019]: Answer delay 152ms

paveldn, 2025-05-02:
Hi @noviardi1986 ,

Sorry, can't check your logs for you at the moment. I am traveling now without my pc and it is not easynto check it from my phone. I will be back at 6th of May. You can try to find parameters yourself using this instruction https://github.com/paveldn/haier-esphome/blob/master/docs/protocol_overview.rst In case it didn't work I will help you once I am back from my trip

noviardi1986, 2025-05-03:
Well have a nice trip...