paveldn/haier-esphome #81: not working - Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST

ser8611, 2025-05-16:
Good morning,

with this code:
```
esphome:
  name: ac-salotto
  friendly_name: AC salotto

esp8266:
  board: esp01_1m

logger:
 baud_rate: 0

api:
  encryption:
    key: "xx"

ota:
  - platform: esphome
    password: "xx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.158
    gateway: 192.168.1.1
    subnet: 255.255.255.0  

  ap:
    ssid: "Ac-Salotto Fallback Hotspot"
    password: "xx"

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600
  id: ac_uart

external_components:
  - source:
      type: git
      url: https://github.com/paveldn/haier-esphome
    components: [ haier ]
    refresh: 60s

climate:
  - platform: haier
    name: "Haier AC Salotto"
    id: haier_climate
    uart_id: ac_uart
```

i have this log, and it doesn't work

```
INFO ESPHome 2025.4.2
INFO Reading configuration /config/esphome/ac-salotto.yaml...
INFO Updating https://github.com/paveldn/haier-esphome@None
INFO Detected log level for Haier protocol: DEBUG
INFO Starting log output from 192.168.1.158 using esphome API
INFO Successfully connected to ac-salotto @ 192.168.1.158 in 0.013s
INFO Successful handshake with ac-salotto @ 192.168.1.158 in 2.412s
[09:04:10][I][app:100]: ESPHome version 2025.4.2 compiled on May 16 2025, 07:51:42
[09:04:10][C][wifi:600]: WiFi:
[09:04:10][C][wifi:428]:   Local MAC: 24:4C:AB:51:89:24
[09:04:10][C][wifi:433]:   SSID: [redacted]
[09:04:10][C][wifi:436]:   IP Address: 192.168.1.158
[09:04:10][C][wifi:439]:   BSSID: [redacted]
[... 93 lines cut ...]
[09:05:39][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[09:05:41][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[09:05:41][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[09:05:41][D][haier.protocol:019]: Answer delay 73ms
[09:05:41][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:05:43][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:05:45][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:05:45][I][haier.climate:096]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[09:05:51][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[09:05:53][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[09:05:53][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[09:05:53][D][haier.protocol:019]: Answer delay 87ms
[09:05:53][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:05:55][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:05:57][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:05:57][I][haier.climate:096]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[09:06:03][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[09:06:05][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[09:06:05][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[09:06:05][D][haier.protocol:019]: Answer delay 78ms
[09:06:05][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:06:07][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:06:09][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[09:06:09][I][haier.climate:096]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
```

paveldn, 2025-05-16:
Hi @ser8611 
Use hOn protocol. Like here:

```
climate:
  - platform: haier
    protocol: hon
    name: "Haier AC Salotto"
    id: haier_climate
    uart_id: ac_uart
```

ser8611, 2025-05-16:
Hi @paveldn , thanks a lot for the fast reply...

perfect, its works perfectly!!