paveldn/haier-esphome #82: Can't send control packet, first poll answer not received

PaulBiod, 2025-05-21:
Hi,
At first thanks for your work.
I have an "equation/haier" AC with the wifi dongle from haier.
I follow the guide to install a d1 mini.
I use the minimal configuration, I tried with hon and smartair2, the same result : if I try to switch on the AC, i get the error : Can't send control packet, first poll answer not received

I enabled verbose.

my conf : 

`esphome:
  name: clim-bas
  friendly_name: Clim bas

esp8266:
  board: esp01_1m

# Enable logging
logger:
  baud_rate : 0
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Clim-Bas Fallback Hotspot"
    password: "MaGpiCtwqYHW"

captive_portal:

uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 9600

      

climate:
  - platform: haier
    name: clim_bas
    protocol: hon

`

And the log : 

`
[17:03:57][I][app:100]: ESPHome version 2024.12.4 compiled on May 21 2025, 17:02:51
[17:03:57][C][wifi:600]: WiFi:
[17:03:57][C][wifi:428]:   Local MAC: 8C:AA:B5:15:D6:BD
[17:03:57][C][wifi:433]:   SSID: [redacted]
[17:03:57][C][wifi:436]:   IP Address: 192.168.1.201
[17:03:57][C][wifi:439]:   BSSID: [redacted]
[17:03:57][C][wifi:441]:   Hostname: 'clim-bas'
[17:03:57][C][wifi:443]:   Signal strength: -24 dB ▂▄▆█
[17:03:57][V][wifi:445]:   Priority: 0.0
[17:03:57][C][wifi:447]:   Channel: 4
[17:03:57][C][wifi:448]:   Subnet: 255.255.255.0
[... 225 lines cut ...]
[17:12:32][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:12:32][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 01 4D 01 59
[17:12:34][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:12:34][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 01 4D 01 59
[17:12:34][I][haier.climate:096]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[17:12:34][V][haier.climate:075]: Phase transition: SENDING_FIRST_STATUS_REQUEST => SENDING_INIT_1
[17:12:40][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[17:12:40][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 61 00 07 72
[17:12:42][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[17:12:42][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 61 00 07 72
[17:12:44][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[17:12:44][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 61 00 07 72
[17:12:44][I][haier.climate:096]: Answer timeout for command 61, phase SENDING_INIT_1
[17:12:44][V][haier.climate:075]: Phase transition: SENDING_INIT_1 => SENDING_INIT_2
[17:12:44][V][haier.climate:075]: Phase transition: SENDING_INIT_2 => SENDING_FIRST_STATUS_REQUEST
[17:12:44][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[17:12:44][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 01 4D 01 59`

Any help would be appreciated..
Thanks

ps: welding is ugly but there is no shorts and it seems ok

[image]
[image]

paveldn, 2025-05-22:
Hi @PaulBiod  
There is no answer from AC, so it should be one of: AC doesn't see ESP requests, or ESP doesn't see AC answers (or both). AC is not sending any data, just answering. Have you tried the most obvious thing, swapping TX <-> RX?
Your issue is somewhere in hardware, so I can help a lot. 
I have small instruction here, please take a look: https://github.com/paveldn/haier-esphome/blob/master/docs/faq.rst#q-i-connected-the-esp-module-but-all-i-can-see-in-the-logs-is-answer-timeout-for-command-61-phase-sending_init_1-and-nothing-is-working-what-should-i-do

PaulBiod, 2025-05-22:
Hi @paveldn 
Thanks for taking time to help.
Sure, first thing I did was switching tx - rx in the code.
Today I'm gonna try another pins, and another esp, maybe I'll use an esp32 as I didn't have too much luck in my projects with d1 mini.
Do you agree that as my AC uses the haier dongle, I should be able to make it work ?  or it could be another protocol that I'd need to sniff ?
Thanks

paveldn, 2025-05-22:
If you want to switch ESP32 I would recommend you try one of these:
https://github.com/paveldn/haier-esphome/blob/master/docs/usb_2_uart_boards.rst
So you would be able to use USB as UART and don't need wiring at all (just maybe type c to type A converter)
I would say I 99% sure we can make your AC work with ESPHome. In Haier ACs I haven't seen any other protocol yet, all of them variations of smartAir2 or hOn. Haier module on your photo can do both hOn and smartAir2 so you will need to try both to find which one supported by your AC.

PaulBiod, 2025-05-22:
Thanks for this info.
I'm trying a spare esp32 wroom 32D, when I connect it to esphome, it seems it can do uart from usb, isn't it ?

[image]

PaulBiod, 2025-05-22:
Hi @paveldn 
I tried with an esp32 wroom 32U, same problem.
The wiring of the tx/rx is done through the tx/rx pins on the esp32, so pin 1 and 3 (I tried reverse the pins also)

```
INFO Upload took 9.09 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.88 using esphome API
INFO Successfully connected to ac-bas @ 192.168.1.88 in 11.315s
INFO Successful handshake with ac-bas @ 192.168.1.88 in 0.142s
[12:10:54][I][app:100]: ESPHome version 2024.12.4 compiled on May 22 2025, 12:10:10
[12:10:54][C][wifi:600]: WiFi:
[12:10:54][C][wifi:428]:   Local MAC: 40:22:D8:02:9A:60
[12:10:54][C][wifi:433]:   SSID: [redacted]
[12:10:54][C][wifi:436]:   IP Address: 192.168.1.88
[12:10:54][C][wifi:440]:   BSSID: [redacted]
[12:10:54][C][wifi:441]:   Hostname: 'ac-bas'
[12:10:54][C][wifi:443]:   Signal strength: -17 dB ▂▄▆█
[12:10:54][V][wifi:445]:   Priority: 0.0
[12:10:54][C][wifi:447]:   Channel: 5
[12:10:54][C][wifi:448]:   Subnet: 255.255.255.0
[12:10:54][C][wifi:449]:   Gateway: 192.168.1.1
[12:10:54][C][wifi:450]:   DNS1: 192.168.1.1
[12:10:54][C][wifi:451]:   DNS2: 192.168.1.1
[12:10:54][C][logger:185]: Logger:
[12:10:54][C][logger:186]:   Level: VERBOSE
[12:10:54][C][logger:188]:   Log Baud Rate: 0
[12:10:54][C][logger:189]:   Hardware UART: UART0
[12:10:54][C][uart.arduino_esp32:151]: UART Bus 0:
[12:10:54][C][uart.arduino_esp32:152]:   TX Pin: GPIO1
[12:10:54][C][uart.arduino_esp32:153]:   RX Pin: GPIO3
[12:10:54][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[12:10:54][C][uart.arduino_esp32:157]:   Baud Rate: 9600 baud
[12:10:54][C][uart.arduino_esp32:158]:   Data Bits: 8
[12:10:54][C][uart.arduino_esp32:159]:   Parity: NONE
[12:10:54][C][uart.arduino_esp32:160]:   Stop bits: 1
[12:10:54][C][haier.climate:256]: Haier Climate 'clim_bas'
[12:10:54][C][haier.climate:257]:   Device communication status: none
[12:10:54][C][haier.climate:125]:   Protocol version: smartAir2
[12:10:54][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[12:10:54][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 61 00 07 72
[12:10:54][D][uart_debug:158]: >>> "\xFF\xFF\n"
[12:10:54][C][mdns:116]: mDNS:
[12:10:54][C][mdns:117]:   Hostname: ac-bas
[12:10:54][V][mdns:118]:   Services:
[12:10:54][V][mdns:120]:   - _esphomelib, _tcp, 6053
[12:10:54][V][mdns:122]:     TXT: friendly_name = AC bas
[12:10:54][V][mdns:122]:     TXT: version = 2024.12.4
[12:10:54][V][mdns:122]:     TXT: mac = 4022d8029a60
[12:10:54][V][mdns:122]:     TXT: platform = ESP32
[12:10:54][V][mdns:122]:     TXT: board = esp32dev
[12:10:54][V][mdns:122]:     TXT: network = wifi
[12:10:54][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[12:10:54][C][esphome.ota:073]: Over-The-Air updates:
[12:10:54][C][esphome.ota:074]:   Address: ac-bas.local:3232
[12:10:54][C][esphome.ota:075]:   Version: 2
[12:10:54][C][esphome.ota:078]:   Password configured
[12:10:54][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[12:10:54][V][haier.protocol:022]: Received data: FF FF 2E 40 00 00 00 00 00 62 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04 5B 8A AB
[12:10:54][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[12:10:54][D][haier.protocol:019]: Answer delay 84ms
[12:10:54][V][haier.climate:075]: Phase transition: SENDING_INIT_1 => SENDING_INIT_2
[12:10:54][C][safe_mode:018]: Safe Mode:
[12:10:54][C][safe_mode:020]:   Boot considered successful after 60 seconds
[12:10:54][C][safe_mode:021]:   Invoke after 10 boot attempts
[12:10:54][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[12:10:54][V][haier.climate:075]: Phase transition: SENDING_INIT_2 => SENDING_FIRST_STATUS_REQUEST
[12:10:54][C][api:140]: API Server:
[12:10:54][C][api:141]:   Address: ac-bas.local:6053
[... 14 lines cut ...]
[12:10:58][I][haier.climate:097]: Answer timeout for command 01, phase SENDING_FIRST_STATUS_REQUEST
[12:10:58][V][haier.climate:075]: Phase transition: SENDING_FIRST_STATUS_REQUEST => SENDING_INIT_1
[12:11:03][D][climate:011]: 'clim_bas' - Setting
[12:11:03][D][climate:015]:   Mode: COOL
[12:11:03][D][Control:372]: Control call
[12:11:03][W][haier.climate:374]: Can't send control packet, first poll answer not received
[12:11:04][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[12:11:04][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 61 00 07 72
[12:11:04][D][uart_debug:158]: >>> "\xFF\xFF\n"
[12:11:04][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[12:11:06][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[12:11:06][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 61 00 07 72
[12:11:06][D][uart_debug:158]: >>> "\xFF\xFF\n"
[12:11:06][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[12:11:06][V][haier.protocol:022]: Received data: FF FF 2E 40 00 00 00 00 00 62 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04 5B 8A AB
[12:11:06][D][haier.protocol:019]: Frame found: type 62, data: 45 2B 2B 32 2E 31 38 00 31 37 30 36 32 36 30 30 F1 00 00 31 37 30 35 32 36 30 30 01 55 2D 41 43 00 00 00 00 00 04
[12:11:06][D][haier.protocol:019]: Answer delay 87ms
[... cut ...]
[12:11:06][V][haier.climate:075]: Phase transition: SENDING_INIT_2 => SENDING_FIRST_STATUS_REQUEST
[12:11:06][D][uart_debug:158]: <<< "\xFF\xFF.@\x00\x00\x00\x00\x00bE++2.18\x0017062600\xF1\x00\x0017052600\x01U-AC\x00\x00\x00\x00\x00\x04[\x8A\xAB"
[12:11:06][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[12:11:06][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 01 4D 01 59
[12:11:06][D][uart_debug:158]: >>> "\xFF\xFF\n"
[12:11:06][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00\x01M\x01Y"
```

PaulBiod, 2025-05-22:
GOOD news :)
Now it works  (with HON protocol)

[image]

```
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/ac-bas.yaml...
INFO Detected log level for Haier protocol: VERBOSE
INFO Generating C++ source...
INFO Compiling app...
Processing ac-bas (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.6
|-- HaierProtocol @ 0.9.31
Compiling .pioenvs/ac-bas/src/main.cpp.o
Linking .pioenvs/ac-bas/firmware.elf
RAM:   [=         ]  14.4% (used 47236 bytes from 327680 bytes)
Flash: [=====     ]  50.6% (used 927861 bytes from 1835008 bytes)
Building .pioenvs/ac-bas/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/ac-bas/firmware.bin"], [".pioenvs/ac-bas/firmware.elf"])
Wrote 0xf3f00 bytes to file /data/build/ac-bas/.pioenvs/ac-bas/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/ac-bas/firmware.bin"], [".pioenvs/ac-bas/firmware.elf"])
========================= [SUCCESS] Took 10.26 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.88 port 3232...
INFO Connected to 192.168.1.88
INFO Uploading /data/build/ac-bas/.pioenvs/ac-bas/firmware.bin (933632 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 9.97 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.88 using esphome API
INFO Successfully connected to ac-bas @ 192.168.1.88 in 11.311s
INFO Successful handshake with ac-bas @ 192.168.1.88 in 0.098s
[12:15:23][I][app:100]: ESPHome version 2024.12.4 compiled on May 22 2025, 12:14:52
[12:15:23][C][wifi:600]: WiFi:
[12:15:23][C][wifi:428]:   Local MAC: 40:22:D8:02:9A:60
[12:15:23][C][wifi:433]:   SSID: [redacted]
[12:15:23][C][wifi:436]:   IP Address: 192.168.1.88
[12:15:23][C][wifi:440]:   BSSID: [redacted]
[12:15:23][C][wifi:441]:   Hostname: 'ac-bas'
[12:15:23][C][wifi:443]:   Signal strength: -16 dB ▂▄▆█
[12:15:23][V][wifi:445]:   Priority: 0.0
[12:15:23][C][wifi:447]:   Channel: 5
[12:15:23][C][wifi:448]:   Subnet: 255.255.255.0
[12:15:23][C][wifi:449]:   Gateway: 192.168.1.1
[12:15:23][C][wifi:450]:   DNS1: 192.168.1.1
[12:15:23][C][wifi:451]:   DNS2: 192.168.1.1
[12:15:23][C][logger:185]: Logger:
[12:15:23][C][logger:186]:   Level: VERBOSE
[12:15:23][C][logger:188]:   Log Baud Rate: 0
[12:15:23][C][logger:189]:   Hardware UART: UART0
[12:15:23][C][uart.arduino_esp32:151]: UART Bus 0:
[12:15:23][C][uart.arduino_esp32:152]:   TX Pin: GPIO1
[12:15:23][C][uart.arduino_esp32:153]:   RX Pin: GPIO3
[12:15:23][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[12:15:23][C][uart.arduino_esp32:157]:   Baud Rate: 9600 baud
[12:15:23][C][uart.arduino_esp32:158]:   Data Bits: 8
[12:15:23][C][uart.arduino_esp32:159]:   Parity: NONE
[12:15:23][C][uart.arduino_esp32:160]:   Stop bits: 1
[... 458 lines cut ...]
[12:17:16][D][haier.climate:1071]: Fan speed Status = 0x3
[12:17:16][D][haier.climate:1072]: Horizontal Swing Status = 0x0
[12:17:16][D][haier.climate:1073]: Vertical Swing Status = 0x0
[12:17:16][D][haier.climate:1074]: Set Point Status = 0x7
[12:17:16][V][haier.climate:075]: Phase transition: SENDING_STATUS_REQUEST => IDLE
[12:17:16][W][component:237]: Component haier.climate took a long time for an operation (52 ms).
[12:17:16][W][component:238]: Components should block for at most 30 ms.
[12:17:16][D][uart_debug:158]: <<< "\xFF\xFF*@\x00\x00\x00\x00\x00\x02m\x01\a\x00#\x00\x02\x01\x00\x00\x00\x001\x00U\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x90\t\xEB"
[12:17:21][V][haier.climate:075]: Phase transition: IDLE => SENDING_STATUS_REQUEST
[12:17:21][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[12:17:21][V][haier.protocol:022]: Sending data: FF FF 0A 40 00 00 00 00 00 01 4D 01 99 B3 B4
[12:17:21][D][uart_debug:158]: >>> "\xFF\xFF\n"
[12:17:21][D][uart_debug:158]: >>> "@\x00\x00\x00\x00\x00\x01M\x01\x99\xB3\xB4"
[12:17:21][V][haier.protocol:022]: Received data: FF FF 2A 40 00 00 00 00 00 02 6D 01 07 00 23 00 02 01 00 00 00 00 31 00 55 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 90 09 EB
[12:17:21][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 07 00 23 00 02 01 00 00 00 00 31 00 55 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[12:17:21][D][haier.protocol:019]: Answer delay 88ms
[12:17:21][D][haier.climate:1070]: HVAC Mode = 0x1
[12:17:21][D][haier.climate:1071]: Fan speed Status = 0x3
[12:17:21][D][haier.climate:1072]: Horizontal Swing Status = 0x0
[12:17:21][D][haier.climate:1073]: Vertical Swing Status = 0x0
[12:17:21][D][haier.climate:1074]: Set Point Status = 0x7
[12:17:21][V][haier.climate:075]: Phase transition: SENDING_STATUS_REQUEST => IDLE
[12:17:21][D][uart_debug:158]: <<< "\xFF\xFF*@\x00\x00\x00\x00\x00\x02m\x01\a\x00#\x00\x02\x01\x00\x00\x00\x001\x00U\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x90\t\xEB"

```

paveldn, 2025-05-22:
Glad it worked!

PaulBiod, 2025-05-22:
Thanks for your awesome work !

closed 2025-05-22 by PaulBiod