paveldn/haier-esphome #88: ACS hp300s1 haier

manias21, 2025-07-13:
I accept help!

[23:59:01][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[23:59:01][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 61 00 07 72
[23:59:01][D][uart_debug:158]: >>> "\xFF\xFF\n"
[23:59:01][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[23:59:01][V][haier.protocol:022]: Received data: FF FF 56 00 00 00 00 00 00 89 00 00 00 00 00
[23:59:01][V][haier.protocol:022]: Received data: 00 00 4F 50 4D 4E 4E 00 00 32 32 32 59 59
[23:59:01][V][haier.protocol:022]: Received data: 32 59 53 32 32 32 00 00 00 00 00 00 00 00 00 00
[23:59:01][V][haier.protocol:022]: Received data: 00 00 00
[23:59:01][V][haier.protocol:022]: Received data: 00 00 00 00 00 00 00 32 32 00 00 00 00 32 32
[23:59:01][V][haier.protocol:022]: Received data: E5 77 00
[23:59:01][V][haier.protocol:022]: Received data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[23:59:01][V][haier.protocol:022]: Received data: 00 00 00 00
[23:59:01][V][haier.protocol:022]: Received data: 00 00 00 00 47
[23:59:01][D][haier.protocol:019]: Frame found: type 89, data: 00 00 00 00 00 00 00 4F 50 4D 4E 4E 00 00 32 32 32 59 59 32 59 53 32 32 32 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 32 00 00 00 00 32 32 E5 77 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[23:59:01][D][haier.protocol:019]: Answer delay 117ms
[23:59:01][V][haier.climate:075]: Phase transition: SENDING_INIT_1 => SENDING_INIT_2
[23:59:01][V][haier.climate:075]: Phase transition: SENDING_INIT_2 => SENDING_FIRST_STATUS_REQUEST
[23:59:01][D][uart_debug:158]: <<< "\xFF\xFFV\x00\x00\x00\x00\x00\x00\x89\x00\x00\x00\x00\x00\x00\x00OPMNN\x00\x00222YY2YS222\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0022\x00\x00\x00\x0022\xE5w\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00G"

[23:59:03][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[23:59:03][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 01 4D 01 59
[23:59:03][D][uart_debug:158]: >>> "\xFF\xFF\n"
[23:59:03][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00\x01M\x01Y"
[23:59:03][V][haier.protocol:022]: Received data: FF FF 56 00 00 00 00 00 00 89 00 00 00 00 00
[23:59:03][V][haier.protocol:022]: Received data: 00 00 4F 50 4D 4E 4E 00 00 32 32 32 59 59 32
[23:59:03][V][haier.protocol:022]: Received data: 59 53 32 32 32 00 00 00 00 00 00 00 00 00 00 00
[23:59:03][V][haier.protocol:022]: Received data: 00 00 00
[23:59:03][V][haier.protocol:022]: Received data: 00 00 00 00 00 00 32 32 00 00 00 00 32 32 E5
[23:59:03][V][haier.protocol:022]: Received data: 77 00 00
[23:59:03][V][haier.protocol:022]: Received data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[23:59:03][V][haier.protocol:022]: Received data: 00 00 00
[23:59:03][V][haier.protocol:022]: Received data: 00 00 47
[23:59:03][D][haier.protocol:019]: Frame found: type 89, data: 00 00 00 00 00 00 00 4F 50 4D 4E 4E 00 00 32 32 32 59 59 32 59 53 32 32 32 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 32 00 00 00 00 32 32 E5 77 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[23:59:03][D][haier.protocol:019]: Answer delay 120ms
[23:59:03][V][haier.climate:075]: Phase transition: SENDING_FIRST_STATUS_REQUEST => SENDING_INIT_1
[23:59:03][W][haier.protocol:013]: Answer handler error, msg=01, answ=89, err=1
[23:59:03][D][uart_debug:158]: <<< "\xFF\xFFV\x00\x00\x00\x00\x00\x00\x89\x00\x00\x00\x00\x00\x00\x00OPMNN\x00\x00222YY2YS222\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0022\x00\x00\x00\x0022\xE5w\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00G"
[23:59:13][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[23:59:13][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 61 00 07 72
[23:59:13][D][uart_debug:158]: >>> "\xFF\xFF\n"
[23:59:13][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[23:59:13][V][haier.protocol:022]: Received data: FF FF 56 00 00 00 00 00 00 89 00 00 00 00 00 00
[23:59:13][V][haier.protocol:022]: Received data: 00 4F 50 4D 4E 4E 00 00 32 32 32 59 59 32
[23:59:13][V][haier.protocol:022]: Received data: 59 53 32 32 32 00 00 00 00 00 00 00 00 00 00
[23:59:13][V][haier.protocol:022]: Received data: 00 00 00
[23:59:13][V][haier.protocol:022]: Received data: 00 00 00 00 00 00 00 32 32 00 00 00 00 32 32
[23:59:13][V][haier.protocol:022]: Received data: E5 77 00 00
[23:59:13][V][haier.protocol:022]: Received data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[23:59:13][V][haier.protocol:022]: Received data: 00 00 00
[23:59:13][V][haier.protocol:022]: Received data: 00 00 00 47
[23:59:13][D][haier.protocol:019]: Frame found: type 89, data: 00 00 00 00 00 00 00 4F 50 4D 4E 4E 00 00 32 32 32 59 59 32 59 53 32 32 32 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 32 00 00 00 00 32 32 E5 77 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[23:59:13][D][haier.protocol:019]: Answer delay 119ms
[23:59:13][V][haier.climate:075]: Phase transition: SENDING_INIT_1 => SENDING_INIT_2
[23:59:13][V][haier.climate:075]: Phase transition: SENDING_INIT_2 => SENDING_FIRST_STATUS_REQUEST
[23:59:13][D][uart_debug:158]: <<< "\xFF\xFFV\x00\x00\x00\x00\x00\x00\x89\x00\x00\x00\x00\x00\x00\x00OPMNN\x00\x00222YY2YS222\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0022\x00\x00\x00\x0022\xE5w\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00G"

[23:59:15][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[23:59:15][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 01 4D 01 59
[23:59:15][D][uart_debug:158]: >>> "\xFF\xFF\n"
[23:59:15][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00\x01M\x01Y"
[23:59:15][V][haier.protocol:022]: Received data: FF FF 56 00 00 00 00 00 00 89 00 00 00 00 00 00
[23:59:15][V][haier.protocol:022]: Received data: 00 4F 50 4D 4E 4E 00 00 32 32 32 59 59 32
[23:59:15][V][haier.protocol:022]: Received data: 59 53 32 32 32 00 00 00 00 00 00 00 00 00 00
[23:59:15][V][haier.protocol:022]: Received data: 00 00 00
[23:59:15][V][haier.protocol:022]: Received data: 00 00 00 00 00 00 00 32 32 00 00 00 00 32 32
[23:59:15][V][haier.protocol:022]: Received data: E5 77 00
[23:59:15][V][haier.protocol:022]: Received data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[23:59:15][V][haier.protocol:022]: Received data: 00 00 00 00 00 00
[... 53 lines cut ...]
[23:59:49][VV][api.connection:190]: Sending keepalive PING
[23:59:49][VV][api.service:013]: send_message ping_request: PingRequest {}
[23:59:49][VV][api.service:068]: on_ping_response: PingResponse {}
[23:59:49][D][haier.protocol:019]: Sending frame: type 61, data: 00 07
[23:59:49][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 61 00 07 72
[23:59:49][D][uart_debug:158]: >>> "\xFF\xFF\n"
[23:59:49][D][uart_debug:158]: >>> "\x00\x00\x00\x00\x00\x00a\x00\ar"
[23:59:49][V][haier.protocol:022]: Received data: FF FF 56 00 00 00 00 00 00 89 00 00 00 00 00 00
[23:59:49][V][haier.protocol:022]: Received data: 00 4F 50 4D 4E 4E 00 00 32 32 32 59 59 32
[23:59:49][V][haier.protocol:022]: Received data: 59 53 32 32 32 00 00 00 00 00 00 00 00 00 00 00
[23:59:49][V][haier.protocol:022]: Received data: 00 00 00
[23:59:49][V][haier.protocol:022]: Received data: 00 00 00 00 00 00 32 32 00 00 00 00 32 32
[23:59:49][V][haier.protocol:022]: Received data: E5 77 00
[23:59:49][V][haier.protocol:022]: Received data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[23:59:49][VV][api.connection:190]: Sending keepalive PING
[23:59:49][VV][api.service:013]: send_message ping_request: PingRequest {}
[23:59:49][V][haier.protocol:022]: Received data: 00 00 00 00 00 00 47
[23:59:49][D][haier.protocol:019]: Frame found: type 89, data: 00 00 00 00 00 00 00 4F 50 4D 4E 4E 00 00 32 32 32 59 59 32 59 53 32 32 32 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 32 00 00 00 00 32 32 E5 77 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[23:59:49][D][haier.protocol:019]: Answer delay 109ms
[23:59:49][V][haier.climate:075]: Phase transition: SENDING_INIT_1 => SENDING_INIT_2
[23:59:49][V][haier.climate:075]: Phase transition: SENDING_INIT_2 => SENDING_FIRST_STATUS_REQUEST
[23:59:49][VV][api.service:068]: on_ping_response: PingResponse {}
[23:59:49][D][uart_debug:158]: <<< "\xFF\xFFV\x00\x00\x00\x00\x00\x00\x89\x00\x00\x00\x00\x00\x00\x00OPMNN\x00\x00222YY2YS222\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0022\x00\x00\x00\x0022\xE5w\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00G"
[23:59:51][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[23:59:51][V][haier.protocol:022]: Sending data: FF FF 0A 00 00 00 00 00 00 01 4D 01 59

paveldn, 2025-07-14:
Hi @manias21 

Protocol doesn't look correct. Frames itself have a proper structure but data inside have completely different layout. What kind of device you are trying to use?

paveldn, 2025-07-14:
I am afraid this heater will not work with my component. It will require separate integration. Transport level protocol is compatible but not data inside packets. Same things as with Haier fridges, washing machines etc.

manias21, 2025-07-14:
True, I'm convinced of that. It's different. I've been testing it because I can't find any other way to control it. Do you see a way to move forward with it?I couldn't find any Haier modules where I could view their questions and answers; it's a jungle.

paveldn, 2025-07-14:
To make it work you will need to figure out meaning of every (or at least most l) byte in a packet, discover how to send control commands, find if there is any kind of additional packets etc. all this will require a lot of work and experiments. Without device to do experiments I can't do it. Sorry

paveldn, 2025-07-14:
I am actually happy that somebody using my work. So I don't mind at all. Just not sure how I can help with this.

manias21, 2025-07-16:
## How the Wall Control Communicates with CPU

Example of ON/OFF communication from Wall Control to CPU:

```
[2025-07-15 13:53:26] CONTROL → CPU ff ff 06 01 10 26 01 07 45   < Control turns ON (00 → 07)
[2025-07-15 13:53:27] CONTROL → CPU ff ff 06 01 10 26 01 00 3e   < Control turns OFF (07 → 00)

Repeated example:
[2025-07-15 13:56:59] CONTROL → CPU ff ff 06 01 10 29 01 07 48
[2025-07-15 13:56:59] CPU → CONTROL ff ff 0c 01 a1 23 04 00 00 00 00 00 52 00 27
[2025-07-15 13:56:59] CONTROL → CPU ff ff 06 01 10 29 01 00 41
[2025-07-15 13:57:01] CPU → CONTROL ff ff 0c 01 91 23 04 00 00 00 00 00 52 00 17

[2025-07-15 13:57:02] CONTROL → CPU ff ff 06 01 10 29 01 07 48
[2025-07-15 13:57:02] CPU → CONTROL ff ff 0c 01 90 03 04 00 00 00 00 00 00 00 a4
[2025-07-15 13:57:03] CONTROL → CPU ff ff 06 01 10 29 01 00 41
```

---

### Message Structure (CONTROL → CPU)

| Byte Position | Description              | Notes                                |
| ------------- | ------------------------ | ------------------------------------ |
| 1-2           | Message Start            | Always `ff ff`                       |
| 3             | Length                   | Always `06`                          |
| 4             | Address/Type             | Always `01`                          |
| 5             | Main Command             | `10` (Control command)               |
| 6             | Subcommand (Function ID) | E.g., `26`, `29` (function specific) |
| 7             | Constant/Action type     | Always `01`                          |
| 8             | State (ON/OFF)           | `07`=ON, `00`=OFF                    |
| 9             | Checksum                 | Variable (message integrity)         |

### CPU Response Interpretation (CPU → CONTROL)

| Byte 5 (response) | State Interpretation     |
| ----------------- | ------------------------ |
| `a1`              | Active state             |
| `91`              | Inactive state (standby) |
| `90`              | Special state/error      |

---

### CPU Status Report Structure

```
BYTE:  0  1  2  3  4  5  6  7  8  9 10 11 12 13 14
       ff ff 0c 01 __ __ 04 __ 00 00 00 00 __ 00 __
```

| Byte 4 | Byte 7 | Byte 12 | Clear State |
| ------ | ------ | ------- | ----------- |
| `90`   | `00`   | `00`    | 🔴          |

|   |
| - |

| **CONTROL OFF** |      |      |                             |
| --------------- | ---- | ---- | --------------------------- |
| `91`            | `00` | `52` | 🟢 **CONTROL ON (standby)** |
| `a1`            | `00` | `52` | 🟢 **CONTROL ON (active)**  |

Examples:

CONTROL OFF:

```
ff ff 0c 01 90 03 04 00 00 00 00 00 00 00 a4
```

CONTROL ON (standby):

```
ff ff 0c 01 91 23 04 00 00 00 00 00 52 00 17
```

CONTROL ON (active):

```
ff ff 0c 01 a1 23 04 00 00 00 00 00 52 00 27
```

---

### Interpretation of Commands Sent from CONTROL

| Byte 6 | Byte 8 | State Interpretation                  |
| ------ | ------ | ------------------------------------- |
| `26`   | `07`   | 🟢 **CONTROL SENDS TURN ON COMMAND**  |
| `26`   | `00`   | 🔴 **CONTROL SENDS TURN OFF COMMAND** |
| `29`   | `07`   | 🟢 **CONTROL SENDS TURN ON COMMAND**  |
| `29`   | `00`   | 🔴 **CONTROL SENDS TURN OFF COMMAND** |

Examples:

CONTROL commands ON (function 26):

```
ff ff 06 01 10 26 01 07 45
```

CONTROL commands OFF (function 26):

```
ff ff 06 01 10 26 01 00 3e
```

CONTROL commands ON (function 29):

```
ff ff 06 01 10 29 01 07 48
```

CONTROL commands OFF (function 29):

```
ff ff 06 01 10 29 01 00 41
```

---

### Brief Conclusion (CONTROL → CPU)

- **Byte 6 (**`**, **`**)**: Specifies the function being toggled.
- **Byte 8 (**`**, **`**)**: Indicates the command (ON/OFF).
- **Checksum (Byte 9)**: Confirms message integrity.

### General Summary

**CONTROL → CPU:**

- `26` or `29` + `07` → 🟢 TURN ON COMMAND
- `26` or `29` + `00` → 🔴 TURN OFF COMMAND

**CPU → CONTROL:**

- `90` + `00` + `00` → 🔴 CONTROL OFF
- `91` + `00` + `52` → 🟢 CONTROL ON (standby)
- `a1` + `00` + `52` → 🟢 CONTROL ON (active)

paveldn, 2025-07-17:
Hi @manias21 
From info you provided, I see that control packet structure is different. In Haier protocol, I implemented packets have much bigger header. So deep modification of the protocol will be required including also transport layer (HaierProtocol library I implemented will not work out of the box). And because I don't have appliance on the hands there will be no way to do experiments it will be too difficult. Sorry, I just don't have enough free time. You need to do it yourself of find somebody with same device who can do it. I can help you to understand how my code works but I can't do protocol analysis and implementation for you.